zigbeefordomoticz/Domoticz-Zigbee #5: 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.

Th1h4ck, 2017-11-15:
Salut,

je teste ton plugin pour ma zigate, mais j'ai un petit problème :
```
2017-11-15 21:51:33.330 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.330 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.330 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.330 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.330 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.330 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.330 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.330 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.330 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.330 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.331 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.331 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.331 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.331 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.331 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.331 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.331 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.533 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.533 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.533 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.533 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.533 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.534 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.534 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.534 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.534 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.534 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.534 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.534 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.534 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.534 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 21:51:33.534 Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
```

J'ai pour le moment seulement inclus un interrupteur xiaomi (le rond), et lors, de l'appui, j'ai ça dans la log.
Domoticz : V3.8160 Build Hash: 7c34639
Ubuntu 16.04.3 LTS (GNU/Linux 4.4.0-98-generic x86_64)
Python 3.5.2

La log en mode debug au redémarrage du plugins : 
```
2017-11-15 22:03:26.122  (Zigate USB) Exiting work loop...
2017-11-15 22:03:30.878  (Zigate USB) Stopped.
2017-11-15 22:03:30.954  (Zigate USB) Initialized version 1.0.2, author 'zaraki673'
2017-11-15 22:03:30.958  (Zigate USB) onStart called
2017-11-15 22:03:30.958  (Zigate USB) Debug log level set to: 'true'.
2017-11-15 22:03:30.959  (Zigate USB) Protocol set to: 'None'.
2017-11-15 22:03:30.959  (Zigate USB) Transport set to: 'Serial', '/dev/ttyUSB15', 115200.
2017-11-15 22:03:30.960  (Zigate USB) Connect directive received, action initiated successfully.
2017-11-15 22:03:30.960  (Zigate USB) Calling message handler 'onConnect'.
2017-11-15 22:03:30.960  (Zigate USB) onConnect called
2017-11-15 22:03:30.960  (Zigate USB) Connected successfully to: /dev/ttyUSB15
2017-11-15 22:03:30.960  (Zigate USB) Sending 18 bytes of data:.
2017-11-15 22:03:30.960  (Zigate USB)     01 02 10 21 02 10 02 14 2d 02 10 02 10 02 18 02 10 03 .. ..     ...!....-.........
2017-11-15 22:03:30.960  (Zigate USB) Sending 12 bytes of data:.
2017-11-15 22:03:30.960  (Zigate USB)     01 02 10 23 02 10 02 11 22 02 10 03 .. .. .. .. .. .. .. ..     ...#...."...
2017-11-15 22:03:30.960  (Zigate USB) Sending 10 bytes of data:.
2017-11-15 22:03:30.960  (Zigate USB)     01 02 10 24 02 10 02 10 24 03 .. .. .. .. .. .. .. .. .. ..     ...$....$.
2017-11-15 22:03:31.036  (Zigate USB) Sending 15 bytes of data:.
2017-11-15 22:03:31.036  (Zigate USB)     01 02 10 49 02 10 02 14 b0 ff fc e1 02 10 03 .. .. .. .. ..     ...I....����...
2017-11-15 22:03:31.162  (Zigate USB) Received 38 bytes of data:.
2017-11-15 22:03:31.162  (Zigate USB)     01 80 02 10 02 10 02 15 a4 02 10 02 10 02 10 21 02 10 03 01     .�......�......!....
2017-11-15 22:03:31.162  (Zigate USB)     80 02 10 02 10 02 15 a6 02 10 02 10 02 10 23 02 10 03 .. ..     �......�......#...
2017-11-15 22:03:31.162  (Zigate USB) Calling message handler 'onMessage'.
2017-11-15 22:03:31.162  Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 22:03:31.162  (Zigate USB) Received 19 bytes of data:.
2017-11-15 22:03:31.162  (Zigate USB)     01 80 02 10 02 10 02 15 a1 02 10 02 10 02 10 24 02 10 03 ..     .�......�......$...
2017-11-15 22:03:31.162  (Zigate USB) Calling message handler 'onMessage'.
2017-11-15 22:03:31.162  Error: (Zigate USB) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-15 22:03:33.719  Incoming connection from: 192.168.0.247
```

zaraki673, 2017-11-16:
il faut mettre a jour en béat malheureusement, la stable inclus le moteur de plugin python mais des grosses modifications y ont été faites depuis.

il faut au minimum la version 3.8179 (surement quelques unes en dessous aussi, mais j'ai pas pu test )

Pour l'interrupteur, il devrait fonctionner (même trame que les détecteurs d'ouvertures/fermetures qui sont supportés).

Cordialement,

zaraki673, 2017-11-21:
Problème lié a Domoticz et non au plugin

closed 2017-11-21 by zaraki673

Th1h4ck, 2017-11-22:
Bonjour,

Désolé de cette réponse tardive, j'ai pas eu le temps de m'en occuper avant...
Après mise à jour en 3.8742, ca fonctionne nickel :)

Merci de ton aide