zigbeefordomoticz/Domoticz-Zigbee #6: Impossibilité de voir le matériel ZiGate dans le menu déroulant.

ebretteville, 2017-11-17:
Bonjour,

Le plug-in est bien transféré dans un répertoire dans le dossier plugin. Impossibilité depuis l'update en beta Version: 3.8740.

Je la voyais dans la dernière stable. Mais j'avais les erreurs de l'issue 
'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.

zaraki673, 2017-11-17:
je viens de Maj ca fonctionne chez moi :(
t as essayer de reboot ? au cas ou ... 
(par contre je suis sur la branche de dev, mais y a pas de raison ...)

ebretteville, 2017-11-17:
Ben comme tjrs, ds le doute, reboot! ;-) Same problem.

Entre beta et dev, comment on tests?!

zaraki673, 2017-11-17:
tu change la branche ds github et retelecharge le plugin.py :D

zaraki673, 2017-11-17:
t as quoi ds les logs? tu dois avoir un probleme de chargement de la
librairie python

2017-11-17 20:10 GMT+01:00 Kevin Larsonneur <azazel673@gmail.com>:

zaraki673, 2017-11-17:
y a que lui que tu ne vois pas en plugins python ?

ebretteville, 2017-11-17:
2017-11-17 20:18:08.096 Domoticz V3.8740 (c)2012-2017 GizMoCuz
2017-11-17 20:18:08.096 Build Hash: a4a23dda, Date: 2017-11-16 08:35:10
2017-11-17 20:18:08.097 Startup Path: /home/pi/domoticz/
2017-11-17 20:18:08.345 Sunrise: 08:06:00 SunSet:17:18:00
2017-11-17 20:18:08.345 EventSystem: reset all events...
2017-11-17 20:18:08.404 PluginSystem: 'nono' Registration ignored, Plugins are not enabled.
2017-11-17 20:18:08.419 Active notification Subsystems: gcm, pushbullet (2/14)
2017-11-17 20:18:08.515 WebServer(HTTP) started on address: :: with port 8080
2017-11-17 20:18:08.550 WebServer(SSL) started on address: :: with port 443
2017-11-17 20:18:08.556 Proxymanager started.
2017-11-17 20:18:08.560 Camera: settings (re)loaded
2017-11-17 20:18:08.604 Starting shared server on: :::6144
2017-11-17 20:18:08.604 TCPServer: shared server started...
2017-11-17 20:18:08.605 RxQueue: queue worker started...
2017-11-17 20:18:10.606 Hardware Monitor: Started
2017-11-17 20:18:10.678 (Téléinfo USB) Teleinfo device uses serial port: /dev/ttyUSB0 at 1200 bauds
2017-11-17 20:18:10.683 (Téléinfo USB) CRC checks will be performed on incoming data
2017-11-17 20:18:10.684 Wunderground: Worker started...
2017-11-17 20:18:10.684 YeeLight Worker started...
2017-11-17 20:18:10.885 RFLink: Using serial port: /dev/ttyACM1
2017-11-17 20:18:10.889 (Téléinfo USB) Teleinfo CRC check mode set to 1
2017-11-17 20:18:11.184 OpenZWave: using config in: /home/pi/domoticz/Config/
2017-11-17 20:18:11.219 OpenZWave: Starting...
2017-11-17 20:18:11.219 OpenZWave: Version: 1.4-2713-ge2021e46-dirty
2017-11-17 20:18:11.454 OpenZWave: Driver Ready
2017-11-17 20:18:11.459 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: SENSOR MULTILEVEL, Label: Temperature, Instance: 1
2017-11-17 20:18:11.460 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: THERMOSTAT SETPOINT, Label: Heating 1, Instance: 1
2017-11-17 20:18:11.461 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Monday, Instance: 1
2017-11-17 20:18:11.461 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Tuesday, Instance: 1
2017-11-17 20:18:11.462 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Wednesday, Instance: 1
2017-11-17 20:18:11.462 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Thursday, Instance: 1
2017-11-17 20:18:11.462 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Friday, Instance: 1
2017-11-17 20:18:11.462 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Saturday, Instance: 1
2017-11-17 20:18:11.463 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Sunday, Instance: 1
2017-11-17 20:18:11.463 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Override State, Instance: 1
2017-11-17 20:18:11.463 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Override Setback, Instance: 1
2017-11-17 20:18:11.463 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: BATTERY, Label: Battery Level, Instance: 1
2017-11-17 20:18:11.463 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLOCK, Label: Day, Instance: 1
2017-11-17 20:18:11.464 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLOCK, Label: Hour, Instance: 1
2017-11-17 20:18:11.464 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLOCK, Label: Minute, Instance: 1
2017-11-17 20:18:11.464 OpenZWave: NodeID: 2 (0x02), Thermostat Clock: Monday 08:51
2017-11-17 20:18:11.465 (Z-Wave) Thermostat ([POPP] Cuisine - Consigne)
2017-11-17 20:18:11.467 OpenZWave: Value_Added: Node: 4 (0x04), CommandClass: SWITCH BINARY, Label: Switch, Instance: 1
2017-11-17 20:18:11.467 OpenZWave: Value_Added: Node: 4 (0x04), CommandClass: SWITCH BINARY, Label: Switch, Instance: 2
2017-11-17 20:18:11.468 (Z-Wave) General/Thermostat Clock ([POPP] Cuisine - Horloge)
2017-11-17 20:18:11.471 OpenZWave: Value_Added: Node: 5 (0x05), CommandClass: SWITCH BINARY, Label: Switch, Instance: 1
2017-11-17 20:18:11.472 OpenZWave: Value_Added: Node: 5 (0x05), CommandClass: SWITCH BINARY, Label: Switch, Instance: 2
2017-11-17 20:18:11.472 OpenZWave: Value_Added: Node: 5 (0x05), CommandClass: SWITCH BINARY, Label: Switch, Instance: 3
2017-11-17 20:18:11.475 OpenZWave: Value_Added: Node: 6 (0x06), CommandClass: SWITCH BINARY, Label: Switch, Instance: 1
2017-11-17 20:18:11.476 OpenZWave: Value_Added: Node: 6 (0x06), CommandClass: SWITCH BINARY, Label: Switch, Instance: 2
2017-11-17 20:18:11.812 RFLink: Controller Initialized!...
2017-11-17 20:18:11.967 RFLink Detected, Version: 1.1 Revision: 48 Build: 4
2017-11-17 20:18:12.525 (Téléinfo USB) P1 Smart Meter (Téléinfo USB)
2017-11-17 20:18:12.527 (Téléinfo USB) Current (Teleinfo Courant)
2017-11-17 20:18:12.528 (Téléinfo USB) General/Percentage (Teleinfo Pourcentage de Charge)
2017-11-17 20:18:12.551 (Téléinfo USB) General/Alert (Teleinfo Tarif en cours)
2017-11-17 20:18:12.555 (Téléinfo USB) General/Alert (Teleinfo Alerte courant)
2017-11-17 20:18:25.742 OpenZWave: Received timeout notification from HomeID: 3744604117, NodeID: 4 (0x04)
2017-11-17 20:18:29.147 (WeatherUnderground) Temp + Humidity + Baro ([WU] Temp + Hum)
2017-11-17 20:18:29.150 (WeatherUnderground) Wind ([WU] Vent)
2017-11-17 20:18:29.152 (WeatherUnderground) UV ([WU] UV)
[... cut ...]
2017-11-17 20:18:38.245 (RFLink) Temp + Humidity (Etage)
2017-11-17 20:18:40.685 OpenZWave: Received timeout notification from HomeID: 3744604117, NodeID: 4 (0x04)
2017-11-17 20:18:45.680 (RFLink) Temp + Humidity (Escalier Cave)
2017-11-17 20:19:13.243 (RFLink) Temp + Humidity (Etage)
2017-11-17 20:19:20.695 (RFLink) Temp + Humidity (Escalier Cave)
2017-11-17 20:19:48.245 (RFLink) Temp + Humidity (Etage)
2017-11-17 20:19:55.681 (RFLink) Temp + Humidity (Escalier Cave)
2017-11-17 20:19:57.752 Incoming connection from: 192.168.0.254

zaraki673, 2017-11-17:
t as bien un problème avec le chargement de python
tu as : 
   PluginSystem: 'nono' Registration ignored, Plugins are not enabled.
et tu devrais avoir : 
   PluginSystem: Started, Python version '3.5.3'.

ebretteville, 2017-11-17:
Ouaip mais là je vois pas comment le régler, je vais tenter de revenir en version release sur domoticz. J'ai vu que domoticz aurait désactivé les plugins python ds les dernières beta.

ebretteville, 2017-11-17:
2017-11-17 22:40:20.591 RxQueue: queue worker started...
2017-11-17 22:40:22.592 Hardware Monitor: Started
2017-11-17 22:40:22.662 (Téléinfo USB) Teleinfo device uses serial port: /dev/ttyUSB0 at 1200 bauds
2017-11-17 22:40:22.667 (Téléinfo USB) CRC checks will be performed on incoming data
2017-11-17 22:40:22.667 Wunderground: Worker started...
2017-11-17 22:40:22.668 YeeLight Worker started...
2017-11-17 22:40:22.669 EventSystem: reset all events...
2017-11-17 22:40:22.671 EventSystem: reset all device statuses...
2017-11-17 22:40:22.869 RFLink: Using serial port: /dev/ttyACM1
2017-11-17 22:40:22.960 (Téléinfo USB) Teleinfo CRC check mode set to 1
2017-11-17 22:40:23.054 Python EventSystem: Module not found - Trying to initialize.
2017-11-17 22:40:23.062 Python EventSystem: Initalizing event module.
2017-11-17 22:40:23.062 EventSystem: Started
2017-11-17 22:40:23.078 PluginSystem: Entering work loop.
2017-11-17 22:40:23.168 OpenZWave: using config in: /home/pi/domoticz/Config/
2017-11-17 22:40:23.567 OpenZWave: Starting...
2017-11-17 22:40:23.567 OpenZWave: Version: 1.4-2501-g5af6780-dirty
2017-11-17 22:40:23.794 RFLink: 20;00;Nodo RadioFrequencyLink - RFLink Gateway V1.1 - R48;
2017-11-17 22:40:23.794 RFLink: Controller Initialized!...
2017-11-17 22:40:23.869 RFLink: 20;01;MySensors=ON;
2017-11-17 22:40:23.954 RFLink: 20;02;VER=1.1;REV=48;BUILD=04;
2017-11-17 22:40:23.954 RFLink Detected, Version: 1.1 Revision: 48 Build: 4
2017-11-17 22:40:24.098 OpenZWave: Driver Ready
2017-11-17 22:40:24.101 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: SENSOR MULTILEVEL, Label: Temperature, Instance: 1
2017-11-17 22:40:24.101 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: THERMOSTAT SETPOINT, Label: Heating 1, Instance: 1
2017-11-17 22:40:24.103 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Monday, Instance: 1
2017-11-17 22:40:24.103 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Tuesday, Instance: 1
2017-11-17 22:40:24.103 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Wednesday, Instance: 1
2017-11-17 22:40:24.103 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Thursday, Instance: 1
2017-11-17 22:40:24.104 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Friday, Instance: 1
2017-11-17 22:40:24.104 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Saturday, Instance: 1
2017-11-17 22:40:24.104 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Sunday, Instance: 1
2017-11-17 22:40:24.104 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Override State, Instance: 1
2017-11-17 22:40:24.104 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLIMATE CONTROL SCHEDULE, Label: Override Setback, Instance: 1
2017-11-17 22:40:24.104 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: BATTERY, Label: Battery Level, Instance: 1
2017-11-17 22:40:24.105 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLOCK, Label: Day, Instance: 1
2017-11-17 22:40:24.105 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLOCK, Label: Hour, Instance: 1
2017-11-17 22:40:24.105 OpenZWave: Value_Added: Node: 2 (0x02), CommandClass: CLOCK, Label: Minute, Instance: 1
2017-11-17 22:40:24.105 OpenZWave: NodeID: 2 (0x02), Thermostat Clock: Monday 10:11
2017-11-17 22:40:24.121 OpenZWave: Value_Added: Node: 4 (0x04), CommandClass: SWITCH BINARY, Label: Switch, Instance: 1
2017-11-17 22:40:24.121 OpenZWave: Value_Added: Node: 4 (0x04), CommandClass: SWITCH BINARY, Label: Switch, Instance: 2
2017-11-17 22:40:24.131 OpenZWave: Value_Added: Node: 5 (0x05), CommandClass: SWITCH BINARY, Label: Switch, Instance: 1
2017-11-17 22:40:24.132 OpenZWave: Value_Added: Node: 5 (0x05), CommandClass: SWITCH BINARY, Label: Switch, Instance: 2
2017-11-17 22:40:24.133 OpenZWave: Value_Added: Node: 5 (0x05), CommandClass: SWITCH BINARY, Label: Switch, Instance: 3
2017-11-17 22:40:24.137 OpenZWave: Value_Added: Node: 6 (0x06), CommandClass: SWITCH BINARY, Label: Switch, Instance: 1
2017-11-17 22:40:24.138 OpenZWave: Value_Added: Node: 6 (0x06), CommandClass: SWITCH BINARY, Label: Switch, Instance: 2
2017-11-17 22:40:24.484 (Z-Wave) Thermostat ([POPP] Cuisine - Consigne)
2017-11-17 22:40:24.625 (nono) Initialized version 0.0.5, author 'mrin'
2017-11-17 22:40:24.667 (Z-Wave) General/Thermostat Clock ([POPP] Cuisine - Horloge)
2017-11-17 22:40:24.907 (ZiGate) Initialized version 1.0.2, author 'zaraki673'
2017-11-17 22:40:24.964 (nono) Debug log level set to: 'true'.
2017-11-17 22:40:24.964 (nono) 'HardwareID':'17'
2017-11-17 22:40:24.965 (nono) 'HomeFolder':'/home/pi/domoticz/plugins/xiaomi-mirobot/'
2017-11-17 22:40:24.965 (nono) 'Version':'0.0.5'
2017-11-17 22:40:24.965 (nono) 'Author':'mrin'
2017-11-17 22:40:24.965 (nono) 'Name':'nono'
2017-11-17 22:40:24.965 (nono) 'Address':'192.168.0.85'
2017-11-17 22:40:24.965 (nono) 'Port':'0'
2017-11-17 22:40:24.965 (nono) 'Key':'xiaomi-mi-robot-vacuum'
2017-11-17 22:40:24.965 (nono) 'Mode1':'5564653571416f317052775743326134'
2017-11-17 22:40:24.965 (nono) 'Mode2':'15'
2017-11-17 22:40:24.965 (nono) 'Mode3':'python3'
2017-11-17 22:40:24.965 (nono) 'Mode4':'Debug'
2017-11-17 22:40:24.965 (nono) 'Mode5':'selector'
2017-11-17 22:40:24.965 (nono) Device count: 4
[... cut ...]
2017-11-17 22:41:01.943 LUA: --- --- --- date et heure :2017-11-17 17:00:00 --- --- ---
2017-11-17 22:41:01.943 LUA: --- --- --- time zone :+01:00 --- --- ---
2017-11-17 22:41:01.943 LUA: --- --- --- timestamp :1510938000 --- --- ---
2017-11-17 22:41:01.943 LUA: --- --- --- hygrometrie : 87 --- --- ---
2017-11-17 22:41:01.943 LUA: --- --- --- température :9.15 --- --- ---
2017-11-17 22:41:01.943 LUA: ========= Fin Qualité de l'air (v1.00) =========
2017-11-17 22:41:02.032 EventSystem: Script event triggered: Affichage pollution AQICN

ebretteville, 2017-11-17:
2017-11-17 22:49:37.591 (ZiGate) onStart called
2017-11-17 22:49:37.591 (ZiGate) Debug log level set to: 'true'.
2017-11-17 22:49:37.593 (nono) Heartbeat interval set to: 15.
2017-11-17 22:49:37.593 (ZiGate) Protocol set to: 'None'.
2017-11-17 22:49:37.593 (ZiGate) Transport set to: 'Serial', '/dev/ttyUSB1', 115200.
2017-11-17 22:49:37.597 (ZiGate) Connect directive received, action initiated successfully.
2017-11-17 22:49:37.597 (ZiGate) Calling message handler 'onConnect'.
2017-11-17 22:49:37.597 (ZiGate) onConnect called
2017-11-17 22:49:37.597 (ZiGate) Connected successfully to: /dev/ttyUSB1
2017-11-17 22:49:37.597 (ZiGate) Sending 18 bytes of data:.
2017-11-17 22:49:37.597 (ZiGate) 01 02 10 21 02 10 02 14 2d 02 10 02 10 02 18 02 10 03 .. .. ...!....-.........
2017-11-17 22:49:37.598 (ZiGate) Sending 12 bytes of data:.
2017-11-17 22:49:37.598 (ZiGate) 01 02 10 23 02 10 02 11 22 02 10 03 .. .. .. .. .. .. .. .. ...#...."...
2017-11-17 22:49:37.673 (ZiGate) Sending 10 bytes of data:.
2017-11-17 22:49:37.673 (ZiGate) 01 02 10 24 02 10 02 10 24 03 .. .. .. .. .. .. .. .. .. .. ...$....$.
2017-11-17 22:49:37.748 (ZiGate) Sending 15 bytes of data:.
2017-11-17 22:49:37.748 (ZiGate) 01 02 10 49 02 10 02 14 b0 ff fc e1 02 10 03 .. .. .. .. .. ...I....����...
2017-11-17 22:49:37.823 (ZiGate) Received 19 bytes of data:.
2017-11-17 22:49:37.823 (ZiGate) 01 80 02 10 02 10 02 15 a4 02 10 02 10 02 10 21 02 10 03 .. .�......�......!...
2017-11-17 22:49:37.823 (ZiGate) Calling message handler 'onMessage'.
2017-11-17 22:49:37.823 Error: (ZiGate) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-17 22:49:37.824 (ZiGate) Received 19 bytes of data:.
2017-11-17 22:49:37.824 (ZiGate) 01 80 02 10 02 10 02 15 a6 02 10 02 10 02 10 23 02 10 03 .. .�......�......#...
2017-11-17 22:49:37.824 (ZiGate) Calling message handler 'onMessage'.
2017-11-17 22:49:37.824 Error: (ZiGate) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-17 22:49:37.824 (ZiGate) Received 19 bytes of data:.
2017-11-17 22:49:37.824 (ZiGate) 01 80 02 10 02 10 02 15 a1 02 10 02 10 02 10 24 02 10 03 .. .�......�......$...
2017-11-17 22:49:37.824 (ZiGate) Calling message handler 'onMessage'.
2017-11-17 22:49:37.824 Error: (ZiGate) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2017-11-17 22:49:38.693 (Téléinfo USB) P1 Smart Meter (Téléinfo USB)
2017-11-17 22:49:38.833 (Téléinfo USB) Current (Teleinfo Courant)
2017-11-17 22:49:38.973 (Téléinfo USB) General/Percentage (Teleinfo Pourcentage de Charge)
2017-11-17 22:49:39.125 (Téléinfo USB) General/Alert (Teleinfo Tarif en cours)
2017-11-17 22:49:39.269 (Téléinfo USB) General/Alert (Teleinfo Alerte courant)
2017-11-17 22:49:50.948 RFLink: 20;03;Alecto V4;ID=5a9a;TEMP=009b;HUM=63;
2017-11-17 22:49:51.107 OpenZWave: Received timeout notification from HomeID: 3744604117, NodeID: 4 (0x04)
2017-11-17 22:49:51.130 (RFLink) Temp + Humidity (Escalier Cave)
2017-11-17 22:49:55.250 (WeatherUnderground) Temp + Humidity + Baro ([WU] Temp + Hum)
2017-11-17 22:49:55.394 (WeatherUnderground) Wind ([WU] Vent)
2017-11-17 22:49:55.735 (nono) Calling message handler 'onHeartbeat'.
2017-11-17 22:49:55.803 OpenZWave: Received Node Dead notification from HomeID: 3744604117, NodeID: 5 (0x05)
2017-11-17 22:49:56.014 (Z-Wave) Light/Switch (Ballon Eau chaude)
2017-11-17 22:49:56.029 (Z-Wave) Light/Switch (Switch)
2017-11-17 22:49:56.033 (Z-Wave) Light/Switch (Switch)
2017-11-17 22:49:56.087 OpenZWave: Received Node Dead notification from HomeID: 3744604117, NodeID: 6 (0x06)
2017-11-17 22:49:56.251 (Z-Wave) Light/Switch (FGS222 - Chaudière?! 1)
2017-11-17 22:49:56.416 (Z-Wave) Light/Switch (FGS222 - Chaudière?! 2)
2017-11-17 22:49:57.570 Error: (nono) Call mirobo-wrapper exception: Command '['python3', '/home/pi/domoticz/plugins/xiaomi-mirobot/mirobo-wrapper.py', '192.168.0.85', '5564653571416f317052775743326134', 'status']' returned non-zero exit status 1.
2017-11-17 22:49:57.571 (ZiGate) Calling message handler 'onHeartbeat'.
2017-11-17 22:49:58.364 OpenZWave: Awake Nodes queried
2017-11-17 22:49:58.365 OpenZWave: Received Node Dead notification from HomeID: 3744604117, NodeID: 4 (0x04)
2017-11-17 22:49:58.540 (Z-Wave) Light/Switch (Radiateur SdB Chambre 1)
2017-11-17 22:49:58.686 (Z-Wave) Light/Switch (Radiateur SdB Chambre 2)
2017-11-17 22:50:01.602 Incoming connection from: 192.168.0.254
2017-11-17 22:50:05.741 (ZiGate) Calling message handler 'onHeartbeat'.
2017-11-17 22:50:06.092 OpenZWave: Received timeout notification from HomeID: 3744604117, NodeID: 4 (0x04)
2017-11-17 22:50:10.755 (nono) Calling message handler 'onHeartbeat'.
2017-11-17 22:50:11.795 Error: (nono) Call mirobo-wrapper exception: Command '['python3', '/home/pi/domoticz/plugins/xiaomi-mirobot/mirobo-wrapper.py', '192.168.0.85', '5564653571416f317052775743326134', 'status']' returned non-zero exit status 1.
2017-11-17 22:50:15.757 (ZiGate) Calling message handler 'onHeartbeat'.
2017-11-17 22:50:18.365 RFLink: 20;04;Alecto V4;ID=5308;TEMP=00c7;HUM=61;
2017-11-17 22:50:18.527 EventSystem: Event triggered: Chauffage_3
2017-11-17 22:50:18.550 (RFLink) Temp + Humidity (Etage)
2017-11-17 22:50:18.559 RFLink Sending: 10;Kaku;000041;1;ON
2017-11-17 22:50:19.135 RFLink: 20;05;OK;
2017-11-17 22:50:19.283 (RFLink) Light/Switch ([Ch.] Contact-sec)
2017-11-17 22:50:19.794 RFLink: 20;06;Alecto V1;ID=0208;WINGS=0000;WINSP=0566;BAT=LOW;
[... cut ...]
2017-11-17 22:50:53.324 RFLink Sending: 10;Kaku;000041;1;ON
2017-11-17 22:50:53.881 RFLink: 20;09;OK;
2017-11-17 22:50:54.039 (RFLink) Light/Switch ([Ch.] Contact-sec)
2017-11-17 22:50:55.770 (nono) Calling message handler 'onHeartbeat'.
2017-11-17 22:50:56.818 Error: (nono) Call mirobo-wrapper exception: Command '['python3', '/home/pi/domoticz/plugins/xiaomi-mirobot/mirobo-wrapper.py', '192.168.0.85', '5564653571416f317052775743326134', 'status']' returned non-zero exit status 1.
2017-11-17 22:50:56.819 (ZiGate) Calling message handler 'onHeartbeat'.

zaraki673, 2017-11-21:
idem : https://github.com/sasu-drooz/Domoticz-Zigate/issues/5

Problème lié a Domoticz et non au plugin

closed 2017-11-21 by zaraki673