zigbeefordomoticz/Domoticz-Zigbee #7: message error

james013, 2017-11-19:
**Domoticz 3.8701 avec Python 3.5.1 sur Synology en DSM 6.1**

Avec la version dev j'ai réussi à synchroniser 3 capteur Xiomi Aqara (temp./Hum./Pres.)
Mais les info remontant ne passe pas tout le temps, et cette erreur revient souvent  ( erreurs sur les lignes **_70_** et **_107_** )

voila un extrait de mon log :

2017-11-19 20:26:43.146 (zigate) Calling message handler 'onMessage'.
2017-11-19 20:26:43.146 (zigate) onMessage called
2017-11-19 20:26:43.146 (zigate) Calling message handler 'onMessage'.
2017-11-19 20:26:43.146 (zigate) onMessage called
2017-11-19 20:26:43.146 (zigate) Calling message handler 'onMessage'.
2017-11-19 20:26:43.146 (zigate) onMessage called
2017-11-19 20:26:43.147 (zigate) ZigateDecode - decodind data : 018102120210021e1e9fc87202110214021302101402102802100211ff7203018102120210021f1b9fc8720211021402130210100210290210021226db7203
2017-11-19 20:26:43.147 (zigate) ZigateRead - decoded data : 018102000e1e9fc872010403001400280001ff7203018102000f1b9fc87201040300100029000226db7203
2017-11-19 20:26:43.148 (zigate) ZigateRead - Message Type : 8102, Data : 9fc872010403001400280001ff7203018102000f1b9fc87201040300100029000226db, RSSI : 72, Length : 000e, Checksum : 1e
2017-11-19 20:26:43.148 (zigate) ZigateRead - MsgType 8102 - reception data : 018102000e1e9fc872010403001400280001ff7203018102000f1b9fc87201040300100029000226db7203 ClusterID : 0403 Attribut ID : 0014 Src Addr : c872 Scr Ep: 01
**_2017-11-19 20:26:43.149 Error: (zigate) 'onMessage' failed 'UnicodeEncodeError'.
2017-11-19 20:26:43.149 Error: (zigate) ----> Line 107 in /usr/local/domoticz/var/plugins/ZIGATE/plugin.py, function onMessage
2017-11-19 20:26:43.149 Error: (zigate) ----> Line 70 in /usr/local/domoticz/var/plugins/ZIGATE/plugin.py, function onMessage_**
2017-11-19 20:26:43.662 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:26:53.646 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:27:03.695 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:27:13.690 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:27:23.728 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:27:33.750 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:27:43.822 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:27:53.841 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:28:03.878 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:28:13.863 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:28:23.894 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:28:33.938 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:28:43.972 (zigate) Calling message handler 'onHeartbeat'.
2017-11-19 20:28:54.008 (zigate) Calling message handler 'onMessage'.
2017-11-19 20:28:54.008 (zigate) onMessage called
2017-11-19 20:28:54.008 (zigate) Calling message handler 'onMessage'.
2017-11-19 20:28:54.008 (zigate) onMessage called

zaraki673, 2017-11-20:
pour ces deux erreurs (ligne 70 et 107) as tu le debug d'activé ? 
je penses que oui et sur syno domoticz n'arrive pas a écrire dans le fichier de debug, c 'est pas gênant, je vais le supprimer pour passer par les logs de Domoticz (mise à jour prévu cette semaine),

Par contre le fait que infos remontent et ne passent pas tout le temps, qu'est ce qui te fait dire ca ?
je ne décode pas toutes les trames encore, juste celles utilisées pour la temp/pression et hum actuellement, les trames d'état de batterie ou autres ne sont pas encore décodées (une partie sera decodé dans la maj de cette semaine, mais surement pas ttes encore :D )

james013, 2017-11-21:
avec la dernière version dev : 

2017-11-21 09:12:36.319 (zigate) Stopped.
2017-11-21 09:13:15.132 (zigate) Started.
2017-11-21 09:13:15.250 Error: (Zigate) failed to load 'plugin.py', Python Path used was '/usr/local/domoticz/var/plugins/ZIGATE/:/volume2/@appstore/py3k/usr/local/lib/python35.zip:/volume2/@appstore/py3k/usr/local/lib/python3.5/:/volume2/@appstore/py3k/usr/local/lib/python3.5/plat-linux:/volume2/@appstore/py3k/usr/local/lib/python3.5/lib-dynload'.
2017-11-21 09:13:15.251 Error: (zigate) Module Import failed, exception: 'IndentationError'
2017-11-21 09:13:15.251 Error: (zigate) Import detail: File: /usr/local/domoticz/var/plugins/ZIGATE/plugin.py, Line: 275, offset: 5
2017-11-21 09:13:15.251 Error: (zigate) Error Line '	elif str(MsgType)=="8000": # Status

zaraki673, 2017-11-21:
oui , j ai oublier une parenthese ds le code quand je l'ai poussé, désolé

zaraki673, 2017-11-21:
c 'est corrigé dans la version de dev actuel

closed 2017-11-21 by zaraki673