zigbeefordomoticz/Domoticz-Zigbee #8: Error: (zigate) Module Import failed, exception: 'SyntaxError'

olicot, 2017-11-20:
Bonjour, 
depuis la version du 14/11, je rencontre cette erreur, et cela bloque complement domoticz:

2017-11-20 20:41:46.855  Error: (Zigate) failed to load 'plugin.py', Python Path used was '/opt/domoticz/plugins/zigate/:/usr/lib/python35.zip:/usr/lib/python3.5/:/usr/lib/python3.5/plat-arm-linux-gnueabihf:/usr/lib/python3.5/lib-dynload'.
2017-11-20 20:41:46.855  Error: (zigate) Module Import failed, exception: 'SyntaxError'
2017-11-20 20:41:46.856  Error: (zigate) Import detail: File: /opt/domoticz/plugins/zigate/plugin.py, Line: -1240278589, offset: 70

A notez que : 
- les versions antérieures au 14/11 fonctionnent 
- les autres plugin python sont fonctionnel (apres avoir désactivér zigate) exemple: 
2017-11-20 20:41:46.715  (RM2_11) Entering work loop.
2017-11-20 20:41:46.715  (RM2_11) Initialized version 3.0.0, author 'zak45'

- Domoticz version: 3.8742
- ARMBIAN 5.31 stable Ubuntu 16.04.3 LTS 3.4.113-sun8i
- Python 3.5.2

des idées ?
MErci 
Olivier

zaraki673, 2017-11-21:
en cour de correction, je pousse une maj sur la branche master des que possible, désolé

olicot, 2017-11-22:
Cela fait partie du processus de dev et debeugage ;-)

zaraki673, 2017-11-23:
c'est corrigé sur les nouvelles versions, tu peux confirmer ?
merci ;)

olicot, 2017-11-23:
snif... meme soucis, avec la branche principale et la dev, pour les dernieres publications.

olicot, 2017-11-24:
j'ai trouvé :-)
c'est dû aux caractères "ç" (ligne 63), et "à" ligne 70, pourtant en commentaire, qui m'embetaient...
donc un souci pas vraiment du au code.
on peut clore cet issue

zaraki673, 2017-11-24:
:( je vais retirer les caracteres speciaux , c est pas plus mal

merci de la recherche , j y aurais pas pensé :(

;)

zaraki673, 2017-11-24:
c est corrigé en dev, bientot publié j esperes :D

zaraki673, 2017-11-25:
publié :p

closed 2017-11-28 by zaraki673