zigbeefordomoticz/Domoticz-Zigbee #14: effacement de la trame suite à une erreur de decodage

deennoo, 2017-11-23:
Domoticz Beta : 3.8742
Plugin Version 1.0.3
Motion Sensor Aqara avec capteur de luminosité

J'essaye de t'aider dans le plugin en ajoutant ce que j'ai comme materiel.

J'ai ajouter le capteur de presence : lumi.sensor_motion.aq2
La fonction detection : ok

Pour le Luxometre j'ai un soucis : 

`2017-11-23 21:02:07.702 (Zigate USB) ZigateDecode - decodind data : 018102120210021f629ae47a02110214021002100210021021021002120210021fc303
2017-11-23 21:02:07.703 (Zigate USB) ZigateRead - decoded data : 018102000f629ae47a010400000000210002000fc303
2017-11-23 21:02:07.703 (Zigate USB) ZigateRead - Message Type : 8102, Data : 9ae47a010400000000210002000f, RSSI : c3, Length : 000f, Checksum : 62
2017-11-23 21:02:07.703 (Zigate USB) ZigateRead - MsgType 8102 - reception data : 018102000f629ae47a010400000000210002000fc303 ClusterID : 0400 Attribut ID : 0000 Src Addr : e47a Scr Ep: 01
2017-11-23 21:02:07.703 (Zigate USB) MajDomoDevice - Device ID : e47a Device EP : 01 Type : Lux Value : 000f
2017-11-23 21:02:07.703 (Zigate USB) onMessage - effacement de la trame suite à une erreur de decodage : 018102120210021f629ae47a02110214021002100210021021021002120210021fc303`

zaraki673, 2017-11-23:
probleme ici : MajDomoDevice - Device ID : e47a Device EP : 01 Type : Lux Value : 000f

tu dois presque y etre, surement un problème de type de donnée (string attendu et integer recu ou inverse ... )
envoie ton code je regarde ca ;)

deennoo, 2017-11-23:
ok en fait cette valeur semble etre de l'hexadecimal a convertir en décimal (source calculatrice prog windows)

zaraki673, 2017-11-23:
vi vi avant de l envoyer dans la commande de maj du capteur ;)

deennoo, 2017-11-23:
je n'ai jamais fais cela...

Ici ?

`		elif MsgClusterId=="0400" :  # (Measurement: LUX)
			MajDomoDevice(MsgSrcAddr,MsgSrcEp,"Lux",MsgClusterData)
			
			Domoticz.Debug("ZigateRead - MsgType 8102 - reception LUX Sensor : " + str(MsgClusterData) )`

zaraki673, 2017-11-23:
oui , tu mets ca je penses : 

MajDomoDevice(MsgSrcAddr,MsgSrcEp,"Lux",str(int(MsgClusterData,16) ))

zaraki673, 2017-11-24:
c'est ajouté à la version 1.0.4

Merci  ;)

closed 2017-11-24 by zaraki673

deennoo, 2017-11-24:
not really closed mon grand

Apres avoir mis 

`MajDomoDevice(MsgSrcAddr,MsgSrcEp,"Lux",str(int(MsgClusterData,16) ))`

J'ai toujours : 

`2017-11-24 10:34:26.071 (Zigate USB) MajDomoDevice - Device ID : a7e7 Device EP : 01 Type : Lux Value : 15
2017-11-24 10:34:26.072 (Zigate USB) onMessage - effacement de la trame suite à une erreur de decodage : 018102120210021f424aa7e702110214021002100210021021021002120210021fed03`

De plus le PIR n’apparaît pas dans les switchs