zigbeefordomoticz/Domoticz-Zigbee #25: Error: (Zigate) 'onHeartbeat' failed 'KeyError'

thiklop, 2017-11-27:
Dans les logs l'appel à la fonction onHeartbeat provoque systématiquement une erreur : 
> 2017-11-27 21:56:41.758 (Zigate) Calling message handler 'onHeartbeat'.
> 2017-11-27 21:56:41.759 Error: (Zigate) 'onHeartbeat' failed 'KeyError'.
> 2017-11-27 21:56:41.759 Error: (Zigate) ----> Line 119 in /home/olivier/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
> 2017-11-27 21:56:41.759 Error: (Zigate) ----> Line 84 in /home/olivier/domoticz/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
> 2017-11-27 21:56:41.759 Error: (Zigate) ----> Line 772 in /home/olivier/domoticz/plugins/Domoticz-Zigate/plugin.py, function ResetDevice 

Noter que cela ne gène pas la réception et le décodage des trames

zaraki673, 2017-11-27:
merci, je regarde ça ;)

thiklop, 2017-11-27:
Trop rapide !

Je précise que ça marchait très bien en 1.0.2.

thiklop, 2017-11-28:
Contrairement à ce que j'ai écrit dans la description le bug bloque la réception des données, je n'ai plus rien depuis hier soir.

zaraki673, 2017-11-28:
:( ok je vais regarder ca asap

zaraki673, 2017-11-28:
hum tu es en quelle version de domoticz ?

zaraki673, 2017-11-28:
suite a la mise à jour a tu effacer les anciens devices ? 
(il faut les redeclarer malheureusement, j ai du ajouter des "tags" dans les optiions des devices domoticz)

thiklop, 2017-11-28:
Alors dans l'ordre, pas d'urgence j'ai rebasculé en 1.0.3 et ça refonctionne.
Je suis en version 3.8742 (beta)
Et pour finir j'ai redeclaré tous mes devices lors du passage en 1.0.3. Ils sont apparus tout seuls après passage dans cette version mais n'étaient pas remontés en 1.0.5.

Si tu as d'autre questions n'hésite pas et merci pour ton boulot.

thiklop, 2017-12-01:
Bon j'ai fini par résoudre le problème et j'ai pu basculer en 1.0.5. En fait il est assez simple, il faut **absolument** supprimer tous ses devices puis redémarrer Domoticz pour que l'erreur n'apparaisse plus.
En résumé pas de problème dans le plugin mais plutôt dans l'installation de celui-ci.

zaraki673, 2017-12-01:
ok, pas besoin avec la nouvelle version, y a pas eu de modification de ce coté la ;)

closed 2017-12-01 by zaraki673