zigbeefordomoticz/Domoticz-Zigbee #29: Prise Salus

zaraki673, 2017-12-05:
Salut,
 
Je bosse sur une prise de marque Salus
 
 
Depuis le zwgui j’ai :
 
Simple req
22:41:17.861 -> 01 02 10 43 02 10 02 13 45 56 5A 02 19 03
22:41:17.884 <- 01 80 00 00 05 59 00 9F 00 43 00 03
22:41:17.913 <- 01 80 43 00 32 BD 9F 00 56 5A 1C 09 01 04 00 51 00 09 00 00 00 01 00 03 00 04 00 05 00 06 04 02 07 02 FC 01 01 00 00 00 01 00 03 00 04 00 05 00 06 04 02 07 02 FC 01 96 03
 
  Length: 5
  Status: 0x00 (Success)
  SQN: 0x9F
  Message:
Type: 0x8043 (Simple Descriptor Response)
  SQN: 0x9F
  Status: 0x00
  Short Address: 0x565A
  Length: 28
  EndPoint: 0x09
  Profile ID: 0x0104 (ZigBee HA)
  Device ID: 0x0051 (Generic - Smart Plug)
  Input Cluster Count: 9
    Cluster 0:  Cluster ID: 0x0000 (General: Basic)
    Cluster 1:  Cluster ID: 0x0001 (General: Power Config)
    Cluster 2:  Cluster ID: 0x0003 (General: Identify)
    Cluster 3:  Cluster ID: 0x0004 (General: Groups)
    Cluster 4:  Cluster ID: 0x0005 (General: Scenes)
    Cluster 5:  Cluster ID: 0x0006 (General: On/Off)
    Cluster 6:  Cluster ID: 0x0402 (Measurement: Temperature)
    Cluster 7:  Cluster ID: 0x0702 (Smart Energy: Metering)
    Cluster 8:  Cluster ID: 0xFC01 (Unknown)
  Output Cluster Count: 1
    Cluster 0:  Cluster ID: 0x0000 (General: Basic)
 
On retrouve bien les cluster dans la ligne retour
 
Depuis Domoticz aussi, je ne vais pas tous te les marquer.
 
2017-12-05 09:00:54.826 (Zigate USB) ZigateRead - Message Type : 8043, Data : 4d00 565a 1c09 0104 0051 0009 0000 0001 0003 0004 0005 0006 0402 0702 fc01 01 0000 0001 0003 0004 0005 0006 0402 0702 fc01, RSSI : a8, Length : 0032, Checksum : 51
 
Les cluster ID sont toujours les memes quelque soit la marque du modules.
 
D’apres mes lecture le device ID :  0051 : c’est toujours une prise (la xiaomi utilise le meme device ID), on pourrait déjà ce servir de cela pour créer le device dans Domoticz qu’en penses tu ?

deennoo, 2017-12-22:
apres enquete, ce type de prise n'envoie jamais son changement d'état si elle est manipulé depuis le bouton physique;

C'est a l'application/plugin d'aller verifier si il y a eu du changement
Idem pour la conso electrique

Depuis zwgui la commande on/off/toggle fonctionne : 

Ici Toggle : 

`09:29:52.978 -> 01 02 10 92 02 10 02 16 58 02 12 02 17 C3 02 11 02 19 02 12 03 
09:29:53.011 <- 01 80 00 00 05 17 00 00 00 92 00 03
09:29:53.040 <- 01 81 01 00 07 0D 00 09 00 06 02 00 87 03`

zaraki673, 2017-12-22:
merci ;)

va avoir plein de taf le heartbeat :P

j ai même pas eu le temps de regarder les ikea :(

j'esperes sortir la nouvelle version avant la fin d'année, elle permettra
de jouer plus facilement (on pourra ajouter manuellement des devices si on
connait la short adress de celui-ci :D )

zaraki673, 2018-01-25:
ca devrait etre bon , en attente du retour

zaraki673, 2018-01-27:
pour le retour d'info conso voir #58

closed 2018-01-27 by zaraki673