zigbeefordomoticz/Domoticz-Zigbee #30: Association OK LOG ok mais pas de nouveau dispositif

seblang, 2017-12-09:
J'ai des périphériques qui communique (d’après les logs) mais aucun nouveau dispositif n'est apparu!
comment résoudre ce problème?
merci

zaraki673, 2017-12-09:
essaie d appuyer sur le bouton, de maniere a ce que le device se represente à nouveau (pas un appuie long, un simple appuie suffit pour la pluspart)

c'est quoi comme périphériques que tu essaies d'ajouter ?

seblang, 2017-12-09:
rien n'y fait.
j'ai mis en détection pour ajout nouveau périphérique, relancer le plugin,
associe lorsque la led clignote.......mais je ne vois tjr pas de
périphérique
en revanche les logs montrent que çà communique

​
il faut peut être que je fasse un RAZ de la clef ou je sais pas...

zaraki673, 2017-12-10:
peut tu m envoyer tes logs 
du demarrage du plugin, jusqu a ce que tu appuies 5 secondes sur le capteur et les trames qui vont suivre

merci

zaraki673, 2018-01-03:
pas de nouvelles, je clos ;)

closed 2018-01-03 by zaraki673