zigbeefordomoticz/Domoticz-Zigbee #31: les températures inférieures à zéro ne sont pas correctes

domowarrior, 2017-12-10:
Bon soir ,
je mexuse mon pauvre francais.
jeinstallez zigate dans mon domoticz ( dans un PI) (Version: 3.8775 Build Hash: 261e2df6)
installez le plug in de python, en voilaa ma Xiaomi temperature a registrez et function bien.

Mais je installez une sensor dans ma balcon , cette sensor a une aqara avec temp, humidité et pression.

le temperature Hunidite et pression sons correct , mais quand le temperature et desous de 0 grade Celsius il na pas pa correct , il y a des valeurs supérieures à 600 grade celsius.

Cette censor regitrez dans le Xiaomi hub , functioneze bien aussi et desous de 0 grade .

peut etre une petit bug ?

je attachez  de log de ma domotics Je mets en dessous du journal à partir du moment où le tempéré va de plus Celsius à moins

2017-12-10 22:53:29.073 (Zigate) onMessage called
2017-12-10 22:53:29.073 (Zigate) ZigateDecode - decodind data : 018102120210021e708710dc02110214021302101402102802100211ff7203
2017-12-10 22:53:29.073 (Zigate) ZigateRead - decoded data : 018102000e708710dc010403001400280001ff7203
2017-12-10 22:53:29.073 (Zigate) ZigateRead - Message Type : 8102, Data : 8710dc010403001400280001ff, RSSI : 72, Length : 000e, Checksum : 70
2017-12-10 22:53:29.074 (Zigate) ZigateRead - MsgType 8102 - reception data : 018102000e708710dc010403001400280001ff7203 ClusterID : 0403 Attribut ID : 0014 Src Addr : 10dc Scr Ep: 01
2017-12-10 22:53:29.074 (Zigate) onMessage - effacement de la trame suite a une erreur de decodage : 018102120210021e708710dc02110214021302101402102802100211ff7203
2017-12-10 22:53:29.074 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.074 (Zigate) onMessage called
2017-12-10 22:53:29.074 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.074 (Zigate) onMessage called
2017-12-10 22:53:29.074 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.075 (Zigate) onMessage called
2017-12-10 22:53:29.075 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.075 (Zigate) onMessage called
2017-12-10 22:53:29.075 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.075 (Zigate) onMessage called
2017-12-10 22:53:29.075 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.075 (Zigate) onMessage called
2017-12-10 22:53:29.075 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.075 (Zigate) onMessage called
2017-12-10 22:53:29.075 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.076 (Zigate) onMessage called
2017-12-10 22:53:29.076 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.076 (Zigate) onMessage called
2017-12-10 22:53:29.076 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.076 (Zigate) onMessage called
2017-12-10 22:53:29.076 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.076 (Zigate) onMessage called
2017-12-10 22:53:29.076 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.076 (Zigate) onMessage called
2017-12-10 22:53:29.077 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.077 (Zigate) onMessage called
2017-12-10 22:53:29.077 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.077 (Zigate) onMessage called
2017-12-10 22:53:29.077 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.077 (Zigate) onMessage called
2017-12-10 22:53:29.077 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.077 (Zigate) onMessage called
2017-12-10 22:53:29.077 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.077 (Zigate) onMessage called
2017-12-10 22:53:29.078 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.078 (Zigate) onMessage called
2017-12-10 22:53:29.078 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.078 (Zigate) onMessage called
2017-12-10 22:53:29.078 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.078 (Zigate) onMessage called
2017-12-10 22:53:29.078 (Zigate) Calling message handler 'onMessage'.
2017-12-10 22:53:29.078 (Zigate) onMessage called
2017-12-10 22:53:29.078 (Zigate) ZigateDecode - decodind data : 018102120210021fbf8710dc0211021402130210100210290210021226117203
2017-12-10 22:53:29.079 (Zigate) ZigateRead - decoded data : 018102000fbf8710dc01040300100029000226117203
2017-12-10 22:53:29.079 (Zigate) ZigateRead - Message Type : 8102, Data : 8710dc0104030010002900022611, RSSI : 72, Length : 000f, Checksum : bf
2017-12-10 22:53:29.079 (Zigate) ZigateRead - MsgType 8102 - reception data : 018102000fbf8710dc01040300100029000226117203 ClusterID : 0403 Attribut ID : 0010 Src Addr : 10dc Scr Ep: 01
2017-12-10 22:53:29.080 (Zigate) MajDomoDevice - Device ID : 10dc - Device EP : 01 - Type : Barometer - Value : 974.5
2017-12-10 22:53:29.080 (Zigate) MajDomoDevice baro CurrentsValue : 8.9;38.2;1;974;974.4
2017-12-10 22:53:29.080 (Zigate) MajDomoDevice bar NewSvalue : 8.9;38.2;1;974.5;974
[... 2202 lines cut ...]
2017-12-10 23:00:59.742 (Zigate) Calling message handler 'onMessage'.
2017-12-10 23:00:59.742 (Zigate) onMessage called
2017-12-10 23:00:59.742 (Zigate) Calling message handler 'onMessage'.
2017-12-10 23:00:59.742 (Zigate) onMessage called
2017-12-10 23:00:59.742 (Zigate) Calling message handler 'onMessage'.
2017-12-10 23:00:59.742 (Zigate) onMessage called
2017-12-10 23:00:59.742 (Zigate) Calling message handler 'onMessage'.
2017-12-10 23:00:59.742 (Zigate) onMessage called
2017-12-10 23:00:59.743 (Zigate) Calling message handler 'onMessage'.
2017-12-10 23:00:59.743 (Zigate) onMessage called
2017-12-10 23:00:59.743 (Zigate) ZigateDecode - decodind data : 018102120210021ff4cc10dc0211021402130210100210290210021226167503
2017-12-10 23:00:59.743 (Zigate) ZigateRead - decoded data : 018102000ff4cc10dc01040300100029000226167503
2017-12-10 23:00:59.743 (Zigate) ZigateRead - Message Type : 8102, Data : cc10dc0104030010002900022616, RSSI : 75, Length : 000f, Checksum : f4
2017-12-10 23:00:59.744 (Zigate) ZigateRead - MsgType 8102 - reception data : 018102000ff4cc10dc01040300100029000226167503 ClusterID : 0403 Attribut ID : 0010 Src Addr : 10dc Scr Ep: 01
2017-12-10 23:00:59.744 (Zigate) MajDomoDevice - Device ID : 10dc - Device EP : 01 - Type : Barometer - Value : 975.0
2017-12-10 23:00:59.744 (Zigate) MajDomoDevice baro CurrentsValue : 648.5;53.7;1;975;974.9
2017-12-10 23:00:59.744 (Zigate) MajDomoDevice bar NewSvalue : 648.5;53.7;1;975.0;975
2017-12-10 23:00:59.744 (Zigate - outside) Updating device from 0:'648.5;53.7;1;975;974.9' to have values 0:'648.5;53.7;1;975.0;975'.
2017-12-10 23:00:59.753 (Zigate) Update 0:'648.5;53.7;1;975.0;975' (Zigate - outside)
2017-12-10 23:00:59.753 (Zigate) MajDomoDevice baro CurrentsValue : 975;974.9
2017-12-10 23:00:59.753 (Zigate - Barometer - 10dc) Updating device from 0:'975;974.9' to have values 0:'975.0;975'.
2017-12-10 23:00:59.755 (Zigate) Update 0:'975.0;975' (Zigate - Barometer - 10dc)
2017-12-10 23:00:59.756 (Zigate) ZigateRead - MsgType 8102 - reception atm : 975.0
2017-12-10 23:01:00.407 (Zigate) Calling message handler 'onHeartbeat'.
2017-12-10 23:01:10.394 (Zigate) Calling message handler 'onHeartbeat'.

zaraki673, 2017-12-13:
Hi,
I will fix this soon,

sorry

zaraki673, 2017-12-27:
Hope this fix is good :(

tell me if not ;)

zaraki673, 2018-12-14:
yes ;) (on two different sensor model)

zaraki673, 2018-12-15:
ReadCluster - 0000/ff01 Saddr: 52ad Temperature : 651.3 :p

[TEMP-LOG.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/2682071/TEMP-LOG.txt)

pipiche38, 2018-12-15:
Thanks ! 

Frame: `018102002ccb6952ad010000ff010042001f0121630b0421a813052187030624010000000064296efe6521fc1f0a2100005403`

Decoding:
```
MsgType: 8102, MsgLenght: 002c, Crc: cb, Data: 6952ad010000ff010042001f0121630b0421a813052187030624010000000064296efe6521fc1f0a210000 , RSSI: 54
SQN: 69, sAddr: 52ad EP: 01 ClusterID: 0000 AttrID: ff01 Status: 00 Type: 42 Size: 001f Value: 0121630b0421a813052187030624010000000064296efe6521fc1f0a210000
```

Then we identify the tag 0x64 with DataType 0x29: `6429 6efe`
Temperature value is : `6efe`

6efe is  as 0x29 ( # 16Bitint )

pipiche38, 2019-01-06:
Fixed on 4.0.5 version

commit 343db722 (2017-12-27): v1.0.8

fix #39
fix #31
add water sensor #37
PR #42 (2017-12-27): version 1.0.8
fix #39
fix #31
add water sensor #37
commit da20de5f (2018-12-15): Should fix #31 and correctly managed negative temp
closed 2019-01-06 by pipiche38