zigbeefordomoticz/Domoticz-Zigbee #33: Nombre max de devices

tornlouses, 2017-12-14:
Bonjour Zaraki673,

Encore merci pour ton boulot.

Je voulais savoir si il y a une limite de device connectés via la Zigate?

En effet j'ai lu sur le site du créateur que la limite logiciel est fixée pour le moment à 36... mais il n'y a pas beaucoup d'info la-dessus.

J'ai l'impression que je ne peux pas dépasser 10 device!
J'arrive pas à en ajouter, mais si je retire un associer, je peux associer ceux que je ne pouvais pas associer avant...
Ce qui est bizarre c'est que dans les log je vois rien passer du tout pour l'ajout de nouveau device lorsque j'en ai déjà 10 d'associés.

Merci d'avance.

zaraki673, 2017-12-14:
Bonjour,

aucune idée, pour le moment je n'ai que 6 devices sur la zigate pour mes tests, je testerais des que possible d'en ajouter (je dois en recevoir de nouveaux bientôt)

james013, 2017-12-14:
J'en ai également 10 et le 11eme ne passe pas !

tornlouses, 2017-12-15:
Arf.
Je ne pense pas que ce soit une limitation au niveau du plugin?
Limite physique de la Zigate?

zaraki673, 2017-12-16:
pas de limitation coté plugin, au pire tu arriverais a associer et il remonterais pas dans domoticz

ou peut etre quand j'initialise la zigate, mais je ne fais q'utilisé les infos de la doc et il me semble pas y voir de limitation (ou j ai mal compris/lu )

tornlouses, 2017-12-17:
Ok,
J'essayerai de regarder cette partie.
J'ai lancé Akila sur le sujet sur son site "Faire soit même". On va voir son retour aussi.
Merci.

tornlouses, 2017-12-18:
Salut,
D'après le retour d'Akila sur le forum, ça coinçait du côté Zigate.
Je pense que tu peux clore ce ticket.
:)
Merci.

closed 2017-12-18 by deennoo