zigbeefordomoticz/Domoticz-Zigbee #34: Plantage Domoticz sur réception d'informations d'un capteur température Xiaomi Aqara

tornlouses, 2017-12-14:
Bonjour Zaraki673,

Comme dit sur le forum j'ai un plantage de Domoticz de temps (aléatoire) suite à la réception d'un message d'un capteur de température Xiaomi Aqara. Comme on pourra remarquer c'est toujours le même.
Cependant en temps normal il fonctionne bien et j'ai régulièrement des update d'info le concernant.

Mes 4 erreurs loguées :

`2017-12-09 17:07:14.102  (Zigate) Calling message handler 'onMessage'.
2017-12-09 17:07:14.102  (Zigate) onMessage called
2017-12-09 17:07:14.102  (Zigate) ZigateDecode - decodind data : 018102120210021f0216293f0214021102140213021010021029021002122720a203
2017-12-09 17:07:14.102  (Zigate) ZigateRead - decoded data : 018102000f06293f040104030010002900022720a203
2017-12-09 17:07:14.102  (Zigate) ZigateRead - Message Type : 8102, Data : 293f040104030010002900022720, RSSI : a2, Length : 000f, Checksum : 06
2017-12-09 17:07:14.103  (Zigate) ZigateRead - MsgType 8102 - reception data : 018102000f06293f040104030010002900022720a203 ClusterID : 0403 Attribut ID : 00
2017-12-09 17:07:14.103  (Zigate) MajDomoDevice - Device ID : 3f04 - Device EP : 01 - Type : Barometer - Value : 1001.6
2017-12-09 17:07:14.103  (Zigate) MajDomoDevice baro CurrentsValue : 21.4;48.9;1;1001;1001.8
2017-12-09 17:07:14.103  (Zigate) MajDomoDevice bar NewSvalue : 21.4;48.9;1;1001.6;1001
2017-12-09 17:07:14.103  (SBD Temperature) Updating device from 0:'21.4;48.9;1;1001;1001.8' to have values 0:'21.4;48.9;1;1001.6;1001'.
2017-12-09 17:07:14.108  (Zigate) Update 0:'21.4;48.9;1;1001.6;1001' (SBD Temperature)
2017-12-09 17:07:14.109  (Zigate) MajDomoDevice baro CurrentsValue : 1001;1001.8
2017-12-09 17:07:14.109  (Zigate - Barometer - 3f04) Updating device from 0:'1001;1001.8' to have values 0:'1001.6;1001'.
2017-12-09 17:07:14.110  (Zigate) Update 0:'1001.6;1001' (Zigate - Barometer - 3f04)
2017-12-09 17:07:14.110  (Zigate) ZigateRead - MsgType 8102 - reception atm : 1001.6
2017-12-09 17:07:14.111  (Zigate) Deallocating connection object 'ZiGate' (/dev/ttyUSB1:).
2017-12-09 17:07:22.483  (Zigate) Calling message handler 'onHeartbeat'.
2017-12-09 17:07:22.498  Error: Domoticz received fatal signal 11 !...
2017-12-09 17:07:22.560  Error:   /home/pi/domoticz/domoticz() [0x236374]
2017-12-09 17:07:22.561  Error:   /home/pi/domoticz/domoticz(_Z14signal_handleri+0x4c) [0x23642c]
2017-12-09 17:07:22.561  Error:   /lib/arm-linux-gnueabihf/libc.so.6(__default_sa_restorer+0) [0x76bc96b0]
`

`2017-12-10 02:06:14.319  (Zigate) Calling message handler 'onMessage'.
2017-12-10 02:06:14.319  (Zigate) onMessage called
2017-12-10 02:06:14.319  (Zigate) ZigateDecode - decodind data : 018102120210021f10a83f02140211021402130210100210290210021226b6a203
2017-12-10 02:06:14.319  (Zigate) ZigateRead - decoded data : 018102000f10a83f0401040300100029000226b6a203
2017-12-10 02:06:14.320  (Zigate) ZigateRead - Message Type : 8102, Data : a83f0401040300100029000226b6, RSSI : a2, Length : 000f, Checksum : 10
2017-12-10 02:06:14.320  (Zigate) ZigateRead - MsgType 8102 - reception data : 018102000f10a83f0401040300100029000226b6a203 ClusterID : 0403 Attribut ID : 00
2017-12-10 02:06:14.320  (Zigate) MajDomoDevice - Device ID : 3f04 - Device EP : 01 - Type : Barometer - Value : 991.0
2017-12-10 02:06:14.320  (Zigate) MajDomoDevice baro CurrentsValue : 21.0;50.8;1;991;991.3
2017-12-10 02:06:14.320  (Zigate) MajDomoDevice bar NewSvalue : 21.0;50.8;1;991.0;991
2017-12-10 02:06:14.320  (SBD Temperature) Updating device from 0:'21.0;50.8;1;991;991.3' to have values 0:'21.0;50.8;1;991.0;991'.
2017-12-10 02:06:14.326  (Zigate) Update 0:'21.0;50.8;1;991.0;991' (SBD Temperature)
2017-12-10 02:06:14.326  (Zigate) MajDomoDevice baro CurrentsValue : 991;991.3
2017-12-10 02:06:14.326  (Zigate - Barometer - 3f04) Updating device from 0:'991;991.3' to have values 0:'991.0;991'.
2017-12-10 02:06:14.328  (Zigate) Update 0:'991.0;991' (Zigate - Barometer - 3f04)
2017-12-10 02:06:14.328  (Zigate) ZigateRead - MsgType 8102 - reception atm : 991.0
2017-12-10 02:06:14.328  (Zigate) Deallocating connection object 'ZiGate' (/dev/ttyUSB1:).
2017-12-10 02:06:15.131  (Zigate) Calling message handler 'onHeartbeat'.
2017-12-10 02:06:15.145  Error: Domoticz received fatal signal 11 !...
2017-12-10 02:06:15.148  Error:   /home/pi/domoticz/domoticz() [0x236374]
2017-12-10 02:06:15.149  Error:   /home/pi/domoticz/domoticz(_Z14signal_handleri+0x4c) [0x23642c]
2017-12-10 02:06:15.149  Error:   /lib/arm-linux-gnueabihf/libc.so.6(__default_sa_restorer+0) [0x76c7a6b0]
`

`2017-12-11 20:04:59.281  (Zigate) Calling message handler 'onMessage'.
2017-12-11 20:04:59.281  (Zigate) onMessage called
2017-12-11 20:04:59.281  (Zigate) ZigateDecode - decodind data : 018102120210021f7c433f02140211021402130210100210290210021226021a9903
2017-12-11 20:04:59.281  (Zigate) ZigateRead - decoded data : 018102000f7c433f04010403001000290002260a9903
2017-12-11 20:04:59.281  (Zigate) ZigateRead - Message Type : 8102, Data : 433f04010403001000290002260a, RSSI : 99, Length : 000f, Checksum : 7c
2017-12-11 20:04:59.282  (Zigate) ZigateRead - MsgType 8102 - reception data : 018102000f7c433f04010403001000290002260a9903 ClusterID : 0403 Attribut ID : 00
2017-12-11 20:04:59.282  (Zigate) MajDomoDevice - Device ID : 3f04 - Device EP : 01 - Type : Barometer - Value : 973.8
2017-12-11 20:04:59.282  (Zigate) MajDomoDevice baro CurrentsValue : 21.1;46.2;1;973;973.5
2017-12-11 20:04:59.282  (Zigate) MajDomoDevice bar NewSvalue : 21.1;46.2;1;973.8;973
2017-12-11 20:04:59.282  (SBD Temperature) Updating device from 0:'21.1;46.2;1;973;973.5' to have values 0:'21.1;46.2;1;973.8;973'.
2017-12-11 20:04:59.284  (Zigate) Update 0:'21.1;46.2;1;973.8;973' (SBD Temperature)
2017-12-11 20:04:59.284  (Zigate) MajDomoDevice baro CurrentsValue : 973;973.5
2017-12-11 20:04:59.284  (Zigate - Barometer - 3f04) Updating device from 0:'973;973.5' to have values 0:'973.8;973'.
2017-12-11 20:04:59.285  (Zigate) Update 0:'973.8;973' (Zigate - Barometer - 3f04)
2017-12-11 20:04:59.286  (Zigate) ZigateRead - MsgType 8102 - reception atm : 973.8
2017-12-11 20:04:59.286  (Zigate) Deallocating connection object 'ZiGate' (/dev/ttyUSB1:).
2017-12-11 20:04:59.389  (Zigate) Deallocating connection object 'ZiGate' (/dev/ttyUSB1:).
[... 55 lines cut ...]
2017-12-10 17:59:24.691  (Zigate) MajDomoDevice baro CurrentsValue : 968;968.9
2017-12-10 17:59:24.691  (Zigate - Barometer - 3f04) Updating device from 0:'968;968.9' to have values 0:'968.6;968'.
2017-12-10 17:59:24.692  (Zigate) Update 0:'968.6;968' (Zigate - Barometer - 3f04)
2017-12-10 17:59:24.693  (Zigate) ZigateRead - MsgType 8102 - reception atm : 968.6
2017-12-10 17:59:32.557  (Zigate) Calling message handler 'onHeartbeat'.
`

`2017-12-08 08:38:23.905  (Zigate) onMessage called
2017-12-08 08:38:23.905  (Zigate) ZigateDecode - decodind data : 018102120210021fc9253f021402110214021302100210021029021002120213e99c03
2017-12-08 08:38:23.905  (Zigate) ZigateRead - decoded data : 018102000fc9253f0401040300000029000203e99c03
2017-12-08 08:38:23.906  (Zigate) ZigateRead - Message Type : 8102, Data : 253f0401040300000029000203e9, RSSI : 9c, Length : 000f, Checksum : c9
2017-12-08 08:38:23.906  (Zigate) ZigateRead - MsgType 8102 - reception data : 018102000fc9253f0401040300000029000203e99c03 ClusterID : 0403 Attribut ID : 0000 Src Addr : 3f04 Scr Ep: 01
2017-12-08 08:38:23.906  (Zigate) MajDomoDevice - Device ID : 3f04 - Device EP : 01 - Type : Barometer - Value : 1001
2017-12-08 08:38:23.906  (Zigate) MajDomoDevice baro CurrentsValue : 21.4;46.0;1;1001.6;1001
2017-12-08 08:38:23.907  (Zigate) MajDomoDevice bar NewSvalue : 21.4;46.0;1;1001;1001.6
2017-12-08 08:38:23.907  (SBD Temperature) Updating device from 0:'21.4;46.0;1;1001.6;1001' to have values 0:'21.4;46.0;1;1001;1001.6'.
2017-12-08 08:38:23.910  (Zigate) Update 0:'21.4;46.0;1;1001;1001.6' (SBD Temperature)
2017-12-08 08:38:23.910  (Zigate) MajDomoDevice baro CurrentsValue : 1001.6;1001
2017-12-08 08:38:23.911  (Zigate - Barometer - 3f04) Updating device from 0:'1001.6;1001' to have values 0:'1001;1001.6'.
2017-12-08 08:38:23.912  (Zigate) Update 0:'1001;1001.6' (Zigate - Barometer - 3f04)
2017-12-08 08:38:23.912  (Zigate) ZigateRead - MsgType 8102 - reception atm : 10.0
2017-12-08 08:38:23.912  (Zigate) Calling message handler 'onMessage'.
`

Merci d'avance pour le retour.

zaraki673, 2017-12-16:
Bonjour, 

quel version de domoticz ? ton système est bien à jour ? rpi1, 2/3 ?

le soucis d'alimentation a ete ecarté, il me semble non ?

tornlouses, 2017-12-17:
Bonjour,
L'alimentation n'est pas complétement mise hors cause... je vais me faire un 2ème domoticz sur un autre rasp pour voir si ça règle le problème.
Depuis l'ouverture du ticket je n'est pas eu le problème... par contre c'est un de mes rflink qui n'est plus joignable de temps en temps...

Sinon mon installation pour le moment j'ai sur un PI3 la version de domoticz :
Version: 3.8775
Build Hash: 261e2df6
Compile Date: 2017-12-06 18:18:34

tornlouses, 2018-01-03:
Bon depuis que je l'ai passé sur un nouveau rapsberry, il tourne sans reproduire le problème de signal 11 depuis le 26/12/2017.
C'était donc peut être un problème d'alimentation.
Merci.

zaraki673, 2018-01-03:
ok, je clos on réouvrira au besoin

;)

closed 2018-01-03 by zaraki673