zigbeefordomoticz/Domoticz-Zigbee #35: Xiaomi Magic Cube

cjonhammar, 2017-12-17:
Problème avec Xiaomi Cube:

2017-12-17 21:45:49.810 (CJ_Zigate) onMessage called
2017-12-17 21:45:49.811 (CJ_Zigate) ZigateDecode - decodind data : 018102120210021f60207b7902120210120210550210210210021202100210a803
2017-12-17 21:45:49.812 (CJ_Zigate) ZigateRead - decoded data : 018102000f60207b790200120055002100020000a803
2017-12-17 21:45:49.812 (CJ_Zigate) ZigateRead - Message Type : 8102, Data : 207b790200120055002100020000, RSSI : a8, Length : 000f, Checksum : 60
2017-12-17 21:45:49.813 (CJ_Zigate) ZigateRead - MsgType 8102 - reception data : 018102000f60207b790200120055002100020000a803 ClusterID : 0012 Attribut ID : 0055 Src Addr : 7b79 Scr Ep: 02
2017-12-17 21:45:49.814 (CJ_Zigate) MajDomoDevice - Device ID : 7b79 - Device EP : 02 - Type : Switch - Value : 0000
2017-12-17 21:45:49.814 (CJ_Zigate) ZigateRead - MsgType 8102 - reception Xiaomi Magic Cube Value : 0000

deennoo, 2017-12-17:
Quel mouvement fais tu ?

cjonhammar, 2017-12-17:
Toute action me donne ce genre d'erreur, voici quelques exemples:

2017-12-17 23:05:41.552 (CJ_Zigate) Received 4 bytes of data:
2017-12-17 23:05:41.552 (CJ_Zigate) 01 81 02 12 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .�..
2017-12-17 23:05:41.552 (CJ_Zigate) Received 22 bytes of data:
2017-12-17 23:05:41.552 (CJ_Zigate) 02 10 02 1f 42 22 7b 79 02 12 02 10 12 02 10 55 02 10 21 02 ....B"{y.......U..!.
2017-12-17 23:05:41.553 (CJ_Zigate) 10 02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..
2017-12-17 23:05:41.553 (CJ_Zigate) Received 7 bytes of data:
2017-12-17 23:05:41.553 (CJ_Zigate) 12 02 10 02 12 8a 03 .. .. .. .. .. .. .. .. .. .. .. .. .. .....�.
2017-12-17 23:05:41.553 (CJ_Zigate) Calling message handler 'onMessage'.
2017-12-17 23:05:41.554 (CJ_Zigate) onMessage called
2017-12-17 23:05:41.554 (CJ_Zigate) Calling message handler 'onMessage'.
2017-12-17 23:05:41.554 (CJ_Zigate) onMessage called
2017-12-17 23:05:41.554 (CJ_Zigate) Calling message handler 'onMessage'.
2017-12-17 23:05:41.555 (CJ_Zigate) onMessage called
2017-12-17 23:05:41.555 (CJ_Zigate) ZigateDecode - decodind data : 018102120210021f42227b79021202101202105502102102100212021002128a03
2017-12-17 23:05:41.557 (CJ_Zigate) ZigateRead - decoded data : 018102000f42227b7902001200550021000200028a03
2017-12-17 23:05:41.557 (CJ_Zigate) ZigateRead - Message Type : 8102, Data : 227b790200120055002100020002, RSSI : 8a, Length : 000f, Checksum : 42
2017-12-17 23:05:41.558 (CJ_Zigate) ZigateRead - MsgType 8102 - reception data : 018102000f42227b7902001200550021000200028a03 ClusterID : 0012 Attribut ID : 0055 Src Addr : 7b79 Scr Ep: 02
2017-12-17 23:05:41.558 (CJ_Zigate) MajDomoDevice - Device ID : 7b79 - Device EP : 02 - Type : Switch - Value : 0002
2017-12-17 23:05:41.559 (CJ_Zigate) onMessage - effacement de la trame suite a une erreur de decodage : 018102120210021f42227b79021202101202105502102102100212021002128a03
2017-12-17 23:05:42.763 (CJ_Zigate) Calling message handler 'onHeartbeat'.
2017-12-17 23:05:46.288 (CJ_Zigate) Received 33 bytes of data:
2017-12-17 23:05:46.288 (CJ_Zigate) 01 81 02 12 02 10 02 1f 54 23 7b 79 02 12 02 10 12 02 10 55 .�......T#{y.......U
2017-12-17 23:05:46.289 (CJ_Zigate) 02 10 21 02 10 02 12 02 10 02 13 9c 03 .. .. .. .. .. .. .. ..!........�.
2017-12-17 23:05:46.289 (CJ_Zigate) Calling message handler 'onMessage'.
2017-12-17 23:05:46.289 (CJ_Zigate) onMessage called
2017-12-17 23:05:46.290 (CJ_Zigate) ZigateDecode - decodind data : 018102120210021f54237b79021202101202105502102102100212021002139c03
2017-12-17 23:05:46.291 (CJ_Zigate) ZigateRead - decoded data : 018102000f54237b7902001200550021000200039c03
2017-12-17 23:05:46.291 (CJ_Zigate) ZigateRead - Message Type : 8102, Data : 237b790200120055002100020003, RSSI : 9c, Length : 000f, Checksum : 54
2017-12-17 23:05:46.292 (CJ_Zigate) ZigateRead - MsgType 8102 - reception data : 018102000f54237b7902001200550021000200039c03 ClusterID : 0012 Attribut ID : 0055 Src Addr : 7b79 Scr Ep: 02
2017-12-17 23:05:46.293 (CJ_Zigate) MajDomoDevice - Device ID : 7b79 - Device EP : 02 - Type : Switch - Value : 0003
2017-12-17 23:05:46.293 (CJ_Zigate - lumi.sensor_cube - 7b79) Updating device from 1:'10' to have values 7:'70'.
2017-12-17 23:05:46.302 (CJ_Zigate) Update 7:'70' (CJ_Zigate - lumi.sensor_cube - 7b79)
2017-12-17 23:05:46.303 (CJ_Zigate) ZigateRead - MsgType 8102 - reception Xiaomi Magic Cube Value : 0003
2017-12-17 23:05:47.556 (CJ_Zigate) Received 6 bytes of data:
2017-12-17 23:05:47.557 (CJ_Zigate) 01 81 02 12 02 10 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .�....
2017-12-17 23:05:47.557 (CJ_Zigate) Received 12 bytes of data:
2017-12-17 23:05:47.557 (CJ_Zigate) 02 1f a0 24 7b 79 02 12 02 10 12 02 .. .. .. .. .. .. .. .. ..�${y......
2017-12-17 23:05:47.557 (CJ_Zigate) Received 13 bytes of data:
2017-12-17 23:05:47.558 (CJ_Zigate) 10 55 02 10 21 02 10 02 12 02 10 02 13 .. .. .. .. .. .. .. .U..!........
2017-12-17 23:05:47.558 (CJ_Zigate) Received 2 bytes of data:
2017-12-17 23:05:47.558 (CJ_Zigate) 6f 03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. o.
2017-12-17 23:05:47.558 (CJ_Zigate) Calling message handler 'onMessage'.
2017-12-17 23:05:47.559 (CJ_Zigate) onMessage called
2017-12-17 23:05:47.559 (CJ_Zigate) Calling message handler 'onMessage'.
2017-12-17 23:05:47.559 (CJ_Zigate) onMessage called
2017-12-17 23:05:47.560 (CJ_Zigate) Calling message handler 'onMessage'.
2017-12-17 23:05:47.560 (CJ_Zigate) onMessage called
2017-12-17 23:05:47.560 (CJ_Zigate) Calling message handler 'onMessage'.
2017-12-17 23:05:47.560 (CJ_Zigate) onMessage called
2017-12-17 23:05:47.561 (CJ_Zigate) ZigateDecode - decodind data : 018102120210021fa0247b79021202101202105502102102100212021002136f03
2017-12-17 23:05:47.562 (CJ_Zigate) ZigateRead - decoded data : 018102000fa0247b7902001200550021000200036f03
2017-12-17 23:05:47.563 (CJ_Zigate) ZigateRead - Message Type : 8102, Data : 247b790200120055002100020003, RSSI : 6f, Length : 000f, Checksum : a0
2017-12-17 23:05:47.564 (CJ_Zigate) ZigateRead - MsgType 8102 - reception data : 018102000fa0247b7902001200550021000200036f03 ClusterID : 0012 Attribut ID : 0055 Src Addr : 7b79 Scr Ep: 02
2017-12-17 23:05:47.564 (CJ_Zigate) MajDomoDevice - Device ID : 7b79 - Device EP : 02 - Type : Switch - Value : 0003
2017-12-17 23:05:47.564 (CJ_Zigate) ZigateRead - MsgType 8102 - reception Xiaomi Magic Cube Value : 0003
2017-12-17 23:05:52.729 (CJ_Zigate) Calling message handler 'onHeartbeat'.
[... cut ...]
2017-12-17 23:06:31.336 (CJ_Zigate) ZigateRead - Message Type : 8102, Data : 277b7903000cff050021000201f4, RSSI : 63, Length : 000f, Checksum : e9
2017-12-17 23:06:31.337 (CJ_Zigate) ZigateRead - MsgType 8102 - reception data : 018102000fe9277b7903000cff050021000201f46303 ClusterID : 000c Attribut ID : ff05 Src Addr : 7b79 Scr Ep: 03
2017-12-17 23:06:31.337 (CJ_Zigate) MajDomoDevice - Device ID : 7b79 - Device EP : 03 - Type : Vert Rot - Value : 01f4
2017-12-17 23:06:31.337 (CJ_Zigate) ZigateRead - MsgType 8102 - reception Xiaomi Magic Cube Value Vert Rot : 01f4

deennoo, 2017-12-18:
Peux tu "nettoyer" les logs stp et ne mettre que ce qui ne fonctionne pas ?

MajDomoDevice - Device ID : 7b79 - Device EP : 02 - Type : Switch - Value : 0002
2017-12-17 23:05:41.559 (CJ_Zigate) onMessage - effacement de la trame suite a une erreur de decodage : 018102120210021f42227b790212021012

Est l'erreur ?

cjonhammar, 2017-12-18:
Si moins est plus? :)
Oui, c'est l'erreur que je vois toujours dans les journaux.

deennoo, 2017-12-18:
Essaye d'identifier exactement le mouvement qui provoque cette erreur.
'move' ? Alerte ? 

Les value de rotation ne sont également pas gérer et peuvent provoquer cette erreur.

Tout n'est pas encore gérer par le plugin, n'ayant pas réussit à reproduire certains mouvement, ou isoler les bonnes valeurs.

zaraki673, 2018-01-14:
voir #50 pour trames et mouvements correspondant

closed 2018-01-14 by zaraki673