zigbeefordomoticz/Domoticz-Zigbee #36: Capteur température non mise à jour

max5962, 2017-12-18:
Mes capteurs températures ne remontent plus d'informations. du coup j'ai fouillé un peu partout, et en mettant la zigate en mode début j'ai trouvé ça : 

18 09:52:25.362 (ZigateManager) ZigateDecode - decodind data : 018102120210021fb5625f4b02110214021202100210021029021002120218107b03
18 09:52:25.362 (ZigateManager) ZigateRead - decoded data : 018102000fb5625f4b01040200000029000208107b03
18 09:52:25.362 (ZigateManager) ZigateRead - Message Type : 8102, Data : 625f4b0104020000002900020810, RSSI : 7b, Length : 000f, Checksum : b5
18 09:52:25.362 (ZigateManager) ZigateRead - MsgType 8102 - reception data : 018102000fb5625f4b01040200000029000208107b03 ClusterID : 0402 Attribut ID : 0000 Src Addr : 5f4b Scr Ep: 01
18 09:52:25.362 (ZigateManager) MajDomoDevice - Device ID : 5f4b - Device EP : 01 - Type : Temperature - Value : 20.6
18 09:52:25.362 (ZigateManager) onMessage - effacement de la trame suite a une erreur de decodage : 018102120210021fb5625f4b02110214021202100210021029021002120218107b03 . 

Mon module 5f4b continu à m'afficher 19.9, nous avons bien une mise à jour non fonctionnel.
J'ai cette erreur au bout de 2 jours :)

max5962, 2017-12-21:
Hello,
Avez vous besoin de plus de logs ?
Max

zaraki673, 2017-12-21:
Bonjour,

oui je veux bien, je n'arrive pas à reproduire ton bug :(

Quel version du plugin et de domoticz ?
vois tu qd meme des trames passer ?

as tu fais une mise à jour (plugin ou domotixz ou les deux) avant que cela
ne commence à bug?

max5962, 2017-12-21:
Hello :)
Merci pour la prise en compte de mon bug.
Voici la version domoticz : Version: 3.8786
Version plugin : 1.0.7

C'est une instal fresh (je viens de recevoir la zigate).

Concrètement , j'ai 2 capteurs température. (Le carré) . Le 1er à fonctionné 2h avant de déconner. Croyant qu'il déconnait je l'ai reset : maintenant il fonctionne.
Le deuxième fonctionnait puis le lendemain boom. D'où le logs.

Des que j'ai plus de logs je le mettrai ici.
Pour info un collegue à le même soucis.
Max

max5962, 2017-12-21:
Je n'ai plus le même niveau d'erreur mais ça semble updater toujours les mêmes valeurs.  

2017-12-21 17:23:40.922 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.922 (ZigateManager) onMessage called
2017-12-21 17:23:40.922 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.922 (ZigateManager) onMessage called
2017-12-21 17:23:40.922 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.922 (ZigateManager) onMessage called
2017-12-21 17:23:40.922 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.922 (ZigateManager) onMessage called
2017-12-21 17:23:40.922 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.922 (ZigateManager) onMessage called
2017-12-21 17:23:40.922 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.922 (ZigateManager) onMessage called
2017-12-21 17:23:40.922 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.922 (ZigateManager) onMessage called
2017-12-21 17:23:40.922 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.922 (ZigateManager) onMessage called
2017-12-21 17:23:40.922 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.922 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.923 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.923 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.923 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.923 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.923 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.923 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.923 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.923 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.923 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.923 (ZigateManager) onMessage called
2017-12-21 17:23:40.923 (ZigateManager) Calling message handler 'onMessage'.
2017-12-21 17:23:40.924 (ZigateManager) onMessage called
2017-12-21 17:23:40.924 (ZigateManager) ZigateDecode - decodind data : 0181021202103262255f4b021102100210ff0211021042021025021121a9021b021421a843021521021702100216240211021002100210021064292a021765212d18662ba79202110210021a21021002107503
2017-12-21 17:23:40.924 (ZigateManager) ZigateRead - decoded data : 018102003262255f4b010000ff01004200250121a90b0421a843052107000624010000000064292a0765212d18662ba79201000a2100007503
2017-12-21 17:23:40.924 (ZigateManager) ZigateRead - Message Type : 8102, Data : 255f4b010000ff01004200250121a90b0421a843052107000624010000000064292a0765212d18662ba79201000a210000, RSSI : 75, Length : 0032, Checksum : 62
2017-12-21 17:23:40.924 (ZigateManager) ZigateRead - MsgType 8102 - reception data : 018102003262255f4b010000ff01004200250121a90b0421a843052107000624010000000064292a0765212d18662ba79201000a2100007503 ClusterID : 0000 Attribut ID : ff01 Src Addr : 5f4b Scr Ep: 01
2017-12-21 17:23:40.924 (ZigateManager) ZigateRead - MsgType 8102 - reception heartbeat (0000) : 0121a90b0421a843052107000624010000000064292a0765212d18662ba79201000a210000
2017-12-21 17:23:40.924 (ZigateManager) ZigateRead - MsgType 8102 - reception batteryLVL (0000) : 100 pour le device addr : 5f4b
2017-12-21 17:23:40.924 (ZigateManager) Devices already exist. Unit=7
2017-12-21 17:23:40.924 (ZigateManager) CurrentnValue = 0
2017-12-21 17:23:40.924 (ZigateManager) CurrentsValue = 19.9;59.5;1;1017.6;1017
2017-12-21 17:23:40.924 (ZigateManager) BatteryLvl = 100
2017-12-21 17:23:40.924 (ZigateManager - Temp+Hum+Baro - 5f4b) Updating device from 0:'19.9;59.5;1;1017.6;1017' to have values 0:'19.9;59.5;1;1017.6;1017'.
2017-12-21 17:23:40.925 (ZigateManager) Devices already exist. Unit=8
2017-12-21 17:23:40.925 (ZigateManager) CurrentnValue = 0
2017-12-21 17:23:40.925 (ZigateManager) CurrentsValue = 19.9;59.5;1
2017-12-21 17:23:40.925 (ZigateManager) BatteryLvl = 100
2017-12-21 17:23:40.925 (ZigateManager - Temp+Hum - 5f4b) Updating device from 0:'19.9;59.5;1' to have values 0:'19.9;59.5;1'.
2017-12-21 17:23:40.926 (ZigateManager) Devices already exist. Unit=9
2017-12-21 17:23:40.926 (ZigateManager) CurrentnValue = 0
2017-12-21 17:23:40.926 (ZigateManager) CurrentsValue = 19.9
2017-12-21 17:23:40.926 (ZigateManager) BatteryLvl = 100
2017-12-21 17:23:40.926 (Temperature Etage) Updating device from 0:'19.9' to have values 0:'19.9'.
2017-12-21 17:23:40.926 (ZigateManager) Devices already exist. Unit=10
2017-12-21 17:23:40.926 (ZigateManager) CurrentnValue = 59
2017-12-21 17:23:40.926 (ZigateManager) CurrentsValue = 0
2017-12-21 17:23:40.926 (ZigateManager) BatteryLvl = 100
2017-12-21 17:23:40.926 (Humidité Etage) Updating device from 59:'0' to have values 59:'0'.
2017-12-21 17:23:40.927 (ZigateManager) Devices already exist. Unit=11
2017-12-21 17:23:40.927 (ZigateManager) CurrentnValue = 0
2017-12-21 17:23:40.927 (ZigateManager) CurrentsValue = 1017.6;1017
2017-12-21 17:23:40.927 (ZigateManager) BatteryLvl = 100
2017-12-21 17:23:40.927 (Pression - Etage) Updating device from 0:'1017.6;1017' to have values 0:'1017.6;1017'.

max5962, 2017-12-21:
Je peux très bien installer une version de tests si besoin.
Max

zaraki673, 2017-12-21:
pour la version de test il va falloir attendre encore, je suis sur une
nouvelle version plus "universelle"

pour ton log, c est normal qu il mette a jour avec les même valeur, en fait
c est juste une trame de niveau de batterie que tu reçois ds ce cas la,
essaie de faire varier la temperature de la sonde (changements de piece,
mis à l exterieur, voir frigo ...), il n y a qu au moment d un changement
de temperature/humidite ou pression que la sonde envoie les trames de
changements d'états

apres pas avoir de variation sur 48h, c est pas possible (sauf pt etre ds
un bunker :D ), donc c'est étrange que ca ne change pas :(

max5962, 2017-12-21:
Vu que ça fonctionnait pas, j'ai décidé de reset à nouveau mon capteur.
Suite à la manip, j'ai des trames du styles : 
2017-12-21 20:52:39.251  (ZigateManager) ZigateDecode - decodind data : 018102120210021f1f0215272f0211021402120210021002102902100212021a3c8403
2017-12-21 20:52:39.251  (ZigateManager) ZigateRead - decoded data : 018102000f1f05272f0104020000002900020a3c8403
2017-12-21 20:52:39.251  (ZigateManager) ZigateRead - Message Type : 8102, Data : 05272f0104020000002900020a3c, RSSI : 84, Length : 000f, Checksum : 1f
2017-12-21 20:52:39.251  (ZigateManager) ZigateRead - MsgType 8102 - reception data : 018102000f1f05272f0104020000002900020a3c8403 ClusterID : 0402 Attribut ID : 0000 Src Addr : 272f Scr Ep: 01
2017-12-21 20:52:39.252  (ZigateManager) MajDomoDevice - Device ID : 272f - Device EP : 01 - Type : Temperature - Value : 26.2
2017-12-21 20:52:39.252  (ZigateManager) ZigateRead - MsgType 8102 - reception temp : 26.2

Je me suis dis chouette, il m'a trouvé un nouveau device, mais en faite non : MajDomoDevice
Sauf que j'ai aucun device qui a ce nom :( ( même dans "inutilisé"
comment je fait ? :)

merci

zaraki673, 2017-12-21:
va dans hardware, tu effaces le plugin zigate,
tu reboot ton domoticz
reconfigure le plugin

zaraki673, 2017-12-21:
ah et mets a jour ton plugin au passage :p (avant de le reconfigurer )

Le 21 décembre 2017 à 23:20, Kevin Larsonneur <azazel673@gmail.com> a écrit
:

max5962, 2017-12-22:
Merci pour ces infos.
je tente ça.
pour la mise à jour, je me met sur la branche master ( dernier commit il y a 20 jours, je suis donc déjà à jour ) ou sur "universal-ZigBee---dev"  ? 

merci

zaraki673, 2017-12-22:
sur master :D

l'autre c est la future :P

;)

max5962, 2017-12-24:
Hello,
Après reset ça refonctionne :)
Je croise les doigts ^^
Maintenant c'est mon détecteur d'ouverture qui ne fonctionne plus :( alors qu'il fonctionnait hier ..)
J'ai un heartbeat mais pas de majdomodevice.

Je clos ce ticket ?
Merci

max5962, 2017-12-24:
Bon après reset du détecteur d'ouverture je crois maintenant les "open" mais plus vraiment les close. 
Doit y avoir un soucis quelque part chez moi :)

zaraki673, 2017-12-24:
dans les logs tu les vois pas passer non plus ?

max5962, 2017-12-27:
J'ai à nouveau reset le détecteur et ça semble fonctionner.
Si ça marche plus au bout de qq jours j'ouvrirai un ticket.
Merci :)

closed 2017-12-27 by max5962