zigbeefordomoticz/Domoticz-Zigbee #37: Xiaomi water detection, button, door detector integration

handfreezer, 2017-12-20:
Hello,

  I have all of those, may I help you to integrate it in any way?

Best regards

deennoo, 2017-12-21:
Of course your welcome.

Provid us some log when water sensor detect water

Button/door are already set

handfreezer, 2017-12-22:
Water detector, hope this help, if not enough just told me.
 2017-12-22 01:15:38.857 (Zigate) Calling message handler 'onHeartbeat'.
2017-12-22 01:15:41.272 (Zigate) Received 12 bytes of data:
2017-12-22 01:15:41.272 (Zigate) 01 80 48 02 10 02 1a 33 02 10 15 8d .. .. .. .. .. .. .. .. .�H....3...�
2017-12-22 01:15:41.272 (Zigate) Received 3 bytes of data:
2017-12-22 01:15:41.273 (Zigate) 02 10 02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ...
2017-12-22 01:15:41.273 (Zigate) Received 1 bytes of data:
2017-12-22 01:15:41.273 (Zigate) 11 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .
2017-12-22 01:15:41.273 (Zigate) Received 2 bytes of data:
2017-12-22 01:15:41.273 (Zigate) d7 8c .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ׌
2017-12-22 01:15:41.273 (Zigate) Received 1 bytes of data:
2017-12-22 01:15:41.273 (Zigate) 5f .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. _
2017-12-22 01:15:41.273 (Zigate) Received 1 bytes of data:
2017-12-22 01:15:41.273 (Zigate) 02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .
2017-12-22 01:15:41.273 (Zigate) Received 2 bytes of data:
2017-12-22 01:15:41.273 (Zigate) 10 6c .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .l
2017-12-22 01:15:41.273 (Zigate) Received 1 bytes of data:
2017-12-22 01:15:41.273 (Zigate) 03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .
2017-12-22 01:15:41.274 (Zigate) Calling message handler 'onMessage'.
2017-12-22 01:15:41.274 (Zigate) onMessage called
2017-12-22 01:15:41.274 (Zigate) Calling message handler 'onMessage'.
2017-12-22 01:15:41.274 (Zigate) onMessage called
2017-12-22 01:15:41.274 (Zigate) Calling message handler 'onMessage'.
2017-12-22 01:15:41.274 (Zigate) onMessage called
2017-12-22 01:15:41.274 (Zigate) Calling message handler 'onMessage'.
2017-12-22 01:15:41.274 (Zigate) onMessage called
2017-12-22 01:15:41.274 (Zigate) Calling message handler 'onMessage'.
2017-12-22 01:15:41.275 (Zigate) onMessage called
2017-12-22 01:15:41.275 (Zigate) Calling message handler 'onMessage'.
2017-12-22 01:15:41.275 (Zigate) onMessage called
2017-12-22 01:15:41.275 (Zigate) Calling message handler 'onMessage'.
2017-12-22 01:15:41.275 (Zigate) onMessage called
2017-12-22 01:15:41.275 (Zigate) Calling message handler 'onMessage'.
2017-12-22 01:15:41.275 (Zigate) onMessage called
2017-12-22 01:15:41.275 (Zigate) ZigateDecode - decodind data : 0180480210021a330210158d02100211d78c5f02106c03
2017-12-22 01:15:41.276 (Zigate) ZigateRead - decoded data : 018048000a3300158d0001d78c5f006c03
2017-12-22 01:15:41.276 (Zigate) ZigateRead - Message Type : 8048, Data : 00158d0001d78c5f00, RSSI : 6c, Length : 000a, Checksum : 33
2017-12-22 01:15:41.276 (Zigate) reception Leave indication : 018048000a3300158d0001d78c5f006c03
2017-12-22 01:15:48.842 (Zigate) Calling message handler 'onHeartbeat'.
2017-12-22 01:15:58.874 (Zigate) Calling message handler 'onHeartbeat'.

handfreezer, 2017-12-22:
About bouton switch, temp and motion detector, I succeed to see them in devices list, but after switch of motion sensor dont work (when I cross it sent a signal to domoticz because of log I saw, but no event are played in domoticz and dashboard do not notify me)

deennoo, 2017-12-22:
maybe it's time to wait a bit, as a new plugin version is on the way to fit more universal with zigbee device (xiaomi one aren't standard)

thx for your log, 

if your read it we just have the leave infomation `2017-12-22 01:15:41.276 (Zigate) ZigateRead - decoded data : 018048000a3300158d0001d78c5f006c03
2017-12-22 01:15:41.276 (Zigate) ZigateRead - Message Type : 8048, Data : 00158d0001d78c5f00, RSSI : 6c, Length : 000a, Checksum : 33
2017-12-22 01:15:41.276 (Zigate) reception Leave indication : 018048000a3300158d0001d78c5f006c03`

We need to have log when device join the zigbee network and when the sensor is sensing water

handfreezer, 2017-12-24:
OK, I understood.
Do you prefer log from nxp tool advices from zigate.fr website, or from domotics log screen (I only know to get them from web pages)?

Also, as I'm configuring my domoticz, I could do some test with next plugin verison if you need as it is not fully installed for now.

zaraki673, 2017-12-24:
easier for you, both should be good ;)

I hope new release will be ready before end of year ;)

handfreezer, 2017-12-26:
Well, I found something strange, in fact, seems recognize, creating the device, but does not appear in domoticz devices screen... log below:

```
grep -i -C5 dd1f /var/log/syslog
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) ZigateDecode - decodind data : 0102104d0210021ce5dd1f0210158d02100211d78c5f807b03
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) ZigateRead - decoded data : 01004d000ce5dd1f00158d0001d78c5f807b03
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) ZigateRead - Message Type : 004d, Data : dd1f00158d0001d78c5f80, RSSI : 7b, Length : 000c, Checksum : e5
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) reception Device announce : Source :dd1f, IEEE : 00158d0001d78c5f, Mac capa : 80
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Received 7 bytes of data:
Dec 26 23:32:10 srvcenter01 domoticz: (zigate)     01 02 10 4d 02 10 02 .. .. .. .. .. .. .. .. .. .. .. .. ..     ...M...
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Received 1 bytes of data:
Dec 26 23:32:10 srvcenter01 domoticz: (zigate)     1c .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Received 2 bytes of data:
--
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) ZigateDecode - decodind data : 0102104d0210021ce6dd1f0210158d02100211d78c5f807803
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) ZigateRead - decoded data : 01004d000ce6dd1f00158d0001d78c5f807803
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) ZigateRead - Message Type : 004d, Data : dd1f00158d0001d78c5f80, RSSI : 78, Length : 000c, Checksum : e6
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) reception Device announce : Source :dd1f, IEEE : 00158d0001d78c5f, Mac capa : 80
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
--
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) ZigateDecode - decodind data : 01810212021022490210dd1f021102100210021002150210420210156c756d692e73656e736f725f776c65616b2e6171317803
Dec 26 23:32:10 srvcenter01 domoticz: (zigate) ZigateRead - decoded data : 01810200224900dd1f0100000005004200156c756d692e73656e736f725f776c65616b2e6171317803
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) ZigateRead - Message Type : 8102, Data : 00dd1f0100000005004200156c756d692e73656e736f725f776c65616b2e617131, RSSI : 78, Length : 0022, Checksum : 49
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) ZigateRead - MsgType 8102 - reception data : 01810200224900dd1f0100000005004200156c756d692e73656e736f725f776c65616b2e6171317803 ClusterID : 0000 Attribut ID : 0005 Src Addr : dd1f Scr Ep: 01
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) ZigateRead - MsgType 8102 - reception heartbeat (0000) : 6c756d692e73656e736f725f776c65616b2e617131
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) ZigateRead - MsgType 8102 - reception heartbeat (0000) - MsgAttrID (0005) - Type de Device : lumi.sensor_wleak.aq1
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) ZigateRead - MsgType 8102 - reception heartbeat (0000) - MsgAttrID (0005) - Type de Device : lumi.sensor_wleak.aq1 Device not found, creating device
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) CreateDomoDevice - Device ID : dd1f Device EP : 01 Type : lumi.sensor_wleak.aq1
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
--
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) Calling message handler 'onMessage'.
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) onMessage called
Dec 26 23:32:11 srvcenter01 domoticz: (zigate) ZigateDecode - decodind data : 018102120210021e150210dd1f021102100210021002110210200210021102137803
[... cut ...]
Dec 26 23:33:47 srvcenter01 domoticz: (zigate) MajDomoDevice - Device ID : dd1f - Device EP : 01 - Type : Switch - Value : 00
Dec 26 23:33:56 srvcenter01 domoticz: (zigate) Calling message handler 'onHeartbeat'.
Dec 26 23:34:00 srvcenter01 domoticz: LUA: FE : script time deco noel
Dec 26 23:34:00 srvcenter01 domoticz: LUA: FE : script time deco noel -- fin
Dec 26 23:34:01 srvcenter01 CRON[8784]: (root) CMD (/root/domoticz/restart_domoticz.sh)
Dec 26 23:34:06 srvcenter01 domoticz: (zigate) Calling message handler 'onHeartbeat'.
Binary file /var/log/syslog matches

```

handfreezer, 2017-12-26:
hope this help, it is easiest for me to get domo log...

zaraki673, 2017-12-27:
Thank's !!!

yes it's help, you give me its name *"lumi.sensor_wleak.aq1"*

as the device name was unknow, i don't really create the device in
domoticz, now i will be able, thk's ;)

handfreezer, 2018-01-01:
Does version 1.0.8 could be drop in place in domoticz without loosing current config and working devices?

zaraki673, 2018-01-01:
yes you can ;)

closed 2018-01-10 by zaraki673

handfreezer, 2018-01-10:
Juste to say it is working with last 1.0.8 (but I lost all previous registered devices... not a problem for me as I'm installing)