zigbeefordomoticz/Domoticz-Zigbee #38: ignore device who send Off value who isn't useful

deennoo, 2017-12-22:
Some Xiaomi device like sw86-2 or sw86-1 switch send value = 0

Value = 0 come just after pression button., this value isn't usefull

on log this give

`0181020011d30a749703000c005500390004c0c0c0c0cf03 ClusterID : 000c Attribut ID : 0055 Src Addr : 7497 Scr Ep: 03
2017-12-22 09:19:26.062 (Zigate USB) MajDomoDevice - Device ID : 7497 - Device EP : 03 - Type : Vert Rot - Value : c0c0c0c0
2017-12-22 09:19:26.062 (Zigate USB) onMessage - effacement de la trame suite a une erreur de decodage : 01810212021011d3021a749702130210021c02105502103902100214c0c0c0c0cf03`

how to ignore this ?

zaraki673, 2017-12-22:
font chier Xiaomi a pas respecter le protocole zigbee :(

si tu envois off, y a deux trames off ?

deennoo, 2017-12-23:
Ce sont des boutons poussoirs, l'action est semblable à une télécommande, il n'y a jamais off. C'est tj on puis off.

zaraki673, 2018-03-20:
en fait c est push/release l'action 
du coup normal d'avoir le retour a l'état 0 (off) une fois l'inter relâché

voir plutôt pour changer le type de switch dans domoticz, qu en penses tu ?
(et du coup on peut gérer le long click , mais coté domoticz)

deennoo, 2018-03-31:
pas de long click sur les  sw86-2 or sw86-1

sw86-1 : click
sw86.2 : click droite, click gauche click des deux en meme temps

commit 48e8c052 (2018-04-14): creation device boutton et double bouttons

fix #38
closed 2018-04-14 by zaraki673