zigbeefordomoticz/Domoticz-Zigbee #39: Capteur de presence Aqara ne se met pas à jour

domopiap, 2017-12-22:
Bonjour
Domoticz V3.8795
Plugin 1.0.7
la mise à jour de la luminosité fonctionne mais pas celle du switch qui a le meme id . Voici les log :
 
2017-12-22 21:41:05.779 (zigate) onMessage called
2017-12-22 21:41:05.779 (zigate) ZigateDecode - decodind data : 018102120210021f4c3318f70211021402100210021002102102100212021002153f03
2017-12-22 21:41:05.782 (zigate) ZigateRead - decoded data : 018102000f4c3318f701040000000021000200053f03
2017-12-22 21:41:05.782 (zigate) ZigateRead - Message Type : 8102, Data : 3318f70104000000002100020005, RSSI : 3f, Length : 000f, Checksum : 4c
2017-12-22 21:41:05.783 (zigate) ZigateRead - MsgType 8102 - reception data : 018102000f4c3318f701040000000021000200053f03 ClusterID : 0400 Attribut ID : 0000 Src Addr : 18f7 Scr Ep: 01
2017-12-22 21:41:05.784 (zigate) MajDomoDevice - Device ID : 18f7 - Device EP : 01 - Type : Lux - Value : 5
2017-12-22 21:41:05.784 (zigate) ZigateRead - MsgType 8102 - reception LUX Sensor : 0005
2017-12-22 21:41:05.785 (zigate) Calling message handler 'onMessage'.
2017-12-22 21:41:05.785 (zigate) onMessage called
2017-12-22 21:41:05.785 (zigate) Calling message handler 'onMessage'.
2017-12-22 21:41:05.786 (zigate) onMessage called
2017-12-22 21:41:05.786 (zigate) Calling message handler 'onMessage'.
2017-12-22 21:41:05.786 (zigate) onMessage called
2017-12-22 21:41:05.787 (zigate) ZigateDecode - decodind data : 018102120210021e723418f7021102140216021002100210180210021102113f03
2017-12-22 21:41:05.789 (zigate) ZigateRead - decoded data : 018102000e723418f7010406000000180001013f03
2017-12-22 21:41:05.789 (zigate) ZigateRead - Message Type : 8102, Data : 3418f701040600000018000101, RSSI : 3f, Length : 000e, Checksum : 72
2017-12-22 21:41:05.790 (zigate) ZigateRead - MsgType 8102 - reception data : 018102000e723418f7010406000000180001013f03 ClusterID : 0406 Attribut ID : 0000 Src Addr : 18f7 Scr Ep: 01
2017-12-22 21:41:05.790 (zigate) MajDomoDevice - Device ID : 18f7 - Device EP : 01 - Type : Switch - Value : 01
2017-12-22 21:41:05.791 (zigate) ZigateRead - MsgType 8102 - reception Occupancy Sensor : 01
2017-12-22 21:41:11.162 (zigate) Calling message handler 'onHeartbeat'.

la MajDomoDevice du switch semble ne pas fonctionner chez moi...
Le capteur de temperature et le switch detecter de porte fonctionne très bien...

zaraki673, 2017-12-22:
Ton switch reste sur on ?
Si oui, mets a jour ton plugin (surtout efface hardware et réinstalle le, histoire d être sur que les devices se créer avec les bonnes infos

De : domopiap
Envoyé le :vendredi 22 décembre 2017 21:46
À : sasu-drooz/Domoticz-Zigate
Cc : Subscribed
Objet :[sasu-drooz/Domoticz-Zigate] Capteur de presence Aqara ne se met pas à jour (#39)

Bonjour
Domoticz V3.8795
Plugin 1.0.7
la mise à jour de la luminosité fonctionne mais pas celle du switch qui a le meme id . Voici les log :
2017-12-22 21:41:05.779 (zigate) onMessage called
2017-12-22 21:41:05.779 (zigate) ZigateDecode - decodind data : 018102120210021f4c3318f70211021402100210021002102102100212021002153f03
2017-12-22 21:41:05.782 (zigate) ZigateRead - decoded data : 018102000f4c3318f701040000000021000200053f03
2017-12-22 21:41:05.782 (zigate) ZigateRead - Message Type : 8102, Data : 3318f70104000000002100020005, RSSI : 3f, Length : 000f, Checksum : 4c
2017-12-22 21:41:05.783 (zigate) ZigateRead - MsgType 8102 - reception data : 018102000f4c3318f701040000000021000200053f03 ClusterID : 0400 Attribut ID : 0000 Src Addr : 18f7 Scr Ep: 01
2017-12-22 21:41:05.784 (zigate) MajDomoDevice - Device ID : 18f7 - Device EP : 01 - Type : Lux - Value : 5
2017-12-22 21:41:05.784 (zigate) ZigateRead - MsgType 8102 - reception LUX Sensor : 0005
2017-12-22 21:41:05.785 (zigate) Calling message handler 'onMessage'.
2017-12-22 21:41:05.785 (zigate) onMessage called
2017-12-22 21:41:05.785 (zigate) Calling message handler 'onMessage'.
2017-12-22 21:41:05.786 (zigate) onMessage called
2017-12-22 21:41:05.786 (zigate) Calling message handler 'onMessage'.
2017-12-22 21:41:05.786 (zigate) onMessage called
2017-12-22 21:41:05.787 (zigate) ZigateDecode - decodind data : 018102120210021e723418f7021102140216021002100210180210021102113f03
2017-12-22 21:41:05.789 (zigate) ZigateRead - decoded data : 018102000e723418f7010406000000180001013f03
2017-12-22 21:41:05.789 (zigate) ZigateRead - Message Type : 8102, Data : 3418f701040600000018000101, RSSI : 3f, Length : 000e, Checksum : 72
2017-12-22 21:41:05.790 (zigate) ZigateRead - MsgType 8102 - reception data : 018102000e723418f7010406000000180001013f03 ClusterID : 0406 Attribut ID : 0000 Src Addr : 18f7 Scr Ep: 01
2017-12-22 21:41:05.790 (zigate) MajDomoDevice - Device ID : 18f7 - Device EP : 01 - Type : Switch - Value : 01
2017-12-22 21:41:05.791 (zigate) ZigateRead - MsgType 8102 - reception Occupancy Sensor : 01
2017-12-22 21:41:11.162 (zigate) Calling message handler 'onHeartbeat'.
la MajDomoDevice du switch semble ne pas fonctionner chez moi...
Le capteur de temperature et le switch detecter de porte fonctionne très bien...

domopiap, 2017-12-22:
Je seul moyen pour le faire fonctionner c'est de commenter la ligne 879... et là ça roule.

zaraki673, 2017-12-23:
mdr effectivement, joli bug ^^
j essaie de corriger ca tte (j esperes, desolé je bosses surtout sur la future version :( )

handfreezer, 2017-12-24:
En tout cas, merci, c'est bien vu!

commit 343db722 (2017-12-27): v1.0.8

fix #39
fix #31
add water sensor #37
PR #42 (2017-12-27): version 1.0.8
fix #39
fix #31
add water sensor #37
closed 2017-12-27 by zaraki673