zigbeefordomoticz/Domoticz-Zigbee #40: IKEA TRADFRI - remote E1524

handfreezer, 2017-12-26:
Hello,

I tried to pair a IKEA Remote E1524 without success. Below is a link from zigate about this remote, and after the debug log.
Do you think this remote could be managed ? (I guess it is a coordinator and could not?)

http://zigate.fr/ikea-tradfri-telecommande-e1524/ 

017-12-26 21:38:05.268 (zigate) Started.
2017-12-26 21:38:05.438 (zigate) Entering work loop.
2017-12-26 21:38:05.438 (zigate) Initialized version 1.0.7, author 'zaraki673'
2017-12-26 21:38:05.455 (zigate) onStart called
2017-12-26 21:38:05.455 (zigate) Debug log level set to: 'true'.
2017-12-26 21:38:05.455 (zigate) Protocol set to: 'None'.
2017-12-26 21:38:05.455 (zigate) Transport set to: 'Serial', '/dev/ttyUSBzigate0', 115200.
2017-12-26 21:38:05.458 (zigate) Connect directive received, action initiated successfully.
2017-12-26 21:38:05.458 (zigate) Calling message handler 'onConnect'.
2017-12-26 21:38:05.458 (zigate) onConnect called
2017-12-26 21:38:05.458 (zigate) Connected successfully to: /dev/ttyUSBzigate0
2017-12-26 21:38:05.458 (zigate) ZigateDecode - Encodind data : 0021
2017-12-26 21:38:05.459 (zigate) Transcode in : 0021 / out :021021
2017-12-26 21:38:05.459 (zigate) ZigateDecode - Encodind data : 0004
2017-12-26 21:38:05.459 (zigate) Transcode in : 0004 / out :02100214
2017-12-26 21:38:05.459 (zigate) getChecksum - Checksum : 0x2d
2017-12-26 21:38:05.459 (zigate) ZigateDecode - Encodind data : 00000800
2017-12-26 21:38:05.460 (zigate) Transcode in : 00000800 / out :0210021002180210
2017-12-26 21:38:05.460 (zigate) sendZigateCmd - Comand send : 01021021021002142d021002100218021003
2017-12-26 21:38:05.460 (zigate) ZigateDecode - Encodind data : 0023
2017-12-26 21:38:05.460 (zigate) Transcode in : 0023 / out :021023
2017-12-26 21:38:05.460 (zigate) ZigateDecode - Encodind data : 0001
2017-12-26 21:38:05.460 (zigate) Transcode in : 0001 / out :02100211
2017-12-26 21:38:05.460 (zigate) getChecksum - Checksum : 0x22
2017-12-26 21:38:05.461 (zigate) ZigateDecode - Encodind data : 00
2017-12-26 21:38:05.461 (zigate) Transcode in : 00 / out :0210
2017-12-26 21:38:05.461 (zigate) sendZigateCmd - Comand send : 010210230210021122021003
2017-12-26 21:38:05.461 (zigate) ZigateDecode - Encodind data : 0024
2017-12-26 21:38:05.461 (zigate) Transcode in : 0024 / out :021024
2017-12-26 21:38:05.461 (zigate) ZigateDecode - Encodind data : 0000
2017-12-26 21:38:05.461 (zigate) Transcode in : 0000 / out :02100210
2017-12-26 21:38:05.462 (zigate) getChecksum - Checksum : 0x24
2017-12-26 21:38:05.462 (zigate) sendZigateCmd - Comand send : 01021024021002102403
2017-12-26 21:38:05.462 (zigate) ZigateDecode - Encodind data : 0049
2017-12-26 21:38:05.462 (zigate) Transcode in : 0049 / out :021049
2017-12-26 21:38:05.462 (zigate) ZigateDecode - Encodind data : 0004
2017-12-26 21:38:05.462 (zigate) Transcode in : 0004 / out :02100214
2017-12-26 21:38:05.463 (zigate) getChecksum - Checksum : 0xb0
2017-12-26 21:38:05.463 (zigate) ZigateDecode - Encodind data : FFFCFE00
2017-12-26 21:38:05.463 (zigate) Transcode in : FFFCFE00 / out :FFFCFE0210
2017-12-26 21:38:05.463 (zigate) sendZigateCmd - Comand send : 0102104902100214b0FFFCFE021003
2017-12-26 21:38:05.463 (zigate) Sending 18 bytes of data:
2017-12-26 21:38:05.463 (zigate) 01 02 10 21 02 10 02 14 2d 02 10 02 10 02 18 02 10 03 .. .. ...!....-.........
2017-12-26 21:38:05.464 (zigate) Sending 12 bytes of data:
2017-12-26 21:38:05.464 (zigate) 01 02 10 23 02 10 02 11 22 02 10 03 .. .. .. .. .. .. .. .. ...#...."...
2017-12-26 21:38:05.539 (zigate) Sending 10 bytes of data:
2017-12-26 21:38:05.539 (zigate) 01 02 10 24 02 10 02 10 24 03 .. .. .. .. .. .. .. .. .. .. ...$....$.
2017-12-26 21:38:05.614 (zigate) Sending 15 bytes of data:
2017-12-26 21:38:05.614 (zigate) 01 02 10 49 02 10 02 14 b0 ff fc fe 02 10 03 .. .. .. .. .. ...I....????...
2017-12-26 21:38:05.689 (zigate) Received 19 bytes of data:
2017-12-26 21:38:05.690 (zigate) 01 80 02 10 02 10 02 15 a4 02 10 02 10 02 10 21 02 10 03 .. .?......?......!...
2017-12-26 21:38:05.690 (zigate) Received 19 bytes of data:
2017-12-26 21:38:05.690 (zigate) 01 80 02 10 02 10 02 15 a6 02 10 02 10 02 10 23 02 10 03 .. .?......?......#...
2017-12-26 21:38:05.690 (zigate) Received 19 bytes of data:
2017-12-26 21:38:05.690 (zigate) 01 80 02 10 02 10 02 15 a1 02 10 02 10 02 10 24 02 10 03 .. .?......?......$...
2017-12-26 21:38:05.690 (zigate) Calling message handler 'onMessage'.
2017-12-26 21:38:05.690 (zigate) onMessage called
2017-12-26 21:38:05.691 (zigate) ZigateDecode - decodind data : 0180021002100215a402100210021021021003
2017-12-26 21:38:05.691 (zigate) ZigateRead - decoded data : 0180000005a4000000210003
2017-12-26 21:38:05.691 (zigate) ZigateRead - Message Type : 8000, Data : 00000021, RSSI : 00, Length : 0005, Checksum : a4
2017-12-26 21:38:05.691 (zigate) ZigateRead - MsgType 8000 - reception status : Success, SQN : 21, Message :
2017-12-26 21:38:05.691 (zigate) Calling message handler 'onMessage'.
2017-12-26 21:38:05.692 (zigate) onMessage called
2017-12-26 21:38:05.692 (zigate) ZigateDecode - decodind data : 0180021002100215a602100210021023021003
2017-12-26 21:38:05.692 (zigate) ZigateRead - decoded data : 0180000005a6000000230003
2017-12-26 21:38:05.692 (zigate) ZigateRead - Message Type : 8000, Data : 00000023, RSSI : 00, Length : 0005, Checksum : a6
2017-12-26 21:38:05.693 (zigate) ZigateRead - MsgType 8000 - reception status : Success, SQN : 23, Message :
2017-12-26 21:38:05.693 (zigate) Calling message handler 'onMessage'.
2017-12-26 21:38:05.693 (zigate) onMessage called
2017-12-26 21:38:05.693 (zigate) ZigateDecode - decodind data : 0180021002100215a102100210021024021003
2017-12-26 21:38:05.693 (zigate) ZigateRead - decoded data : 0180000005a1000000240003
2017-12-26 21:38:05.694 (zigate) ZigateRead - Message Type : 8000, Data : 00000024, RSSI : 00, Length : 0005, Checksum : a1
2017-12-26 21:38:05.694 (zigate) ZigateRead - MsgType 8000 - reception status : Success, SQN : 24, Message :
2017-12-26 21:38:05.794 (zigate) Received 18 bytes of data:
2017-12-26 21:38:05.794 (zigate) 01 80 02 10 02 10 02 15 e3 02 10 2f 02 10 49 02 10 03 .. .. .?......?../..I...
2017-12-26 21:38:05.794 (zigate) Calling message handler 'onMessage'.
2017-12-26 21:38:05.795 (zigate) onMessage called
2017-12-26 21:38:05.795 (zigate) ZigateDecode - decodind data : 0180021002100215e302102f021049021003
2017-12-26 21:38:05.795 (zigate) ZigateRead - decoded data : 0180000005e3002f00490003
2017-12-26 21:38:05.795 (zigate) ZigateRead - Message Type : 8000, Data : 002f0049, RSSI : 00, Length : 0005, Checksum : e3
2017-12-26 21:38:05.796 (zigate) ZigateRead - MsgType 8000 - reception status : Success, SQN : 49, Message :
2017-12-26 21:38:14.965 (zigate) Calling message handler 'onHeartbeat'.
2017-12-26 21:38:24.957 (zigate) Calling message handler 'onHeartbeat'.
2017-12-26 21:38:34.953 (zigate) Calling message handler 'onHeartbeat'.
2017-12-26 21:38:44.949 (zigate) Calling message handler 'onHeartbeat'.
2017-12-26 21:38:53.768 (zigate) Stop directive received.
2017-12-26 21:38:53.794 (zigate) Disconnect directive received for '/dev/ttyUSBzigate0'.
2017-12-26 21:38:53.947 (zigate) Exiting work loop.
2017-12-26 21:38:58.776 Error: (zigate) Plugin did not stop after 5 seconds, flushing event queue...
2017-12-26 21:38:58.779 (zigate) Stopping threads.
2017-12-26 21:38:58.781 (zigate) Stopped.

zaraki673, 2017-12-27:
Hi,

Your debug log don't speak with your Remote, i can see only normal init
message send to and receive from the zigate ...

deennoo, 2017-12-27:
Remote only speak with bulb, zigate isn't able to sniff it and hear what she says

closed 2018-01-01 by handfreezer