zigbeefordomoticz/Domoticz-Zigbee #41: Registering twice the same device geenrate two different device id

handfreezer, 2017-12-26:
Hi,

Me again... 

The origin is the fact that I could not register aqara temp/hum/baro from xiaomi anymore.
What I notice when testing is the fact that each time I registered the same device, its ID changed, the old one stay present in domoticz but never updated, only the last ID is updated.

Now, what I'm guessing is the fact that I registered a max node count on the zigate or in the plugin. I don't know, do you have any idea?

zaraki673, 2017-12-27:
Max in zigate, a fix will come with a new firmware,

and yes , your right, each time you reset your aqara devices, it will give
you a new short address, and the old one is not use anymore,
for now, i don't check the long address, tha"s why you have a new devices
in domoticz

;)

handfreezer, 2018-01-01:
OK, that's clear.
Is it possible to use tha MAC as device ID in Domoticz?

zaraki673, 2018-01-01:
I will try to put it in domoticz's DB, but not has Domoticz device's ID,
it's too long if i remember true

deennoo, 2018-01-02:
This Can be good, working like Xiaomi, who is able to re set name when à device is re-add After some test

handfreezer, 2018-01-30:
see a devicelist feature in v2.* but failed to use it

zaraki673, 2018-01-30:
Hi,

yes it don't work now :( have make some bug, i will try to fix this soon
(hope find time before end of the week

handfreezer, 2018-01-30:
note : in last version, aqara temp/pressure/humidity sensors shows only in three separates devices. will it be possible to go back in feature (it was good to get one sensor with temp and humidity together ). Maybe, it will be possible to describe this choice in devicelist or deviceconf?

zaraki673, 2018-01-30:
yes will see to add back this feature as soon as all my last change will be
repair ;)

handfreezer, 2018-02-02:
is it cleared in 2.2.2?

zaraki673, 2018-02-02:
yes reopen, not yet clear :p

zaraki673, 2018-03-22:
sorry i m not able tu change deviceid in domoticz
I've try it with version 2.2.7, keep code in but as comment
only way i have it's to delete old one before create a new one :(