zigbeefordomoticz/Domoticz-Zigbee #43: Capteur de temperature qui ne fonctionne plus : erreur de decodage

domopiap, 2018-01-02:
Hello 
est-ce que mon problème est lié à #41 et au nombre de reset que j'ai fait et qui a donc rempli le "compteur" capteur sur la zigate ?
voici le symptôme : 
j'ai 2 capteurs de temperature. L'un fonctionne normalement et l'autre n'est pas reconnue car il semble y a voir une erreur systématique dans le décodage de la trame. 

voici le log du capteur qui fonctionne:

2018-01-02 10:42:11.558 (zigate) Received 10 bytes of data:
2018-01-02 10:42:11.558 (zigate) 01 81 02 12 02 10 32 5e 98 7d .. .. .. .. .. .. .. .. .. .. .�....2^�}
2018-01-02 10:42:11.558 (zigate) Received 13 bytes of data:
2018-01-02 10:42:11.558 (zigate) 97 02 11 02 10 02 10 ff 02 11 02 10 42 .. .. .. .. .. .. .. �......�....B
2018-01-02 10:42:11.559 (zigate) Received 22 bytes of data:
2018-01-02 10:42:11.559 (zigate) 02 10 25 02 11 21 bd 02 1b 02 14 21 a8 33 02 15 21 02 1f 02 ..%..!�....!�3..!...
2018-01-02 10:42:11.560 (zigate) 10 02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..
2018-01-02 10:42:11.560 (zigate) Received 17 bytes of data:
2018-01-02 10:42:11.560 (zigate) 16 24 02 10 02 10 02 10 02 10 02 10 64 29 82 02 19 .. .. .. .$..........d)�..
2018-01-02 10:42:11.560 (zigate) Received 8 bytes of data:
2018-01-02 10:42:11.560 (zigate) 65 21 15 12 66 2b 02 16 .. .. .. .. .. .. .. .. .. .. .. .. e!..f+..
2018-01-02 10:42:11.561 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:42:11.561 (zigate) onMessage called
2018-01-02 10:42:11.565 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:42:11.565 (zigate) onMessage called
2018-01-02 10:42:11.566 (zigate) Received 14 bytes of data:
2018-01-02 10:42:11.566 (zigate) 88 02 11 02 10 02 1a 21 02 10 02 10 48 03 .. .. .. .. .. .. �......!....H.
2018-01-02 10:42:11.567 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:42:11.567 (zigate) onMessage called
2018-01-02 10:42:11.567 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:42:11.568 (zigate) onMessage called
2018-01-02 10:42:11.568 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:42:11.568 (zigate) onMessage called
2018-01-02 10:42:11.569 (zigate) Received 42 bytes of data:
2018-01-02 10:42:11.569 (zigate) 01 81 02 12 02 10 19 a0 99 7d 97 02 11 02 10 02 10 02 10 02 .�.....��}�.........
2018-01-02 10:42:11.569 (zigate) 15 02 10 42 02 10 02 1c 6c 75 6d 69 2e 77 65 61 74 68 65 72 ...B....lumi.weather
2018-01-02 10:42:11.569 (zigate) 48 03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. H.
2018-01-02 10:42:11.569 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:42:11.570 (zigate) onMessage called
2018-01-02 10:42:11.570 (zigate) ZigateDecode - decodind data : 018102120210325e987d97021102100210ff0211021042021025021121bd021b021421a833021521021f021002162402100210021002100210642982021965211512662b02168802110210021a21021002104803
2018-01-02 10:42:11.574 (zigate) ZigateRead - decoded data : 01810200325e987d97010000ff01004200250121bd0b0421a83305210f00062400000000006429820965211512662b068801000a2100004803
2018-01-02 10:42:11.574 (zigate) ZigateRead - Message Type : 8102, Data : 987d97010000ff01004200250121bd0b0421a83305210f00062400000000006429820965211512662b068801000a210000, RSSI : 48, Length : 0032, Checksum : 5e
2018-01-02 10:42:11.575 (zigate) ZigateRead - MsgType 8102 - reception data : 01810200325e987d97010000ff01004200250121bd0b0421a83305210f00062400000000006429820965211512662b068801000a2100004803 ClusterID : 0000 Attribut ID : ff01 Src Addr : 7d97 Scr Ep: 01
2018-01-02 10:42:11.575 (zigate) ZigateRead - MsgType 8102 - reception heartbeat (0000) : 0121bd0b0421a83305210f00062400000000006429820965211512662b068801000a210000
2018-01-02 10:42:11.576 (zigate) ZigateRead - MsgType 8102 - reception batteryLVL (0000) : 100 pour le device addr : 7d97
2018-01-02 10:42:11.576 (zigate) Devices already exist. Unit=16
2018-01-02 10:42:11.577 (zigate) CurrentnValue = 0
2018-01-02 10:42:11.577 (zigate) CurrentsValue = 24.7;46.1;1;1003.6;1003
2018-01-02 10:42:11.577 (zigate) BatteryLvl = 100
2018-01-02 10:42:11.596 (zigate) Devices already exist. Unit=17
2018-01-02 10:42:11.597 (zigate) CurrentnValue = 0
2018-01-02 10:42:11.599 (zigate) CurrentsValue = 24.7;46.1;1
2018-01-02 10:42:11.599 (zigate) BatteryLvl = 100
2018-01-02 10:42:11.600 (zigate - Temp+Hum - 7d97) Updating device from 0:'24.7;46.1;1' to have values 0:'24.7;46.1;1'.
2018-01-02 10:42:11.605 (zigate) Devices already exist. Unit=18
2018-01-02 10:42:11.605 (zigate) CurrentnValue = 0
2018-01-02 10:42:11.605 (zigate) CurrentsValue = 24.7
2018-01-02 10:42:11.605 (zigate) BatteryLvl = 100
2018-01-02 10:42:11.606 (zigate - Temperature - 7d97) Updating device from 0:'24.7' to have values 0:'24.7'.
2018-01-02 10:42:11.610 (zigate) Devices already exist. Unit=19
2018-01-02 10:42:11.611 (zigate) CurrentnValue = 46
2018-01-02 10:42:11.611 (zigate) CurrentsValue = 0
2018-01-02 10:42:11.611 (zigate) BatteryLvl = 100
2018-01-02 10:42:11.612 (zigate - Humidity - 7d97) Updating device from 46:'0' to have values 46:'0'.
2018-01-02 10:42:11.616 (zigate) Devices already exist. Unit=20
2018-01-02 10:42:11.617 (zigate) CurrentnValue = 0
2018-01-02 10:42:11.617 (zigate) CurrentsValue = 1003.6;1003
2018-01-02 10:42:11.617 (zigate) BatteryLvl = 100
2018-01-02 10:42:11.618 (zigate - Barometer - 7d97) Updating device from 0:'1003.6;1003' to have values 0:'1003.6;1003'.
2018-01-02 10:42:11.622 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:42:11.623 (zigate) onMessage called

---------------------------------------------
[... 32 lines cut ...]
2018-01-02 10:43:53.096 (zigate) Received 9 bytes of data:
2018-01-02 10:43:53.096 (zigate) 77 65 61 74 68 65 72 48 03 .. .. .. .. .. .. .. .. .. .. .. weatherH.
2018-01-02 10:43:53.097 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:43:53.097 (zigate) onMessage called
2018-01-02 10:43:53.097 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:43:53.098 (zigate) onMessage called
2018-01-02 10:43:53.098 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:43:53.098 (zigate) onMessage called
2018-01-02 10:43:53.099 (zigate) Calling message handler 'onMessage'.
2018-01-02 10:43:53.099 (zigate) onMessage called
2018-01-02 10:43:53.099 (zigate) ZigateDecode - decodind data : 01810212021019e62b233d021102100210021002150210420210021c6c756d692e776561746865724803
2018-01-02 10:43:53.101 (zigate) ZigateRead - decoded data : 0181020019e62b233d01000000050042000c6c756d692e776561746865724803
2018-01-02 10:43:53.101 (zigate) ZigateRead - Message Type : 8102, Data : 2b233d01000000050042000c6c756d692e77656174686572, RSSI : 48, Length : 0019, Checksum : e6
2018-01-02 10:43:53.102 (zigate) ZigateRead - MsgType 8102 - reception data : 0181020019e62b233d01000000050042000c6c756d692e776561746865724803 ClusterID : 0000 Attribut ID : 0005 Src Addr : 233d Scr Ep: 01
2018-01-02 10:43:53.103 (zigate) ZigateRead - MsgType 8102 - reception heartbeat (0000) : 6c756d692e77656174686572
2018-01-02 10:43:53.103 (zigate) ZigateRead - MsgType 8102 - reception heartbeat (0000) - MsgAttrID (0005) - Type de Device : lumi.weather
2018-01-02 10:43:53.103 (zigate) onMessage - effacement de la trame suite a une erreur de decodage : 01810212021019e62b233d021102100210021002150210420210021c6c756d692e776561746865724803
2018-01-02 10:43:58.123 (zigate) Calling message handler 'onHeartbeat'.

 
 un reset avec appui de 5s sur le petit bouton du capteur amène au meme message d'erreur : "effacement de la trame suite a une erreur de decodage"...

 une idée ?

 merci

zaraki673, 2018-01-02:
je viens de tester de desassocier/réassocier 25 fois de suite un capteur identique au tiens et pas d'erreur :(

domopiap, 2018-01-02:
:(
D'ou peut venir : effacement de la trame suite a une erreur de decodage"...  a t on avis?
Comment peut-on tout effacer sur la zigate et les capteurs pour repartir de zéro?

domopiap, 2018-01-02:
Lorsque tu as fait des 25 essais à chaque fois tu avais un nouveau devis qui remontait dans domoticz avec un nouvel ID?

deennoo, 2018-01-03:
Oui obliger, une inclusion : un nouveau device ID.

Est ce que lorsque tu enlèves le device tu vois bien "leave"?

domopiap, 2018-01-03:
Pour enlever :j'enlève les piles?
Ou le fait de rester appuyer sur le bouton pendant 5s enlève puis le " remet"  avec une autre ID?
(Ces manipes sont bien à faire losque la Led bleu du zigate clignote, n'est ce pas?).
Merci

zaraki673, 2018-01-03:
pour dissocier tu dois rester appuyer 5secondes sur le bouton du capteur, comme pour associer,
si tu le fais pendant que ca clignotte, ton capteur va automatiquement se ré-associer à la zigate, sinon il va juste se desassocier (message de log : "reception Leave indication :" )

domopiap, 2018-01-03:
merci :
donc 
- capteur dissocié, zigate non clignotante :j'appuie sur le bouton = rien dans les log
- zigate clignotante : j'appuie sur le bouton 5s = 
2018-01-03 22:42:17.583 (zigate) ZigateRead - MsgType 8102 - reception data : 018102000e0e0ebaa9010403001400280001ff5a03 ClusterID : 0403 Attribut ID : 0014 Src Addr : baa9 Scr Ep: 01
2018-01-03 22:42:17.584 (zigate) onMessage - effacement de la trame suite a une erreur de decodage : 018102120210021e021e021ebaa902110214021302101402102802100211ff5a03018102120210021ff2021ebaa90211021402130210100210290210021227235a03
ensuite le meme message "effacement...." à chaque fois que j'appuie sur le bouton du capteur.
- un appui sur le bouton du capteur 5s alors que la zingage ne clignote pas = rien ne se passe pas de message : "reception Leave indication"...

je vais refaire une installation propre de domoticz (sur version stable) afin de tirer cela au claire ...est-ce le capteur ou un de mes tests qui a laissé un état pas propre quelque part...

 

- [ ]

zaraki673, 2018-01-03:
reste peut etre appuyer un peu plus de 5secondes tu dois voir ton capteur
clignotter 3fois

domopiap, 2018-01-04:
oui il clignote bien 3 fois... mais meme chose.
Pour la nouvelle installe propre faut-il un domoticz beta ou cela devrait fonctionner avec la version stable 3.8153. Car j'ai le message : "Error: (Zigate) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'"
Au démarrage j'ai : PluginSystem: Started, Python version '3.4.2'. je suis en jessie

zaraki673, 2018-01-04:
non il faut être en bêta obligatoirement :(

sinon essaie un full reset de ta zigate :
[image: Images intégrées 2]

domopiap, 2018-01-04:
oui je pense qu'un full reset serait le bienvenue . je ne vois pas les images que tu as postées... (c'est comment faire un full reset je suppose... ça tombe bien car je ne sais pas faire :))

zaraki673, 2018-01-04:
https://preview.ibb.co/evwTpw/image_1.png

ebretteville, 2018-01-06:
Une petite question, je vois toujours des logs mais rien dans les dispositifs... J'aurai voulu réaliser le reset comme suggéré dans le commentaire d'il ya deux jours avec la capture d'écran mais je ne sais pas quel logiciel est utilisé.

Mes logs pr infos:

`2018-01-06 11:43:57.438 (zigate) Stop directive received.
2018-01-06 11:43:57.479 (zigate) Disconnect directive received for '/dev/ttyUSB1'.
2018-01-06 11:43:57.547 (zigate) Exiting work loop.
2018-01-06 11:43:58.242 (RFLink) Temp + Humidity (RDC)
2018-01-06 11:44:02.445 Error: (zigate) Plugin did not stop after 5 seconds, flushing event queue...
2018-01-06 11:44:02.445 (zigate) Stopping threads.
2018-01-06 11:44:02.445 (zigate) Stopped.
2018-01-06 11:44:02.446 (zigate) Started.
2018-01-06 11:44:02.623 (zigate) Entering work loop.
2018-01-06 11:44:02.623 (zigate) Initialized version 1.1.0, author 'zaraki673'
2018-01-06 11:44:02.627 (zigate) onStart called
2018-01-06 11:44:02.627 (zigate) Debug log level set to: 'true'.
2018-01-06 11:44:02.627 (zigate) Protocol set to: 'None'.
2018-01-06 11:44:02.627 (zigate) Transport set to: 'Serial', '/dev/ttyUSB1', 115200.
2018-01-06 11:44:02.630 (zigate) Connect directive received, action initiated successfully.
2018-01-06 11:44:02.630 (zigate) Calling message handler 'onConnect'.
2018-01-06 11:44:02.630 (zigate) onConnect called
2018-01-06 11:44:02.630 (zigate) Connected successfully
2018-01-06 11:44:02.631 (zigate) getChecksum - Checksum : 0x2d
2018-01-06 11:44:02.631 (zigate) ZigateDecode - Encodind data : 0021
2018-01-06 11:44:02.631 (zigate) Transcode in : 0021 / out :021021
2018-01-06 11:44:02.631 (zigate) ZigateDecode - Encodind data : 0004
2018-01-06 11:44:02.631 (zigate) Transcode in : 0004 / out :02100214
2018-01-06 11:44:02.631 (zigate) ZigateDecode - Encodind data : 00000800
2018-01-06 11:44:02.631 (zigate) Transcode in : 00000800 / out :0210021002180210
2018-01-06 11:44:02.631 (zigate) sendZigateCmd - Comand send : 01021021021002142d021002100218021003
2018-01-06 11:44:02.631 (zigate) getChecksum - Checksum : 0x22
2018-01-06 11:44:02.631 (zigate) ZigateDecode - Encodind data : 0023
2018-01-06 11:44:02.632 (zigate) Transcode in : 0023 / out :021023
2018-01-06 11:44:02.632 (zigate) ZigateDecode - Encodind data : 0001
2018-01-06 11:44:02.632 (zigate) Transcode in : 0001 / out :02100211
2018-01-06 11:44:02.632 (zigate) ZigateDecode - Encodind data : 00
2018-01-06 11:44:02.632 (zigate) Transcode in : 00 / out :0210
2018-01-06 11:44:02.632 (zigate) sendZigateCmd - Comand send : 010210230210021122021003
2018-01-06 11:44:02.632 (zigate) getChecksum - Checksum : 0x24
2018-01-06 11:44:02.632 (zigate) ZigateDecode - Encodind data : 0024
2018-01-06 11:44:02.632 (zigate) Transcode in : 0024 / out :021024
2018-01-06 11:44:02.632 (zigate) ZigateDecode - Encodind data : 0000
2018-01-06 11:44:02.632 (zigate) Transcode in : 0000 / out :02100210
2018-01-06 11:44:02.633 (zigate) sendZigateCmd - Comand send : 01021024021002102403
2018-01-06 11:44:02.633 (zigate) getChecksum - Checksum : 0xb4
2018-01-06 11:44:02.633 (zigate) ZigateDecode - Encodind data : 0049
2018-01-06 11:44:02.633 (zigate) Transcode in : 0049 / out :021049
2018-01-06 11:44:02.633 (zigate) ZigateDecode - Encodind data : 0004
2018-01-06 11:44:02.633 (zigate) Transcode in : 0004 / out :02100214
2018-01-06 11:44:02.633 (zigate) ZigateDecode - Encodind data : FFFCfa00
2018-01-06 11:44:02.633 (zigate) Transcode in : FFFCfa00 / out :FFFCfa0210
2018-01-06 11:44:02.633 (zigate) sendZigateCmd - Comand send : 0102104902100214b4FFFCfa021003
2018-01-06 11:44:02.633 (zigate) Sending 18 bytes of data:
2018-01-06 11:44:02.633 (zigate) 01 02 10 21 02 10 02 14 2d 02 10 02 10 02 18 02 10 03 .. .. ...!....-.........
2018-01-06 11:44:02.634 (zigate) Sending 12 bytes of data:
2018-01-06 11:44:02.634 (zigate) 01 02 10 23 02 10 02 11 22 02 10 03 .. .. .. .. .. .. .. .. ...#...."...
2018-01-06 11:44:02.709 (zigate) Sending 10 bytes of data:
2018-01-06 11:44:02.709 (zigate) 01 02 10 24 02 10 02 10 24 03 .. .. .. .. .. .. .. .. .. .. ...$....$.
2018-01-06 11:44:02.784 (zigate) Sending 15 bytes of data:
2018-01-06 11:44:02.784 (zigate) 01 02 10 49 02 10 02 14 b4 ff fc fa 02 10 03 .. .. .. .. .. ...I....����...
2018-01-06 11:44:02.859 (zigate) Received 16 bytes of data:
2018-01-06 11:44:02.859 (zigate) 10 02 10 02 15 a4 02 10 02 10 02 10 21 02 10 03 .. .. .. .. .....�......!...
2018-01-06 11:44:02.859 (zigate) Received 19 bytes of data:
2018-01-06 11:44:02.859 (zigate) 01 80 02 10 02 10 02 15 a6 02 10 02 10 02 10 23 02 10 03 .. .�......�......#...
2018-01-06 11:44:02.859 (zigate) Received 19 bytes of data:
2018-01-06 11:44:02.859 (zigate) 01 80 02 10 02 10 02 15 a1 02 10 02 10 02 10 24 02 10 03 .. .�......�......$...
2018-01-06 11:44:02.860 (zigate) Calling message handler 'onMessage'.
2018-01-06 11:44:02.860 (zigate) ZigateDecode - decodind data : 1002100215a402100210021021021003
2018-01-06 11:44:02.860 (zigate) ZigateRead - decoded data : 100005a4000000210003
2018-01-06 11:44:02.860 (zigate) ZigateRead - Message Type : 0005, Data : 0021, RSSI : 00, Length : a400, Checksum : 00
2018-01-06 11:44:02.860 (zigate) ZigateRead - Unknow Message Type 0005
2018-01-06 11:44:02.860 (zigate) Calling message handler 'onMessage'.
2018-01-06 11:44:02.860 (zigate) ZigateDecode - decodind data : 0180021002100215a602100210021023021003
2018-01-06 11:44:02.861 (zigate) ZigateRead - decoded data : 0180000005a6000000230003
[... cut ...]
2018-01-06 11:44:04.892 Error: (nono) Call mirobo-wrapper exception: Command '['python3', '/home/pi/domoticz/plugins/xiaomi-mirobot/mirobo-wrapper.py', '192.168.0.85', '5564653571416f317052775743326134', 'status']' returned non-zero exit status 1.
2018-01-06 11:44:08.228 Error: WUnderground: Receiving old data from WU! (No new data return for more than 30 minutes)
2018-01-06 11:44:09.850 (RFLink) Temp + Humidity (Etage)
2018-01-06 11:44:12.164 (zigate) Calling message handler 'onHeartbeat'.
2018-01-06 11:44:18.683 (nono) Calling message handler 'onHeartbeat'.`

zaraki673, 2018-01-06:
Ca c'est juste la phase normal de démarrage du plugin

mets à jour en 1.10, reparamêtres le plugin, tu y verras la fonction full reset, tu fais mettre a jour les parametres du plugin, en activant le full reset, par contre une fois le full reset effectué repasser par la pages paraetrages pour desactivé le full reset, sinon a chaque démarrage il y aura un full reset :P

ebretteville, 2018-01-06:
Ah oui, j'ai bien sélectionné le full reset mais la maintenant il est désactivé, je clique sur modifier pour faire clignoter la led bleu mais elle ne clignote pas, elle est fixe. J'autorise les nouveaux périphériques pendant 5 minutes puis clique sur le bouton de l'un des capteurs, je vois bien des choses passer (date) mais rien dans les dispositifs non-utilisés...

ebretteville, 2018-01-06:
J'ai désactivé le reset puis tenté l'association, voila qqs logs:

`2018-01-06 12:36:40.910 New sensors allowed for 5 minutes...
2018-01-06 12:36:40.912 (zigate) Received 7 bytes of data:
2018-01-06 12:36:40.912 (zigate) 00 00 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .......
2018-01-06 12:36:40.912 (zigate) Received 3 bytes of data:
2018-01-06 12:36:40.912 (zigate) 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ...
2018-01-06 12:36:40.912 (zigate) Received 5 bytes of data:
2018-01-06 12:36:40.912 (zigate) 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .....
2018-01-06 12:36:40.913 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:40.913 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:40.913 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:40.963 (zigate) Received 5 bytes of data:
2018-01-06 12:36:40.963 (zigate) 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .....
2018-01-06 12:36:40.963 (zigate) Received 5 bytes of data:
2018-01-06 12:36:40.964 (zigate) 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .....
2018-01-06 12:36:40.964 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:40.964 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.014 (zigate) Received 1 bytes of data:
2018-01-06 12:36:41.014 (zigate) 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .
2018-01-06 12:36:41.014 (zigate) Received 5 bytes of data:
2018-01-06 12:36:41.014 (zigate) 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .....
2018-01-06 12:36:41.014 (zigate) Received 7 bytes of data:
2018-01-06 12:36:41.015 (zigate) 00 00 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .......
2018-01-06 12:36:41.015 (zigate) Received 4 bytes of data:
2018-01-06 12:36:41.015 (zigate) 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ....
2018-01-06 12:36:41.015 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.015 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.015 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.015 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.066 (zigate) Received 4 bytes of data:
2018-01-06 12:36:41.066 (zigate) 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ....
2018-01-06 12:36:41.066 (zigate) Received 5 bytes of data:
2018-01-06 12:36:41.066 (zigate) 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .....
2018-01-06 12:36:41.066 (zigate) Received 5 bytes of data:
2018-01-06 12:36:41.066 (zigate) 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .....
2018-01-06 12:36:41.066 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.066 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.067 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.117 (zigate) Received 1 bytes of data:
2018-01-06 12:36:41.117 (zigate) 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .
2018-01-06 12:36:41.117 (zigate) Received 1 bytes of data:
2018-01-06 12:36:41.117 (zigate) 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .
2018-01-06 12:36:41.117 (zigate) Received 6 bytes of data:
2018-01-06 12:36:41.117 (zigate) 00 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. ......
2018-01-06 12:36:41.117 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.118 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.118 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.168 (zigate) Received 5 bytes of data:
2018-01-06 12:36:41.168 (zigate) 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .....
2018-01-06 12:36:41.168 (zigate) Received 6 bytes of data:
2018-01-06 12:36:41.168 (zigate) 00 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. ......
2018-01-06 12:36:41.168 (zigate) Received 5 bytes of data:
2018-01-06 12:36:41.168 (zigate) 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .....
2018-01-06 12:36:41.169 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.169 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:41.169 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:45.778 (zigate) Received 7 bytes of data:
2018-01-06 12:36:45.779 (zigate) 00 00 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .......
2018-01-06 12:36:45.779 (zigate) Received 5 bytes of data:
2018-01-06 12:36:45.779 (zigate) 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .....
2018-01-06 12:36:45.779 (zigate) Received 5 bytes of data:
2018-01-06 12:36:45.779 (zigate) 00 00 00 00 00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .....
2018-01-06 12:36:45.779 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:45.779 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:36:45.780 (zigate) Calling message handler 'onMessage'.`

`2018-01-06 12:33:44.478 (zigate) Stopping threads.
2018-01-06 12:33:44.478 (zigate) Stopped.
2018-01-06 12:33:44.478 (zigate) Started.
[... 63 lines cut ...]
2018-01-06 12:33:44.901 (zigate) ZigateRead - Message Type : 0005, Data : 0023, RSSI : 00, Length : a600, Checksum : 00
2018-01-06 12:33:44.901 (zigate) ZigateRead - Unknow Message Type 0005
2018-01-06 12:33:44.901 (zigate) Calling message handler 'onMessage'.
2018-01-06 12:33:44.901 (zigate) ZigateDecode - decodind data : 0180021002100215a102100210021024021003
[... cut ...]
2018-01-06 12:33:49.939 New sensors allowed for 5 minutes...
2018-01-06 12:33:54.173 (zigate) Calling message handler 'onHeartbeat'.
2018-01-06 12:34:04.196 (zigate) Calling message handler 'onHeartbeat'.
2018-01-06 12:34:08.271 (RFLink) Temp + Humidity (RDC)
2018-01-06 12:34:14.168 (zigate) Calling message handler 'onHeartbeat'.
2018-01-06 12:34:19.890 (RFLink) Temp + Humidity (Etage)
2018-01-06 12:34:24.191 (zigate) Calling message handler 'onHeartbeat'.
2018-01-06 12:34:34.213 (zigate) Calling message handler 'onHeartbeat'.`

zaraki673, 2018-01-06:
autoriser les nouveaux péripheriques est inutiles

pourquoi tes logs ne sont pas ds l ordre chronologique ? je suis desolé c
est illisibles comme ca,

exemple de logs lisibles :

2018-01-05 04:42:35.951  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.951  (zig) ZigateDecode - decodind data :
018102120210021f353fcb5402110214021302100210021029021002120213d6e103
2018-01-05 04:42:35.952  (zig) ZigateRead - decoded data :
018102000f353fcb5401040300000029000203d6e103
2018-01-05 04:42:35.952  (zig) ZigateRead - Message Type : 8102, Data :
3fcb5401040300000029000203d6, RSSI : e1, Length : 000f, Checksum : 35
2018-01-05 04:42:35.952  (zig) ZigateRead - MsgType 8102 - reception data :
018102000f353fcb5401040300000029000203d6e103 ClusterID : 0403 Attribut ID :
0000 Src Addr : cb54 Scr Ep: 01
2018-01-05 04:42:35.952  (zig) MajDomoDevice - Device ID : cb54 - Device EP
: 01 - Type : Barometer - Value : 982
2018-01-05 04:42:35.952  (zig) MajDomoDevice baro CurrentsValue :
25.0;44.8;1;982.8;982
2018-01-05 04:42:35.952  (zig) MajDomoDevice bar NewSvalue :
25.0;44.8;1;982;982.8
2018-01-05 04:42:35.952  (zig - Temp+Hum+Baro - cb54) Updating device from
0:'25.0;44.8;1;982.8;982' to have values 0:'25.0;44.8;1;982;982.8'.
2018-01-05 04:42:35.953  (zig) Update 0:'25.0;44.8;1;982;982.8' (zig -
Temp+Hum+Baro - cb54)
2018-01-05 04:42:35.953  (zig) MajDomoDevice baro CurrentsValue : 982.8;982
2018-01-05 04:42:35.953  (zig - Barometer - cb54) Updating device from
0:'982.8;982' to have values 0:'982;982.8'.
2018-01-05 04:42:35.954  (zig) Update 0:'982;982.8' (zig - Barometer - cb54)
2018-01-05 04:42:35.954  (zig) ZigateRead - MsgType 8102 - reception atm :
9.8
2018-01-05 04:42:35.954  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.955  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.956  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.956  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.956  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.956  (zig) Calling message handler 'onMessage'.
2018-01-05 04:42:35.956  (zig) ZigateDecode - decodind data :
018102120210021e02183fcb5402110214021302101402102802100211ffe103
2018-01-05 04:42:35.956  (zig) ZigateRead - decoded data :
018102000e083fcb54010403001400280001ffe103
2018-01-05 04:42:35.956  (zig) ZigateRead - Message Type : 8102, Data :
3fcb54010403001400280001ff, RSSI : e1, Length : 000e, Checksum : 08
2018-01-05 04:42:35.956  (zig) ZigateRead - MsgType 8102 - reception data :
018102000e083fcb54010403001400280001ffe103 ClusterID : 0403 Attribut ID :
0014 Src Addr : cb54 Scr Ep: 01
2018-01-05 04:42:35.956  (zig) onMessage - effacement de la trame suite a
une erreur de decodage :
018102120210021e02183fcb5402110214021302101402102802100211ffe103
2018-01-05 04:42:35.956  (zig) Calling message handler 'onMessage'.

domopiap, 2018-01-06:
Bon en ce qui me concerne.
Nouvelle installation.
installation propre de Domoticz V3.8798
reset total de la zigate avec ZGWUI
association avec ZGWUI de mes capteurs : 2 Capteur de température / humidité / pression atmosphérique , 1 Détecteur ouverture de porte Aqara, 1 Détecteur de mouvement aqara
Plug de la zigate sur mon nouveau domoticz: config du plugin puis ... après une nuit... appuie une fois sur chaque détecteur qui remonte dans domoticz.
L'effacement de la trame suite à une erreur de décodage continue d'être là ...je suppute le décodage de la tension batterie à travers la trame propriétaire de Xiaomi : http://zigate.fr/xiaomi-capteur-temperature-humidite-et-pression-atmospherique-clusters/
clos pour moi
merci ;)

zaraki673, 2018-01-06:
> capteur-temperature-humidite-et-pression-atmospherique-clusters/

sur le baro il y a une trame que je ne sais pas decodé et qui donne ce
message, si tu as le message entre deux Baro c est donc normal, sinon y a
quelques choses qui cloches tout de même

bon pour l association, le mieux est de le faire par domoticz, tous les
capteurs ne renvois pas forcement la trame permettant de les identifier en
dehors de la phase d'association :(
(les anciens model de detecteur d'ouverture d porte/fenetre, il me semble,
j ' ai pas de quoi tester sur moi :p )

closed 2018-01-14 by zaraki673