zigbeefordomoticz/Domoticz-Zigbee #46: Version 1.10 non fonctionnel

max5962, 2018-01-05:
Hello,
Je suis en train de charger la version 1.10 et j'ai cette erreur : 
2018-01-05 17:47:55.993 Error: (ZigateUSB) failed to load 'plugin.py', Python Path used was ':/usr/lib/python35.zip:/usr/lib/python3.5:/usr/lib/python3.5/plat-x86_64-linux-gnu:/usr/lib/python3.5/lib-dynload'.
2018-01-05 17:47:55.993 Error: (zigatePlugin) Module Import failed, exception: 'ImportError'
2018-01-05 17:47:55.993 Error: (zigatePlugin) Module Import failed: ' Name: plugin'
2018-01-05 17:47:55.993 Error: (zigatePlugin) Error Line details not available.
2018-01-05 17:49:00.539 Error: zigatePlugin hardware (7) thread seems to have ended unexpectedly

Que faire :) ? 

merci

zaraki673, 2018-01-05:
passe par la page parametres, il y a eu des modifications de ce coté, du
coup faut reparamétrer et mettre à jour, pas besoin de le supprimer et
ré-associer les devices

max5962, 2018-01-06:
Ça semble fonctionner :)
J'attend les premiers update de device pour clôturer.
Merci

max5962, 2018-01-06:
C'est OK merci :)

closed 2018-01-06 by max5962