zigbeefordomoticz/Domoticz-Zigbee #50: Trames du Cube Xiaomi

tornlouses, 2018-01-12:
Je met les trames que j'obtiens avec les mouvements sur le cube Xiaomi :

Shake :

> 2018-01-11 16:16:55.943  (Zigate) Received 6 bytes of data:
2018-01-11 16:16:55.943  (Zigate)     01 81 02 12 02 10 .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .▒....
2018-01-11 16:16:55.943  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.943  (Zigate)     02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.943  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.943  (Zigate)     1f .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.943  (Zigate) Received 2 bytes of data:
2018-01-11 16:16:55.943  (Zigate)     b5 4b .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ▒K
2018-01-11 16:16:55.944  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.944  (Zigate)     14 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.944  (Zigate) Received 2 bytes of data:
2018-01-11 16:16:55.944  (Zigate)     b1 02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ▒.
2018-01-11 16:16:55.944  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.944  (Zigate)     12 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.944  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.944  (Zigate)     02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.944  (Zigate) Received 2 bytes of data:
2018-01-11 16:16:55.944  (Zigate)     10 12 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..
2018-01-11 16:16:55.944  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.944  (Zigate)     02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.944  (Zigate) Received 2 bytes of data:
2018-01-11 16:16:55.945  (Zigate)     10 55 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .U
2018-01-11 16:16:55.945  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.945  (Zigate)     02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.945  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.945  (Zigate)     10 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.945  (Zigate) Received 2 bytes of data:
2018-01-11 16:16:55.945  (Zigate)     21 02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     !.
2018-01-11 16:16:55.945  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.945  (Zigate)     10 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.945  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.945  (Zigate)     02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.945  (Zigate) Received 2 bytes of data:
2018-01-11 16:16:55.945  (Zigate)     12 02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..
2018-01-11 16:16:55.946  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.946  (Zigate)     10 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.946  (Zigate) Received 2 bytes of data:
2018-01-11 16:16:55.946  (Zigate)     02 10 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..
2018-01-11 16:16:55.946  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.946  (Zigate)     b1 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ▒
2018-01-11 16:16:55.946  (Zigate) Received 1 bytes of data:
2018-01-11 16:16:55.946  (Zigate)     03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-11 16:16:55.946  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.946  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.947  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.947  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.947  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.947  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.947  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.947  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.947  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.947  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.947  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.948  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.948  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.948  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.948  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.948  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.948  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.948  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.948  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.949  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.949  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:16:55.949  (Zigate) ZigateDecode - decodind data : 018102120210021fb54b14b102120210120210550210210210021202100210b103
2018-01-11 16:16:55.949  (Zigate) ZigateRead - decoded data : 018102000fb54b14b10200120055002100020000b103
2018-01-11 16:16:55.949  (Zigate) ZigateRead - Message Type : 8102, Data : 4b14b10200120055002100020000, RSSI : b1, Length : 000f, Checksum : b5
[... 587 lines cut ...]
2018-01-11 16:24:26.699  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.699  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.700  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.700  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.700  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.701  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.701  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.702  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.702  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.702  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.703  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.703  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.704  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.704  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.704  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.705  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.705  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.706  (Zigate) Calling message handler 'onMessage'.
2018-01-11 16:24:26.706  (Zigate) ZigateDecode - decodind data : 018102120210021fac5a14b102120210120210550210210210021202100213ba03
2018-01-11 16:24:26.707  (Zigate) ZigateRead - decoded data : 018102000fac5a14b10200120055002100020003ba03
2018-01-11 16:24:26.707  (Zigate) ZigateRead - Message Type : 8102, Data : 5a14b10200120055002100020003, RSSI : ba, Length : 000f, Checksum : ac
2018-01-11 16:24:26.708  (Zigate) ZigateRead - MsgType 8102 - reception data : 018102000fac5a14b10200120055002100020003ba03 ClusterID : 0012 Attribut ID : 0055 Src Addr : 14b1 Scr Ep: 02
2018-01-11 16:24:26.708  (Zigate) MajDomoDevice - Device ID : 14b1 - Device EP : 02 - Type : Switch - Value : 0003
2018-01-11 16:24:26.709  (Zigate) onMessage - effacement de la trame suite a une erreur de decodage : 018102120210021fac5a14b102120210120210550210210210021202100213ba03
2018-01-11 16:24:27.311  (Zigate) Calling message handler 'onHeartbeat'.

zaraki673, 2018-01-14:
tu peux tester avec la dernière version (2.0.0), une partie devrait déjà fonctionner depuis un moment :(

tornlouses, 2018-01-16:
Salut avec la version 2.0.1 :

Shake :
> 2018-01-16 08:41:10.338  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.338  (Zigate)     01 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-16 08:41:10.338  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.339  (Zigate)     81 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ▒
2018-01-16 08:41:10.339  (Zigate) Received 2 bytes of data:
2018-01-16 08:41:10.339  (Zigate)     02 12 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..
2018-01-16 08:41:10.339  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.339  (Zigate)     02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-16 08:41:10.339  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.340  (Zigate)     10 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-16 08:41:10.340  (Zigate) Received 4 bytes of data:
2018-01-16 08:41:10.340  (Zigate)     02 1f f6 02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..▒.
2018-01-16 08:41:10.340  (Zigate) Received 2 bytes of data:
2018-01-16 08:41:10.340  (Zigate)     16 83 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .▒
2018-01-16 08:41:10.340  (Zigate) Received 2 bytes of data:
2018-01-16 08:41:10.340  (Zigate)     02 17 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..
2018-01-16 08:41:10.341  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.341  (Zigate)     02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-16 08:41:10.341  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.341  (Zigate)     12 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-16 08:41:10.341  (Zigate) Received 2 bytes of data:
2018-01-16 08:41:10.341  (Zigate)     02 10 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..
2018-01-16 08:41:10.342  (Zigate) Received 2 bytes of data:
2018-01-16 08:41:10.342  (Zigate)     12 02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..
2018-01-16 08:41:10.342  (Zigate) Received 2 bytes of data:
2018-01-16 08:41:10.342  (Zigate)     10 55 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .U
2018-01-16 08:41:10.342  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.342  (Zigate)     02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-16 08:41:10.342  (Zigate) Received 2 bytes of data:
2018-01-16 08:41:10.343  (Zigate)     10 21 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .!
2018-01-16 08:41:10.343  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.343  (Zigate)     02 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-16 08:41:10.343  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.343  (Zigate)     10 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-16 08:41:10.343  (Zigate) Received 2 bytes of data:
2018-01-16 08:41:10.343  (Zigate)     02 12 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..
2018-01-16 08:41:10.344  (Zigate) Received 2 bytes of data:
2018-01-16 08:41:10.344  (Zigate)     02 10 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..
2018-01-16 08:41:10.344  (Zigate) Received 2 bytes of data:
2018-01-16 08:41:10.344  (Zigate)     02 12 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ..
2018-01-16 08:41:10.344  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.344  (Zigate)     9c .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ▒
2018-01-16 08:41:10.345  (Zigate) Received 1 bytes of data:
2018-01-16 08:41:10.345  (Zigate)     03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-16 08:41:10.345  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.345  (Zigate) onMessage called
2018-01-16 08:41:10.345  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.346  (Zigate) onMessage called
2018-01-16 08:41:10.346  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.346  (Zigate) onMessage called
2018-01-16 08:41:10.346  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.346  (Zigate) onMessage called
2018-01-16 08:41:10.347  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.347  (Zigate) onMessage called
2018-01-16 08:41:10.347  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.347  (Zigate) onMessage called
2018-01-16 08:41:10.347  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.347  (Zigate) onMessage called
2018-01-16 08:41:10.348  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.348  (Zigate) onMessage called
2018-01-16 08:41:10.348  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.348  (Zigate) onMessage called
2018-01-16 08:41:10.349  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.349  (Zigate) onMessage called
2018-01-16 08:41:10.349  (Zigate) Calling message handler 'onMessage'.
2018-01-16 08:41:10.349  (Zigate) onMessage called
2018-01-16 08:41:10.349  (Zigate) Calling message handler 'onMessage'.
[... 912 lines cut ...]
2018-01-16 09:16:11.059  (Zigate) onMessage called
2018-01-16 09:16:11.059  (Zigate) Calling message handler 'onMessage'.
2018-01-16 09:16:11.059  (Zigate) onMessage called
2018-01-16 09:16:11.060  (Zigate) Calling message handler 'onMessage'.
2018-01-16 09:16:11.060  (Zigate) onMessage called
2018-01-16 09:16:11.060  (Zigate) Calling message handler 'onMessage'.
[... cut ...]
On dirait qu'il met du temps entre deux actions pour pouvoir les comprendre... je veux dire par là que si tu enchaine 2 actions en genre moins d'une minute il comprend que la première mais pas la deuxième. Et pourtant la deuxième il la comprend si tu l'as fait 5 minutes plus tard... normal?

J'avais mis sur le forum un exemple de configuration d'actions en fonction des mouvements du cube... j'ai essayé de nouveau et qu'une seule action a fonctionné.

Je te laisse jugé si c'est fonctionnel et donc fermer le ticket.

zaraki673, 2018-01-27:
il manque beaucoup a decoder sur ce capteur, il y a des trames ep03 qui sont inconnus encore

pour juger si c'est fonctionnel, je ne sais pas, je n'ai pas ce capteur :(
mais au vu de ce que tu me dis ca fonctionne pas :(

deennoo, 2018-01-29:
Pour le decodages des trames du cube voici ce que j'avais commencé a faire : 

`			if DType=="lumi.sensor_cube"   :  # Xiaomi Magic Cube
				if Type==Dtypename :
					if Type=="Switch" :
						if Ep == "02" :
							if value == "0000" : #shake
								state="10"
								data="01"
						
							elif value == "0204" or value == "0200" or value == "0203" or value == "0201" or value == "0202" or value == "0205": #tap
								state="50"
								data="05"
							
							elif value == "0103" or value == "0100" or value == "0104" or value == "0101" or value == "0102" or value == "0105": #Slide
								state="20"
								data="02"
							
							elif value == "0003" : #Free Fall
								state="70"
								data="07"
								
							elif value >= "0004" and value <= "0059": #90°
								state="30"
								data="03"
								
							elif value >= "0060" : #180°
								state="90"
								data="09"
					# elif Type=="Switch" : #rotation
						# elif MsgClusterId=="000c":
							# if Ep == "03" :
								# if value == "40404040" or value=="41414141" or value=="42424242" or value=="43434343" : #clockwise
								  # state="40"
								  # data="04"
						
							# # elif value == "0204" or value == "0200": #tap
								# # state="50"
								# # data="05"
								
							#Devices[x].Update(nValue = int(data),sValue = str(state))
							UpdateDevice(x,int(data),str(state))`

par exemple clockwise et anti clock wise ne donne pas toujours les memes valeurs : 404040 ou 41414141ou 42424242 c'est un peu la merde ce capteur...beaucoup de fonction a gerer, sans compter les alertes et move qui viennent parasiter le tout.

Je passe par zwgui pour avoir les trames.

pipiche38, 2018-08-10:
J'attends un Cube d'ici une 15ene de jour, je pourrais mettre à jour le plugin pour ce type de capteur. D'après ce que j'ai vu sur d'autres forums, il semble que les valeurs envoyées prennent en compte également la face de départ et face d'arrivée. Donc cela est un peu plus complexe.

Il y a également un état 'wakeup'/reveil value = 0002 qu'il faudrait traiter .

Comme signalé #111 le cube via le Xiaomi gateway est implémenté en natif sur domoticz. Je pense qu'il serait donc bien d'aligner les comportements notamment vis-à-vis du switch selector

pipiche38, 2018-09-03:
Close. Implemented in dev banch and will push to master with version 2.4.0

closed 2018-09-03 by pipiche38