zigbeefordomoticz/Domoticz-Zigbee #58: Erreur suite ajout prise xiaomi

Cortexlegeni, 2018-01-25:
Bonjour,
Domoticz tourne bien avec la Zigate depuis 2 jour et là j'ai voulu rajouter une prise xiaomi et depuis plus aucun capteur ne remonte d'infos:
Voici le log du moment:

[prise.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/1663051/prise.txt)

Merci beaucoup. (je suis disponible si besoin de faire des tests ou autre)
Rémi

Cortexlegeni, 2018-01-25:
J'ai reset la zigate via les reglages du plugins et je n'arrive toujours pas à ajouter la prise.
[ajout prise apres reset zigate.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/1663093/ajout.prise.apres.reset.zigate.txt)

zaraki673, 2018-01-25:
d apres les logs, aucun problème avec une prise mais avec un detecteur d'eau : 'Model': 'lumi.sensor_wleak.aq1'

apres, comme toujours, : 
   quel version du plugin?
   quel version de domoticz?

d'apres la log, la variable DeviceConf, as-tu bien copié tous les fichiers du plugins et non pas juste le plugin.py ?

Cortexlegeni, 2018-01-25:
Bonsoir,
Plugin en version 2.0.1 et domoticz 3.8834
J'ai bien copié tous les fichiers, et tout marchait bien depuis 2 jours alors que le détecteur d'eau était déjà là.
j'avais 2 capteurs de temperature, un detecteur de fuite d'eau, un inter et un detecteur de mouvement, juste pour faire des tests.

zaraki673, 2018-01-25:
bonsoir,

humm, peux tu test avec la version 2.1.0, j ai modifier la fonction GetType, j'esperes que ca solutionne ton problème,

;)

Cortexlegeni, 2018-01-25:
j'ai mis la version 2.1, mis Domoticz à jour, fait un reset via le plugin et j'ai toujours ça dans le log

`
2018-01-25 20:09:59.042 (Zi-test) Calling message handler 'onHeartbeat'.
2018-01-25 20:09:59.043 (Zi-test) ListOfDevices : {'7902': {'RIA': '2', 'Ep': {'01': {'0006': {}, '0000': {}}}, 'IEEE': '00158d0001dbc932', 'MacCapa': '80', 'Status': 'inDB', 'Battery': {}, 'Type': 'Door', 'Model': 'lumi.sensor_magnet.aq2', 'ZDeviceID': {}, 'ProfileID': {}, 'Heartbeat': '201'}, 'e0f9': {'RIA': '2', 'Ep': {'01': {'0006': {}, '0000': {}}}, 'IEEE': '00158d0001bbe9bf', 'MacCapa': '80', 'Status': 'inDB', 'Battery': {}, 'Type': 'Water', 'Model': 'lumi.sensor_wleak.aq1', 'ZDeviceID': {}, 'ProfileID': {}, 'Heartbeat': '201'}, '5a19': {}, '9d3e': {'RIA': '2', 'Ep': {'01': {'0000': {}, '0400': {}, '0406': {}}}, 'IEEE': '00158d0001a33a9b', 'MacCapa': '80', 'Status': 'inDB', 'Battery': {}, 'Type': 'Motion/Lux', 'Model': 'lumi.sensor_motion.aq2', 'ZDeviceID': {}, 'ProfileID': {}, 'Heartbeat': '0'}}
2018-01-25 20:09:59.043 Error: (Zi-test) 'onHeartbeat' failed 'KeyError'.
2018-01-25 20:09:59.043 Error: (Zi-test) ----> Line 286 in /home/pi/domoticz/plugins/zigate/plugin.py, function onHeartbeat
2018-01-25 20:09:59.043 Error: (Zi-test) ----> Line 193 in /home/pi/domoticz/plugins/zigate/plugin.py, function onHeartbeat`

zaraki673, 2018-01-25:
hum redemmarre le plugin ou domoticz et vide le fichier devicelist.txt (il doit y avoir que {} )
il y a eu un probleme sur un device, juste l adresse c est ajouter et pas le contenu par defaut, (pas normal :( )

Cortexlegeni, 2018-01-25:
le fichier DeviceList.txt est bien vide ( juste les accolades),
mais je n'ai pas encore viré les devices de Domoticz par contre. je le fais.

Cortexlegeni, 2018-01-25:
[Apres reboot plugin et device supprime.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/1665807/Apres.reboot.plugin.et.device.supprime.txt)

Après un reboot du plugin et le device supprimé de domoticz

Cortexlegeni, 2018-01-25:
Bilan je n'arrive plus non plus à ajouter de nouveaux devices dans Domoticz, le reset du plugin ne semble pas suffire.
Dois-je tenter un reset via l'appli windows ou je laisse comme cela pour pouvoir te donner les résultats des essais que tu veux que je réalise ? (je ne suis pas pressé, je fais des tests, et je vais sans doute commander une deuxième Zigate pour tester un peu plus loin)

zaraki673, 2018-01-25:
essaie de supprimer le plugin de domoticz, normalement apres avoir tout supprimer tu devrais avoir dans les logs : 
(Zi-test) ListOfDevices : {}
alors que dans ta derniere log je vois encore ton detecteur d'ouverture xiaomi et detecteur de mouvement et un device adresse '5a19': {} qui fou le bordel :( 

(je vais regarder pour mettre une exception dans ce cas la )

Cortexlegeni, 2018-01-25:
c'est fait, j'arrive à remettre des dispositifs mais toujours pas ma prise Xiaomi.

[apres full plugin remove.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/1665945/apres.full.plugin.remove.txt)

Cortexlegeni, 2018-01-25:
je te dis ça dès que tu commit.

Cortexlegeni, 2018-01-25:
j'ai encore quelques minutes avant de me coucher, et si tu as besoin d'autres choses n'hésites pas. Je commence à avoir un bon nombre de capteurs xiaomi si jamais tu as besoin de cobaïe

zaraki673, 2018-01-25:
c'est commit version 2.1.1 ;)

merci

zaraki673, 2018-01-25:
ca creer le device ds domoticz, tu dois pouvoir la piloté, mais par contre je sais pas encore decodé ces retours d'état (030303)

zaraki673, 2018-01-25:
desolé je viens de repousser, j avais oublier de definir le type pour l envoie de commande :p

zaraki673, 2018-01-25:
pour ce qui est des capteurs xiaomi, jen ai pas mal également, mais test tout ce que tu peux, ca permet detre sur que tout fonctionne ;)

merci

Cortexlegeni, 2018-01-25:
ok, j'avais pris la version intermédiaire et je n'arrivais plus a activer la detection/ajout.

zaraki673, 2018-01-25:
oui j ai fais une erreur (meme sur le dernier commit), je corrige ca vite

zaraki673, 2018-01-25:
dsl

Cortexlegeni, 2018-01-25:
ok, je me disais aussi, car même là ça ne fonctionnait pas.
Faut que je passe un peu de temps à analyser ton code pour mettre le nez dedans aussi.

zaraki673, 2018-01-25:
c'est bon cette fois, desole

zaraki673, 2018-01-25:
il est pas tres compliqué tu veras, je ne suis loin d etre expert python, du coup il y a bcp de raccourci (ou optimisation) que je ne fais :(

Cortexlegeni, 2018-01-25:
ok je teste.
je ne suis pas expert Python non plus ne t'inquiètes pas ;)

Cortexlegeni, 2018-01-25:
elle ne veut toujours pas venir ma prise (j'espère que je ne l'ai pas flinguée.)
voici deux logs.
[test ajout 1.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/1666079/test.ajout.1.txt)
[test ajout 1 et 2.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/1666080/test.ajout.1.et.2.txt)

Cortexlegeni, 2018-01-25:
ok, j'ai vu, tu n'as pas mis le code d'ajout du plug dans le domoDevice, je verrai demain si j'ai le temps car là je vais me coucher, bonne nuit.

zaraki673, 2018-01-25:
humm etrange il je voit un plug qui devrait etre ds les devices (id 5294)

ok, bonne nuit,

pour demain : 

tu va dans hardware, tu effaces le plugin,
tu verifis que le fichier devicelist.txt est bien vide (enfin juste {} )
tu rajoutes le plugin en mettant la fonction reset a true
tu attends quelques secondes
tu selectionnes le plugin dans la liste,tu mets reset PD a false puis update

et a partir de la tu reassocie le plug (juste lui pour le moment)

Cortexlegeni, 2018-01-25:
il n'apparait pas dans les devices.
je teste ce que tu viens de dire et te réponds, après j'irai au lit. (j'aime pas laisser un truc pas fini)

Cortexlegeni, 2018-01-25:
j'ai suivi ta liste, ça fonctionne, voilà le rapport si besoin.
[plug ok.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/1666102/plug.ok.txt)

zaraki673, 2018-01-25:
merci :D 

du coup un nouveau bug, j ai deux trames vu comme en reponse de l envoie d'une commande 

merci pour tous ces test , bonne nuit ;)

Cortexlegeni, 2018-01-26:
tu ne remontes pas la conso par contre ? je ne sais pas si ce modèle le fait ?
bonne nuit.

zaraki673, 2018-01-26:
non pas encore, c'est prevue, un peu plus tard :D

Cortexlegeni, 2018-01-26:
J'ai regardé pour la remontée de la conso de la prise, mais je me heurte à quelques soucis, je n'ai que très peu de valeurs dans le champs de retour. J'ai aussi modifié la ligne de calcul de la longeur dans le send, car il ne passait pas en hexa:
length=returnlen(4,(str(hex(int(round(len(datas)/2)))).split('x')[-1]))

Tous les cas testés ne renvoient dans le champ de conso que 4 fois le même octet : 00, 40, 41 ou 42 quelque soit la charge

zaraki673, 2018-01-26:
pour la conso tu envoi un readattribute pour avoir ces valeurs ? (tu passes
par le soft windows por tester ?) (cf :
http://zigate.fr/xiaomi-power-socket-prise-pilotee-clusters/ )

pour la longueur du send je pensais l avoir corrigé :(, je vais checker ca
et mettre ta solution si c est pas fait :) )

:

Cortexlegeni, 2018-01-26:
J'ai testé la conso via ton plugin et via l'appli windows et à chaque fois, les 4 octets qui doivent représenter la conso sont identiques et me donnent des valeurs du type 40, 41 ou 42, ce qui donne:
21:42:26.528 <- 01 81 02 00 11 04 CF 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 9F 03
(donc que des 0 en l'absence de courant.

et sur ton plugin : 2018-01-26 14:15:04.344 (Zi-Test) Received 82 bytes of data:
2018-01-26 14:15:04.344 (Zi-Test) 01 80 02 10 02 10 02 15 9a 02 10 1e 02 11 02 10 02 10 03 01 .�......�...........
2018-01-26 14:15:04.344 (Zi-Test) 81 02 12 02 10 11 e1 1e 83 23 02 12 02 10 02 1c 02 10 55 02 �.....�.�#........U.
2018-01-26 14:15:04.344 (Zi-Test) 10 39 02 10 02 14 41 41 41 41 ab 03 01 81 02 12 02 10 02 1d .9....AAAA�..�......
2018-01-26 14:15:04.344 (Zi-Test) ec 1e 83 23 02 12 02 10 02 1c 02 10 02 10 86 ff 02 10 02 10 �.�#..........��....
2018-01-26 14:15:04.345 (Zi-Test) ab 03 .
donc 41 41 41 41 avec une petite charge, un grosse charge ,ne change rien

Cortexlegeni, 2018-01-26:
22:59:33.255 <- 01 81 02 00 11 7F 53 83 23 02 00 0C 00 55 00 39 00 04 44 44 44 44 78 03
1012w de conso d'aprés une seconde prise branchée en série et reliée au home xiaomi.

zaraki673, 2018-01-26:
bon bah va falloir la chercher la trame, car ca doit pas etre celle ci :(,
a voir si y en a pas d autres, je v regarder pour ajouter le decodage des
ce cluster cette nuit pour que tu puisses tester demain ;)

Cortexlegeni, 2018-01-26:
t'embètes pas trop, je ne suis pas chez moi du week end.
Le truc c'est qu'une fois que tu as fais la demande, la prise t'envoie de façon périodique la trame de réponse, donc elle doit bien servir à quelque chose.
Sinon faudrait refabriquer un sniffer comme akila pour tester ça, mais j'ai pas trop le temps de me lancer là dedans en ce moment, J'ai des malades à la maison.

Cortexlegeni, 2018-01-26:
en fait elle balance cette trame tous les 15 minutes, alors va savoir ce qu'elle contient.
23:18:45.401 <- 01 81 02 00 11 E2 2A 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 9C 03
23:19:01.422 <- 01 81 02 00 11 F3 3B 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 9C 03
23:19:17.384 <- 01 81 02 00 11 85 4D 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 9C 03
23:19:33.397 <- 01 81 02 00 11 99 5E 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 93 03
23:19:49.382 <- 01 81 02 00 11 A8 6F 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 93 03
23:20:05.414 <- 01 81 02 00 11 47 80 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 93 03
23:20:21.390 <- 01 81 02 00 11 5A 92 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 9C 03
23:20:37.393 <- 01 81 02 00 11 6E A3 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 99 03
23:20:53.400 <- 01 81 02 00 11 79 B4 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 99 03
23:21:09.407 <- 01 81 02 00 11 0B C6 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 99 03
23:21:25.407 <- 01 81 02 00 11 1A D7 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 99 03
23:21:41.429 <- 01 81 02 00 11 21 E9 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 9C 03
23:21:57.413 <- 01 81 02 00 11 3D FA 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 93 03

Cortexlegeni, 2018-01-26:
Quand tu modifie la charge, elle renvoie bien cette trame:
23:22:45.377 <- 01 81 02 00 11 E6 2E 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 9C 03
23:22:56.991 <- 01 81 02 00 11 F3 3B 83 23 02 00 0C 00 55 00 39 00 04 44 44 44 44 9C 03
23:23:08.038 <- 01 81 02 00 11 84 49 83 23 02 00 0C 00 55 00 39 00 04 44 44 44 44 99 03
23:23:10.820 <- 01 81 02 00 11 86 4B 83 23 02 00 0C 00 55 00 39 00 04 43 43 43 43 99 03
23:23:35.402 <- 01 81 02 00 11 AE 65 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 9F 03
23:23:51.408 <- 01 81 02 00 11 BD 76 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 9F 03
23:24:07.391 <- 01 81 02 00 11 43 88 83 23 02 00 0C 00 55 00 39 00 04 00 00 00 00 9F 03

Cortexlegeni, 2018-01-26:
je ne sais pas trop comment faire pour trouver la bonne trame de façon simple.

zaraki673, 2018-01-26:
va alloir passer la zigate en mode sniffer et regarder ce qui passe entre
la prise et la gateway ...

ca va pas etre simple quoi :)

Cortexlegeni, 2018-01-30:
Bonjour,
Comment est-ce que je passe la zigate en sniffer ?
Merci

zaraki673, 2018-01-30:
il faut lui charger un autre firmware
cf le site zigate.fr (dsl pas acces depuis le taf pour mettre le lien :( )

Cortexlegeni, 2018-01-30:
j'ai regardé le lien vers la prise Salus, cela semble plus simple vu qu'il propose l'accès au cluster 0x0702 qui est le standard pour la gestion d'energie.
Je regarderai pour le sniffer ce soir aussi car au boulot je n'ai pas pris ma centrale, je n'ai que la clé et la prise commandée.

deennoo, 2018-01-30:
Voici tout les elements de la salus sur le ep : 09

Input Cluster Count: 9
    Cluster 0:  Cluster ID: 0x0000 (General: Basic)
    Cluster 1:  Cluster ID: 0x0001 (General: Power Config)
    Cluster 2:  Cluster ID: 0x0003 (General: Identify)
    Cluster 3:  Cluster ID: 0x0004 (General: Groups)
    Cluster 4:  Cluster ID: 0x0005 (General: Scenes)
    Cluster 5:  Cluster ID: 0x0006 (General: On/Off)
    Cluster 6:  Cluster ID: 0x0402 (Measurement: Temperature)
    Cluster 7:  Cluster ID: 0x0702 (Smart Energy: Metering)
    Cluster 8:  Cluster ID: 0xFC01 (Unknown)

Cortexlegeni, 2018-01-30:
OK, il doit falloir taper dans le clusters 0x0702 pour cette prise la, mais sur la xiaomi, je ne trouve pas ce clusters, ni le 0x0B04 qui semble être l'autre qui gère la conso. Il n'y a que le 0x0064 qui me semble étrange mais je vais essayer de pousser si j'arrive a sniffer le zigbee, pour le moment je peine avec ubiqua.

Cortexlegeni, 2018-02-14:
J'ai bien avancé sur la conso, j'ai quasi bouclé le truc, comment puis-je t'envoyer ça (ya le plugin ainsi que le fichier texte de modifié)

zaraki673, 2018-02-14:
soit via github, tu pull sur la branch dev2
soit tu copies colles ta modification ici et j'ajouterais sur git ;)

zaraki673, 2018-04-10:
Salut, je comprends pas ta modif, le on/off fonctionnait déjà sur la salus,
dans ton code tu utilises le EP 9 et le cluster 0005, pourquoi le 0005 ? c est celui pour modifier une scene et non faire on/off qui est le 0006

deennoo, 2018-04-12:
Elle ne fonctionnait pas dans les version precedente avant le comit.

Tu as raison erreur de ma part c'est bien le cluster 0006 pour le On/Off et le 0005 pour les scene on/off (en fait les deux fonctionnes tres bien).

Reste maintenant le smart energy metering : 
Dans un premier temps, il sera pas mal de créer un widget TEXT ou tu viens ecrires la valeur du cluster, ainsi on peut avoir un historique de ce que y est envoyer et essayer des decodages

zaraki673, 2018-04-13:
ok, merci, fix en 2.3.5

pour le smart energy, te fais ca ds la prochaine version ;)

pipiche38, 2018-08-22:
le cluster 0x000c est aussi utilisé par le cub Xiaomi pour remonter des informations de type "Rotation". Je viens de pousser une modif sur la branche "developement", je suis interessé sur les retour, notamment si je n'ai rien cassé coté Power Meter ;-)

pipiche38, 2018-09-10:
Implementation done 0d1074d4a2d3c4ff747c34cd2b99de8db16fa1bd

commit 8072acaa (2018-01-25): fix verification existance device in db

fix #58
closed 2018-09-10 by pipiche38