zigbeefordomoticz/Domoticz-Zigbee #59: Crash domoticz

Miloune, 2018-01-26:
Bonjour,

J'ai un problème depuis quelques jours avec le plugin, et je pense que c'est lié au capteur ouverture / fermeture de porte.
Ce problème est assez gênant du au fait que mon Domoticz crash complètement.
Ci joint la stacktrace. Si besoin de complément d'informations, n'hésite pas ;)

```
2018-01-26 16:22:02.574  (Zigate) Calling message handler 'onMessage'.
2018-01-26 16:22:02.574  (Zigate) Calling message handler 'onMessage'.
2018-01-26 16:22:02.574  (Zigate) Calling message handler 'onMessage'.
2018-01-26 16:22:02.574  (Zigate) Calling message handler 'onMessage'.
2018-01-26 16:22:02.574  (Zigate) Calling message handler 'onMessage'.
2018-01-26 16:22:02.574  (Zigate) Received 1 bytes of data:
2018-01-26 16:22:02.574  (Zigate)     8a .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ~J
2018-01-26 16:22:02.574  (Zigate) Calling message handler 'onMessage'.
2018-01-26 16:22:02.574  (Zigate) Calling message handler 'onMessage'.
2018-01-26 16:22:02.574  (Zigate) Calling message handler 'onMessage'.
2018-01-26 16:22:02.574  (Zigate) Received 1 bytes of data:
2018-01-26 16:22:02.574  (Zigate)     03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     .
2018-01-26 16:22:02.574  (Zigate) Calling message handler 'onMessage'.
2018-01-26 16:22:02.574  (Zigate) Deallocating connection object 'ZiGate' (/dev/ttyUSB0:).
2018-01-26 16:22:02.677  (Zigate) Calling message handler 'onMessage'.
2018-01-26 16:22:02.677  (Zigate) ZigateDecode - decodind data : 018102120210021e19f0557f02110214021302101402102802100211ff8a03
2018-01-26 16:22:02.677  (Zigate) ZigateRead - decoded data : 018102000e19f0557f010403001400280001ff8a03
2018-01-26 16:22:02.677  Error: Domoticz received fatal signal 11 !...
2018-01-26 16:22:02.678  Error: Domoticz received fatal signal 11 while backtracing !...
2018-01-26 16:22:02.690  Error:   /root/domoticz/domoticz() [0x6119a8]
2018-01-26 16:22:02.690  Error:   /root/domoticz/domoticz(_Z14signal_handleri+0x4f) [0x611a5f]
2018-01-26 16:22:02.690  Error:   /lib/x86_64-linux-gnu/libpthread.so.0(+0xf890) [0x7f943c36a890]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(+0x76bdf) [0x7f9437c47bdf]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(+0x76fc4) [0x7f9437c47fc4]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(+0x1b0581) [0x7f9437d81581]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(PyUnicode_Append+0x455) [0x7f9437d29c55]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(+0x5f55b) [0x7f9437c3055b]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(PyEval_EvalFrameEx+0x673d) [0x7f9437dbbaed]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(PyEval_EvalFrameEx+0x685c) [0x7f9437dbbc0c]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(PyEval_EvalFrameEx+0x685c) [0x7f9437dbbc0c]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(PyEval_EvalFrameEx+0x685c) [0x7f9437dbbc0c]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(PyEval_EvalCodeEx+0x883) [0x7f9437dbdaa3]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(+0xd8ff9) [0x7f9437ca9ff9]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(PyObject_Call+0x68) [0x7f9437ce0cd8]
2018-01-26 16:22:02.690  Error:   /usr/lib/x86_64-linux-gnu/libpython3.4m.so(PyEval_CallObjectWithKeywords+0x47) [0x7f9437db4787]
2018-01-26 16:22:02.690  Error:   /root/domoticz/domoticz(_ZN7Plugins7CPlugin8CallbackESsPv+0xb9) [0xa03ca9]
2018-01-26 16:22:02.690  Error:   /root/domoticz/domoticz(_ZN7Plugins13CCallbackBase8CallbackEP7_object+0x3b) [0xa0d8cb]
2018-01-26 16:22:02.690  Error:   /root/domoticz/domoticz(_ZN7Plugins17onMessageCallback13ProcessLockedEv+0x39) [0xa1d8e9]
2018-01-26 16:22:02.690  Error:   /root/domoticz/domoticz(_ZN7Plugins13CCallbackBase7ProcessEv+0x17) [0xa0dd87]
2018-01-26 16:22:02.690  Error:   /root/domoticz/domoticz(_ZN7Plugins13CPluginSystem7Do_WorkEv+0x5b3) [0xa12883]
2018-01-26 16:22:02.690  Error:   /root/domoticz/domoticz() [0xc6922a]
2018-01-26 16:22:02.690  Error:   /lib/x86_64-linux-gnu/libpthread.so.0(+0x8064) [0x7f943c363064]
2018-01-26 16:22:02.690  Error:   /lib/x86_64-linux-gnu/libc.so.6(clone+0x6d) [0x7f943bb9362d]

```

Merci d'avance !

zaraki673, 2018-01-26:
bonjour,

a chaque reception de ce capteur ?

Miloune, 2018-01-26:
Non, c'est assez aléatoire, mais j'ai l'impression que c'est depuis que j'ai connecté celui là à ma Zigate, après ce n'est peut être pas lié.

zaraki673, 2018-01-26:
je penses pas :(, mais c'est bien la partie plugin python qui plante, t as
d autres plugin python activé ?

impossible de savoir si c est un probleme du plugin ou de domoticz (meme si
je penses plus au second)

en quel version tu es ?

ebretteville, 2018-01-26:
Moi perso, dès que je mets le plugin avec tous ses fichiers, domoticz ne se lance pas. Dès que je renomme le plugin avec un .pyth plus de soucis. En gros depuis la version 2. Je suis sur la dernière bêta d'hier. Raspbian Jessie. En revanche python 2.7.4 est me python principal. J'ai essayé de passé sur la branche 3.6 mais soit je me suis planté soit ya un soucis. Juste mon retour. Je penses passer mon rasp domoticz sous stretch. Et lettre direct la ZiGate et le plug-in pour discriminer tout autre problème.

Miloune, 2018-01-26:
J'ai un autre plugin en python : xiaomi-mirobot.
Version: 3.8850
Build Hash: 3e46d9cd
Compile Date: 2018-01-25 23:08:58
dzVents Version: 2.4.1

J'ai mis à jour ce matin pour voir si ça pouvait résoudre mon problème, mais en vain.

zaraki673, 2018-01-26:
voir : https://github.com/sasu-drooz/Domoticz-Zigate/issues/34
surement un problème systeme qui pertube domoticz

closed 2018-02-02 by zaraki673