zigbeefordomoticz/Domoticz-Zigbee #60: Ampoule IKEA LED1622G12 qui ne remonte pas dans Domoticz

domopiap, 2018-01-27:
Hello
Domoticz 3.8798 Plugin 'Version':'2.1.1' friemxare Zigate 3.0b

Je fais un reset factory mais rien ne remonte comme device dans domptiez voici les log:

il y a bien les: MsgType 004d - Reception Device announce avec avant listOfDevice vide et ensuite :{'1e9a': {'Ep': {}, 'IEEE': '000b57fffe993f2b', 'Battery': {}, 'Model': {}, 'MacCapa': '8e', 'Status': '004d', 'RIA': '0', 'Heartbeat': '0', 'ZDeviceID': {}, 'Type': {}, 'ProfileID': {}}}

merci

---------------------------------------------------------------------------------
### 2018-01-27 10:35:15.259 (Zigate) Calling message handler 'onHeartbeat'.
2018-01-27 10:35:15.260 (Zigate) ListOfDevices : {}
2018-01-27 10:35:15.260 (Zigate) HB count = 5
2018-01-27 10:35:25.290 (Zigate) Calling message handler 'onHeartbeat'.
2018-01-27 10:35:25.291 (Zigate) ListOfDevices : {}
2018-01-27 10:35:25.291 (Zigate) HB count = 6
2018-01-27 10:35:29.805 (Zigate) Received 21 bytes of data:
2018-01-27 10:35:29.805 (Zigate) 01 02 10 4d 02 10 02 1c e9 1e 9a 02 10 02 1b 57 ff fe 99 3f ...M....�.�....W���?
2018-01-27 10:35:29.805 (Zigate) 2b .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. +
2018-01-27 10:35:29.805 (Zigate) Received 3 bytes of data:
2018-01-27 10:35:29.805 (Zigate) 8e 72 03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. �r.
2018-01-27 10:35:29.806 (Zigate) Calling message handler 'onMessage'.
2018-01-27 10:35:29.806 (Zigate) onMessage called
2018-01-27 10:35:29.807 (Zigate) Calling message handler 'onMessage'.
2018-01-27 10:35:29.807 (Zigate) onMessage called
2018-01-27 10:35:29.807 (Zigate) ZigateDecode - decodind data : 0180021002100215a602100210021023021003
2018-01-27 10:35:29.808 (Zigate) ZigateRead - decoded data : 0180000005a6000000230003
2018-01-27 10:35:29.809 (Zigate) ZigateRead - Message Type : 8000, Data : 00000023, RSSI : 00, Length : 0005, Checksum : a6
2018-01-27 10:35:29.809 (Zigate) ZigateRead - MsgType 8000 - reception status : 0180000005a6000000230003
2018-01-27 10:35:29.809 (Zigate) Decode8000 - Reception status : Success, SQN : 23, Message :
2018-01-27 10:35:29.810 (Zigate) ZigateDecode - decodind data : 0180021002100215a1021002100210240210030180021002100215da0210160210490210030102104d0210021ce91e9a0210021b57fffe993f2b8e7203
2018-01-27 10:35:29.812 (Zigate) ZigateRead - decoded data : 0180000005a10000002400030180000005da00160049000301004d000ce91e9a000b57fffe993f2b8e7203
2018-01-27 10:35:29.813 (Zigate) ZigateRead - Message Type : 8000, Data : 0000002400030180000005da00160049000301004d000ce91e9a000b57fffe993f2b8e, RSSI : 72, Length : 0005, Checksum : a1
2018-01-27 10:35:29.813 (Zigate) ZigateRead - MsgType 8000 - reception status : 0180000005a10000002400030180000005da00160049000301004d000ce91e9a000b57fffe993f2b8e7203
2018-01-27 10:35:29.813 (Zigate) Decode8000 - Reception status : Success, SQN : 24, Message :
2018-01-27 10:35:35.283 (Zigate) Calling message handler 'onHeartbeat'.
2018-01-27 10:35:35.283 (Zigate) ListOfDevices : {}
2018-01-27 10:35:35.283 (Zigate) HB count = 7
2018-01-27 10:35:35.635 (Zigate) Received 24 bytes of data:
2018-01-27 10:35:35.635 (Zigate) 01 02 10 4d 02 10 02 1c e9 1e 9a 02 10 02 1b 57 ff fe 99 3f ...M....�.�....W���?
2018-01-27 10:35:35.635 (Zigate) 2b 8e 72 03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. +�r.
2018-01-27 10:35:35.635 (Zigate) Calling message handler 'onMessage'.
2018-01-27 10:35:35.636 (Zigate) onMessage called
2018-01-27 10:35:35.636 (Zigate) ZigateDecode - decodind data : 0102104d0210021ce91e9a0210021b57fffe993f2b8e7203
2018-01-27 10:35:35.637 (Zigate) ZigateRead - decoded data : 01004d000ce91e9a000b57fffe993f2b8e7203
2018-01-27 10:35:35.637 (Zigate) ZigateRead - Message Type : 004d, Data : 1e9a000b57fffe993f2b8e, RSSI : 72, Length : 000c, Checksum : e9
2018-01-27 10:35:35.638 (Zigate) ZigateRead - MsgType 004d - Reception Device announce : 01004d000ce91e9a000b57fffe993f2b8e7203
2018-01-27 10:35:35.638 (Zigate) Decode004d - Reception Device announce : Source :1e9a, IEEE : 000b57fffe993f2b, Mac capa : 8e
2018-01-27 10:35:35.689 (Zigate) Received 24 bytes of data:
2018-01-27 10:35:35.689 (Zigate) 01 02 10 4d 02 10 02 1c e3 1e 9a 02 10 02 1b 57 ff fe 99 3f ...M....�.�....W���?
2018-01-27 10:35:35.689 (Zigate) 2b 8e 78 03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. +�x.
2018-01-27 10:35:35.689 (Zigate) Received 15 bytes of data:
2018-01-27 10:35:35.689 (Zigate) 01 87 02 11 02 10 02 13 fd 02 10 02 10 78 03 .. .. .. .. .. .�......�....x.
2018-01-27 10:35:35.690 (Zigate) Calling message handler 'onMessage'.
2018-01-27 10:35:35.690 (Zigate) onMessage called
2018-01-27 10:35:35.690 (Zigate) ZigateDecode - decodind data : 0102104d0210021ce31e9a0210021b57fffe993f2b8e7803
2018-01-27 10:35:35.692 (Zigate) ZigateRead - decoded data : 01004d000ce31e9a000b57fffe993f2b8e7803
2018-01-27 10:35:35.692 (Zigate) ZigateRead - Message Type : 004d, Data : 1e9a000b57fffe993f2b8e, RSSI : 78, Length : 000c, Checksum : e3
2018-01-27 10:35:35.692 (Zigate) ZigateRead - MsgType 004d - Reception Device announce : 01004d000ce31e9a000b57fffe993f2b8e7803
2018-01-27 10:35:35.693 (Zigate) Decode004d - Reception Device announce : Source :1e9a, IEEE : 000b57fffe993f2b, Mac capa : 8e
2018-01-27 10:35:35.693 (Zigate) Calling message handler 'onMessage'.
2018-01-27 10:35:35.693 (Zigate) onMessage called
2018-01-27 10:35:35.694 (Zigate) ZigateDecode - decodind data : 0187021102100213fd021002107803
2018-01-27 10:35:35.694 (Zigate) ZigateRead - decoded data : 0187010003fd00007803
2018-01-27 10:35:35.695 (Zigate) ZigateRead - Message Type : 8701, Data : 0000, RSSI : 78, Length : 0003, Checksum : fd
2018-01-27 10:35:35.695 (Zigate) ZigateRead - MsgType 8701 - Reception Router discovery confirm : 0187010003fd00007803
2018-01-27 10:35:45.275 (Zigate) Calling message handler 'onHeartbeat'.
2018-01-27 10:35:45.276 (Zigate) ListOfDevices : {'1e9a': {'Ep': {}, 'IEEE': '000b57fffe993f2b', 'Battery': {}, 'Model': {}, 'MacCapa': '8e', 'Status': '004d', 'RIA': '0', 'Heartbeat': '0', 'ZDeviceID': {}, 'Type': {}, 'ProfileID': {}}}
2018-01-27 10:35:45.276 (Zigate) sendZigateCmd - length is : 0002
2018-01-27 10:35:45.277 (Zigate) getChecksum - Checksum : 0xc3
[... 127 lines cut ...]
2018-01-27 10:36:35.445 (Zigate) Decode8043 - Reception Simple descriptor response : In Cluster Count : 08
2018-01-27 10:36:35.446 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0000
2018-01-27 10:36:35.446 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0003
2018-01-27 10:36:35.446 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0004
2018-01-27 10:36:35.447 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0005
2018-01-27 10:36:35.447 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0006
2018-01-27 10:36:35.447 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0008
2018-01-27 10:36:35.448 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0b05
2018-01-27 10:36:35.448 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 1000
2018-01-27 10:36:35.448 (Zigate) Decode8043 - Reception Simple descriptor response : Out Cluster Count : 04
2018-01-27 10:36:35.449 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0000
2018-01-27 10:36:35.449 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0003
2018-01-27 10:36:35.449 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0004
2018-01-27 10:36:35.450 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0005
2018-01-27 10:36:35.450 (Zigate) ZigateDecode - decodind data : 0180021002100215dc02101a021043021003
2018-01-27 10:36:35.451 (Zigate) ZigateRead - decoded data : 0180000005dc001a00430003
2018-01-27 10:36:35.451 (Zigate) ZigateRead - Message Type : 8000, Data : 001a0043, RSSI : 00, Length : 0005, Checksum : dc
2018-01-27 10:36:35.452 (Zigate) ZigateRead - MsgType 8000 - reception status : 0180000005dc001a00430003
2018-01-27 10:36:35.452 (Zigate) Decode8000 - Reception status : Success, SQN : 43, Message :
2018-01-27 10:36:45.291 (Zigate) Calling message handler 'onHeartbeat'.
2018-01-27 10:36:45.292 (Zigate) ListOfDevices : {'1e9a': {'Ep': {'01': {'0000': {}, '0003': {}, '0008': {}, '0b05': {}, '1000': {}, '0006': {}, '0004': {}, '0005': {}}}, 'IEEE': '000b57fffe993f2b', 'Battery': {}, 'Model': {}, 'MacCapa': '8e', 'Status': '8043', 'RIA': '0', 'Heartbeat': '0', 'ZDeviceID' [... cut ...]
2018-01-27 10:36:45.293 (Zigate) HB count = 14
2018-01-27 10:36:55.323 (Zigate) Calling message handler 'onHeartbeat'.
2018-01-27 10:36:55.324 (Zigate) ListOfDevices : {'1e9a': {'Ep': {'01': {'0000': {}, '0003': {}, '0008': {}, '0b05': {}, '1000': {}, '0006': {}, '0004': {}, '0005': {}}}, 'IEEE': '000b57fffe993f2b', 'Battery': {}, 'Model': {}, 'MacCapa': '8e', 'Status': '8043', 'RIA': '0', 'Heartbeat': '1', 'ZDeviceID': '0100', 'Type': {}, 'ProfileID': '0104'}}
2018-01-27 10:36:55.324 (Zigate) HB count = 15

zaraki673, 2018-01-27:
Bonjour,

elle ne fait que du On/Off ? ou dimmable et changement de couleur ?

merci ;)

domopiap, 2018-01-27:
elle doit juste faire on/off dimmable.

zaraki673, 2018-01-27:
oki, nickel merci je vais rajouter ca ;)

deennoo, 2018-01-29:
j'ai aussi cette ampoule, je vais un test avec la version 2.1.1 mais toujours pas fonctionnelle

zaraki673, 2018-01-29:
mes dernieres maj ou tout peter je crois, les devices avec model remontés
lors du device announce ne sont plus creer correctement (ca n'utilise plus
le chier de conf externe)

:(

deennoo, 2018-01-29:
ok compris.

le "chier" device list doit ce remplir egalment ?

zaraki673, 2018-01-29:
oui le devicelist doit se remplir et est checker au demarrage si on le
modifie (avant que le device soit inserer en base, sinon il faut effacer le
device en base puis relancer le plugin pour que le nouveau device soit
creer avec les modif apportées)

les devices se creer en fonctionne uniquement si des trames d'infos sont
recu, pas lisant le deviceconf.txt :(

dans l ensemble ca fonctionne, mais pas comme ca devrait :(
(en mettant les bonnes images et typename de devices ds domoticz)

deennoo, 2018-03-31:
Tu as eu l'occasion de test avec le nouveau FW des ampoules IKEA ? elles apparaisent avec le ZLL maintenant, 

Type: 0x8043 (Simple Descriptor Response)
  SQN: 0x85
  Status: 0x00
  Short Address: 0x9749
  Length: 32
  EndPoint: 0x01
  Profile ID: 0xC05E (ZigBee LL)
  Device ID: 0x0100 (Lighting - On/Off Light)
  Input Cluster Count: 8
    Cluster 0:  Cluster ID: 0x0000 (General: Basic)
    Cluster 1:  Cluster ID: 0x0003 (General: Identify)
    Cluster 2:  Cluster ID: 0x0004 (General: Groups)
    Cluster 3:  Cluster ID: 0x0005 (General: Scenes)
    Cluster 4:  Cluster ID: 0x0006 (General: On/Off)
    Cluster 5:  Cluster ID: 0x0008 (General: Level Control)
    Cluster 6:  Cluster ID: 0x0B05 (Misc: Diagnostics)
    Cluster 7:  Cluster ID: 0x1000 (ZLL: Commissioning)
  Output Cluster Count: 4
    Cluster 0:  Cluster ID: 0x0005 (General: Scenes)
    Cluster 1:  Cluster ID: 0x0019 (General: OTA)
    Cluster 2:  Cluster ID: 0x0020 (General: Poll Control)
    Cluster 3:  Cluster ID: 0x1000 (ZLL: Commissioning)

zaraki673, 2018-03-31:
> je viens de modifier le plugin pour qu'elle remonte avec le Nouveau FW
> ZLL, maintenant j'attend qu'elle apparaisse automatiquement dans le
> DeviceList.txt...cela prend combien de temps ?

il faut éteindre et rallumer électriquement l ampoule pour qu elle s
annonce à nouveau et ca devrait se faire ds la minutes max

zaraki673, 2018-04-12:
normalement c'est bon ?

commit ee19ba7d (2018-01-27): ajout IKEA LED1622G12

fix #60
closed 2018-04-12 by zaraki673