zigbeefordomoticz/Domoticz-Zigbee #61: variateur d’intensité sans fil IKEA (dimmer)

domopiap, 2018-01-27:
Lors de l'association du variateur voici les log si cela peut t'aider ...
je peux avoir la meme chose avec le détecteur de mouvement si tu veux...

Calling message handler 'onHeartbeat'.
2018-01-27 11:13:54.372 (Zigate) ListOfDevices : {}
2018-01-27 11:13:54.372 (Zigate) HB count = 3
2018-01-27 11:13:58.836 (Zigate) Received 11 bytes of data:
2018-01-27 11:13:58.836 (Zigate) 01 02 10 4d 02 10 02 1c fe 90 21 .. .. .. .. .. .. .. .. .. ...M....��!
2018-01-27 11:13:58.836 (Zigate) Received 14 bytes of data:
2018-01-27 11:13:58.837 (Zigate) 02 10 02 1b 57 ff fe 2c 89 02 14 80 72 03 .. .. .. .. .. .. ....W��,�..�r.
2018-01-27 11:13:58.837 (Zigate) Calling message handler 'onMessage'.
2018-01-27 11:13:58.837 (Zigate) onMessage called
2018-01-27 11:13:58.838 (Zigate) Calling message handler 'onMessage'.
2018-01-27 11:13:58.838 (Zigate) onMessage called
2018-01-27 11:13:58.838 (Zigate) ZigateDecode - decodind data : 0102104d0210021cfe90210210021b57fffe2c890214807203
2018-01-27 11:13:58.839 (Zigate) ZigateRead - decoded data : 01004d000cfe9021000b57fffe2c8904807203
2018-01-27 11:13:58.840 (Zigate) ZigateRead - Message Type : 004d, Data : 9021000b57fffe2c890480, RSSI : 72, Length : 000c, Checksum : fe
2018-01-27 11:13:58.840 (Zigate) ZigateRead - MsgType 004d - Reception Device announce : 01004d000cfe9021000b57fffe2c8904807203
2018-01-27 11:13:58.840 (Zigate) Decode004d - Reception Device announce : Source :9021, IEEE : 000b57fffe2c8904, Mac capa : 80
2018-01-27 11:13:58.941 (Zigate) Received 17 bytes of data:
2018-01-27 11:13:58.941 (Zigate) 01 02 10 4d 02 10 02 1c db 90 21 02 10 02 1b 57 ff .. .. .. ...M....ې!....W�
2018-01-27 11:13:58.941 (Zigate) Received 8 bytes of data:
2018-01-27 11:13:58.942 (Zigate) fe 2c 89 02 14 80 57 03 .. .. .. .. .. .. .. .. .. .. .. .. �,�..�W.
2018-01-27 11:13:58.942 (Zigate) Received 15 bytes of data:
2018-01-27 11:13:58.942 (Zigate) 01 87 02 11 02 10 02 13 d2 02 10 02 10 57 03 .. .. .. .. .. .�......�....W.
2018-01-27 11:13:58.942 (Zigate) Calling message handler 'onMessage'.
2018-01-27 11:13:58.943 (Zigate) onMessage called
2018-01-27 11:13:58.943 (Zigate) Calling message handler 'onMessage'.
2018-01-27 11:13:58.943 (Zigate) onMessage called
2018-01-27 11:13:58.944 (Zigate) ZigateDecode - decodind data : 0102104d0210021cdb90210210021b57fffe2c890214805703
2018-01-27 11:13:58.945 (Zigate) ZigateRead - decoded data : 01004d000cdb9021000b57fffe2c8904805703
2018-01-27 11:13:58.945 (Zigate) ZigateRead - Message Type : 004d, Data : 9021000b57fffe2c890480, RSSI : 57, Length : 000c, Checksum : db
2018-01-27 11:13:58.945 (Zigate) ZigateRead - MsgType 004d - Reception Device announce : 01004d000cdb9021000b57fffe2c8904805703
2018-01-27 11:13:58.946 (Zigate) Decode004d - Reception Device announce : Source :9021, IEEE : 000b57fffe2c8904, Mac capa : 80
2018-01-27 11:13:58.946 (Zigate) Calling message handler 'onMessage'.
2018-01-27 11:13:58.946 (Zigate) onMessage called
2018-01-27 11:13:58.947 (Zigate) ZigateDecode - decodind data : 0187021102100213d2021002105703
2018-01-27 11:13:58.947 (Zigate) ZigateRead - decoded data : 0187010003d200005703
2018-01-27 11:13:58.948 (Zigate) ZigateRead - Message Type : 8701, Data : 0000, RSSI : 57, Length : 0003, Checksum : d2
2018-01-27 11:13:58.948 (Zigate) ZigateRead - MsgType 8701 - Reception Router discovery confirm : 0187010003d200005703
2018-01-27 11:14:04.327 (Zigate) Calling message handler 'onHeartbeat'.
2018-01-27 11:14:04.328 (Zigate) ListOfDevices : {'9021': {'ZDeviceID': {}, 'MacCapa': '80', 'Ep': {}, 'IEEE': '000b57fffe2c8904', 'Type': {}, 'ProfileID': {}, 'Model': {}, 'RIA': '0', 'Heartbeat': '0', 'Battery': {}, 'Status': '004d'}}
2018-01-27 11:14:04.329 (Zigate) sendZigateCmd - length is : 0002
2018-01-27 11:14:04.329 (Zigate) getChecksum - Checksum : 0xf6
2018-01-27 11:14:04.330 (Zigate) ZigateDecode - Encodind data : 0045
2018-01-27 11:14:04.330 (Zigate) Transcode in : 0045 / out :021045
2018-01-27 11:14:04.330 (Zigate) ZigateDecode - Encodind data : 0002
2018-01-27 11:14:04.331 (Zigate) Transcode in : 0002 / out :02100212
2018-01-27 11:14:04.331 (Zigate) ZigateDecode - Encodind data : f6
2018-01-27 11:14:04.331 (Zigate) Transcode in : f6 / out :f6
2018-01-27 11:14:04.332 (Zigate) ZigateDecode - Encodind data : 9021
2018-01-27 11:14:04.332 (Zigate) Transcode in : 9021 / out :9021
2018-01-27 11:14:04.332 (Zigate) sendZigateCmd - Comand send : 0102104502100212f6902103
2018-01-27 11:14:04.333 (Zigate) HB count = 4
2018-01-27 11:14:04.333 (Zigate) Sending 12 bytes of data:
2018-01-27 11:14:04.333 (Zigate) 01 02 10 45 02 10 02 12 f6 90 21 03 .. .. .. .. .. .. .. .. ...E....��!.
2018-01-27 11:14:04.434 (Zigate) Received 19 bytes of data:
2018-01-27 11:14:04.435 (Zigate) 01 80 02 10 02 10 02 15 c2 02 10 02 12 02 10 45 02 10 03 .. .�......�......E...
2018-01-27 11:14:04.435 (Zigate) Calling message handler 'onMessage'.
2018-01-27 11:14:04.435 (Zigate) onMessage called
2018-01-27 11:14:04.436 (Zigate) ZigateDecode - decodind data : 0180021002100215c202100212021045021003
2018-01-27 11:14:04.436 (Zigate) ZigateRead - decoded data : 0180000005c2000200450003
2018-01-27 11:14:04.437 (Zigate) ZigateRead - Message Type : 8000, Data : 00020045, RSSI : 00, Length : 0005, Checksum : c2
2018-01-27 11:14:04.437 (Zigate) ZigateRead - MsgType 8000 - reception status : 0180000005c2000200450003
2018-01-27 11:14:04.437 (Zigate) Decode8000 - Reception status : Success, SQN : 45, Message :
2018-01-27 11:14:05.140 (Zigate) Received 16 bytes of data:
2018-01-27 11:14:05.140 (Zigate) 01 80 45 02 10 02 17 33 02 12 02 10 90 21 02 11 .. .. .. .. .�E....3....�!..
2018-01-27 11:14:05.140 (Zigate) Received 4 bytes of data:
2018-01-27 11:14:05.141 (Zigate) 02 11 42 03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..B.
2018-01-27 11:14:05.141 (Zigate) Calling message handler 'onMessage'.
[... 49 lines cut ...]
2018-01-27 11:14:16.507 (Zigate) ZigateRead - decoded data : 018043002676030090212001c05e0810020600000001000300090b0510000600000001000300090b0510008403
2018-01-27 11:14:16.507 (Zigate) ZigateRead - Message Type : 8043, Data : 030090212001c05e0810020600000001000300090b0510000600000001000300090b051000, RSSI : 84, Length : 0026, Checksum : 76
2018-01-27 11:14:16.508 (Zigate) ZigateRead - MsgType 8043 - Reception Simple descriptor response 018043002676030090212001c05e0810020600000001000300090b0510000600000001000300090b0510008403
2018-01-27 11:14:16.508 (Zigate) Decode8043 - Reception Simple descriptor response : SQN : 03, Status : 00, short Addr : 9021, Lenght : 20
2018-01-27 11:14:16.508 (Zigate) Decode8043 - Reception Simple descriptor response : EP : 01, Profile : c05e, Device Id : 0810, Bit Field : 02
2018-01-27 11:14:16.508 (Zigate) Decode8043 - Reception Simple descriptor response : In Cluster Count : 06
2018-01-27 11:14:16.509 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0000
2018-01-27 11:14:16.509 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0001
2018-01-27 11:14:16.509 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0003
2018-01-27 11:14:16.510 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0009
2018-01-27 11:14:16.510 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 0b05
2018-01-27 11:14:16.511 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster in: 1000
2018-01-27 11:14:16.512 (Zigate) Decode8043 - Reception Simple descriptor response : Out Cluster Count : 06
2018-01-27 11:14:16.512 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0000
2018-01-27 11:14:16.512 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0001
2018-01-27 11:14:16.513 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0003
2018-01-27 11:14:16.513 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0009
2018-01-27 11:14:16.513 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 0b05
2018-01-27 11:14:16.514 (Zigate) Decode8043 - Reception Simple descriptor response : Cluster out: 1000
2018-01-27 11:14:24.343 (Zigate) Calling message handler 'onHeartbeat'.
2018-01-27 11:14:24.344 (Zigate) ListOfDevices : {'9021': {'ZDeviceID': '0810', 'MacCapa': '80', 'Ep': {'01': {'0001': {}, '0000': {}, '0b05': {}, '1000': {}, '0009': {}, '0003': {}}}, 'IEEE': '000b57fffe2c8904', 'Type': {}, 'ProfileID': 'c05e', 'Model': {}, 'RIA': '0', 'Heartbeat': '0', 'Battery': {}, 'Status': '8043'}}
2018-01-27 11:14:24.344 (Zigate) HB count = 6
2018-01-27 11:14:34.374 (Zigate) Calling message handler 'onHeartbeat'.
2018-01-27 11:14:34.375 (Zigate) ListOfDevices : {'9021': {'ZDeviceID': '0810', 'MacCapa': '80', 'Ep': {'01': {'0001': {}, '0000': {}, '0b05': {}, '1000': {}, '0009': {}, '0003': {}}}, 'IEEE': '000b57fffe2c8904', 'Type': {}, 'ProfileID': 'c05e', 'Model': {}, 'RIA': '0', 'Heartbeat': '1', 'Battery': {}, 'Status': '8043'}}
2018-01-27 11:14:34.375 (Zigate) HB count = 7

zaraki673, 2018-01-27:
nickel, merci

c est quel variateur ? (le beau blanc ? )

domopiap, 2018-01-27:
c'est le petit rond 
 http://www.ikea.com/fr/fr/catalog/products/80356911/

zaraki673, 2018-01-27:
hummm cool, je vais l'ajouter aussi  :D

normalement si tu l action une dizaine de fois, il devrait apparaître ds
domoticz (en mode pseudo forcé :p )

domopiap, 2018-01-27:
l'actionne ...tu veux dire quoi ?

J'ai celui la aussi si tu veux : 
http://www.ikea.com/fr/fr/catalog/products/30338849/

zaraki673, 2018-01-27:
> en tout cas les Ikea LED1545G12, 980lm, dimmable, couleur chaud/froid
> fonctionnent impec et sont reconnues direct.

si celui la (d inter) t arrives a avoir quelques choses ... il s annonce,
me renvois ces ep et cluster (de temps en temps) mais jamais les trames qd
je l actionnes :(

zaraki673, 2018-02-03:
hum tu peux m envoyer des logs qd tu actionnes ton switch ? (on/off et variation)
aucun des clusters listés n'est actuellement décodé :(

domopiap, 2018-02-03:
voici les log : démarrage avec aucun device et reset et association du variateur Ikea bouton jaune...puis après plus rien lorsque je bouge le bouton....

pour ton info :

2018-02-03 21:48:02.034 (zigate) Initialized version 2.2.3, author 'zaraki673'
2018-02-03 21:48:02.034 (zigate) Entering work loop.
2018-02-03 21:48:02.092 (zigate) onStart called
2018-02-03 21:48:02.092 (zigate) Debug log level set to: 'true'.
2018-02-03 21:48:02.092 (zigate) 'DomoticzVersion':'3.8798'
2018-02-03 21:48:02.092 (zigate) 'Name':'zigate'
2018-02-03 21:48:02.093 (zigate) 'HardwareID':'5'
2018-02-03 21:48:02.093 (zigate) 'Mode5':'11'
2018-02-03 21:48:02.093 (zigate) 'Mode1':'USB'
2018-02-03 21:48:02.093 (zigate) 'Address':'0.0.0.0'
2018-02-03 21:48:02.093 (zigate) 'DomoticzBuildTime':'2018-01-02 02:17:32'
2018-02-03 21:48:02.094 (zigate) 'Mode3':'False'
2018-02-03 21:48:02.094 (zigate) 'Author':'zaraki673'
2018-02-03 21:48:02.094 (zigate) 'Port':'9999'
2018-02-03 21:48:02.094 (zigate) 'HomeFolder':'/home/pi/domoticz/plugins/Zigate/'
2018-02-03 21:48:02.094 (zigate) 'Version':'2.2.3'
2018-02-03 21:48:02.095 (zigate) 'Mode2':'254'
2018-02-03 21:48:02.095 (zigate) 'DomoticzHash':'8bb0420c'
2018-02-03 21:48:02.095 (zigate) 'SerialPort':'/dev/ttyUSB0'
2018-02-03 21:48:02.095 (zigate) 'Mode6':'Debug'
2018-02-03 21:48:02.095 (zigate) 'Key':'Zigate'
2018-02-03 21:48:02.096 (zigate) Device count: 0
2018-02-03 21:48:02.097 (zigate) DeviceConf.txt = {'lumi.weather':{'Ep':{'01':{'0402','0405','0403'}},'Type':'Temp/Humi/Baro'},'lumi.sensor_ht':{'Ep':{'01':{'0402','0405'}},'Type':'Temp/Humi'},'lumi.sensor_magnet.aq2':{'Ep':{'01':{'0006'}},'Type':'Door'},'lumi.sensor_motion':{'Ep':{'01':{'0006'}},'Type':'Motion'},'lumi.sensor_switch.aq2':{'Ep':{'01':{'0006'}},'Type':'Switch'},'lumi.sensor_switch':{'Ep':{'01':{'0006'}},'Type':'MSwitch'},'lumi.sensor_86sw2':{'Ep':{'01':{'0006'},'02':{'0006'}},'Type':'DSwitch'},'lumi.sensor_smoke':{'Ep':{'01':{'0006'}},'Type':'Smoke'},'lumi.sensor_wleak.aq1':{'Ep':{'01':{'0006'}},'Type':'Water'},'lumi.sensor_86sw1':{'Ep':{'01':{'0006'}},'Type':'Switch'},'lumi.sensor_motion.aq2':{'Ep':{'01':{'0406','0400'}},'Type':'Motion/Lux'},'lumi.sensor_cube':{'Ep':{'01':{'0012','000c'}},'Type':'XCube'},'lumi.plug':{'Ep':{'01':{'0006'}},'Type':'Plug'},'Ampoule.LED1545G12.Tradfri':{'Ep':{'01':{'0006','0008','0300'}},'Type':'Switch/LvlControl/ColorControl'},'Ampoule.LED1622G12.Tradfri':{'Ep':{'01':{'0006','0008'}},'Type':'Switch/LvlControl'},'Ampoule.phillips.hue':{'Ep':{'01':{'0006','0008'}},'Type':'Switch/LvlControl'},'plug.Osram':{'Ep':{'03':{'0006'}},'Type':'Plug'}}
2018-02-03 21:48:02.105 (zigate) DeviceList.txt open
2018-02-03 21:48:02.106 Error: (zigate) 'onStart' failed 'ValueError'.
2018-02-03 21:48:02.106 Error: (zigate) ----> Line 278 in /home/pi/domoticz/plugins/Zigate/plugin.py, function onStart
2018-02-03 21:48:02.106 Error: (zigate) ----> Line 83 in /home/pi/domoticz/plugins/Zigate/plugin.py, function onStart
2018-02-03 21:48:02.106 (zigate) Protocol set to: 'None'.
2018-02-03 21:48:02.106 (zigate) Transport set to: 'Serial', '/dev/ttyUSB0', 115200.
2018-02-03 21:48:02.110 (zigate) Connect directive received, action initiated successfully.
2018-02-03 21:48:02.110 (zigate) Calling message handler 'onConnect'.
2018-02-03 21:48:02.110 (zigate) onConnect called
2018-02-03 21:48:02.110 (zigate) Connected successfully
2018-02-03 21:48:02.111 (zigate) sendZigateCmd - length is : 0004
2018-02-03 21:48:02.111 (zigate) getChecksum - Checksum : 0x2e
2018-02-03 21:48:02.111 (zigate) ZigateEncode - Encodind data : 0021
2018-02-03 21:48:02.112 (zigate) Transcode in : 0021 / out :021021
2018-02-03 21:48:02.112 (zigate) ZigateEncode - Encodind data : 0004
2018-02-03 21:48:02.112 (zigate) Transcode in : 0004 / out :02100214
2018-02-03 21:48:02.112 (zigate) ZigateEncode - Encodind data : 2e
2018-02-03 21:48:02.113 (zigate) Transcode in : 2e / out :2e
2018-02-03 21:48:02.113 (zigate) ZigateEncode - Encodind data : 00000b00
2018-02-03 21:48:02.113 (zigate) Transcode in : 00000b00 / out :02100210021b0210
2018-02-03 21:48:02.113 (zigate) sendZigateCmd - Comand send : 01021021021002142e02100210021b021003
2018-02-03 21:48:02.114 (zigate) sendZigateCmd - length is : 0001
2018-02-03 21:48:02.114 (zigate) getChecksum - Checksum : 0x22
2018-02-03 21:48:02.114 (zigate) ZigateEncode - Encodind data : 0023
2018-02-03 21:48:02.115 (zigate) Transcode in : 0023 / out :021023
2018-02-03 21:48:02.115 (zigate) ZigateEncode - Encodind data : 0001
2018-02-03 21:48:02.115 (zigate) Transcode in : 0001 / out :02100211
2018-02-03 21:48:02.115 (zigate) ZigateEncode - Encodind data : 22
2018-02-03 21:48:02.116 (zigate) Transcode in : 22 / out :22
2018-02-03 21:48:02.116 (zigate) ZigateEncode - Encodind data : 00
2018-02-03 21:48:02.116 (zigate) Transcode in : 00 / out :0210
2018-02-03 21:48:02.116 (zigate) sendZigateCmd - Comand send : 010210230210021122021003
2018-02-03 21:48:02.117 (zigate) getChecksum - Checksum : 0x24
2018-02-03 21:48:02.117 (zigate) ZigateEncode - Encodind data : 0024
2018-02-03 21:48:02.117 (zigate) Transcode in : 0024 / out :021024
2018-02-03 21:48:02.117 (zigate) ZigateEncode - Encodind data : 0000
2018-02-03 21:48:02.118 (zigate) Transcode in : 0000 / out :02100210
2018-02-03 21:48:02.118 (zigate) ZigateEncode - Encodind data : 24
2018-02-03 21:48:02.118 (zigate) Transcode in : 24 / out :24
2018-02-03 21:48:02.118 (zigate) sendZigateCmd - Comand send : 01021024021002102403
[... cut ...] D': '0810'}}
2018-02-03 21:49:41.601 (zigate) HB count = 9
2018-02-03 21:49:51.583 (zigate) Calling message handler 'onHeartbeat'.
2018-02-03 21:49:51.584 (zigate) ListOfDevices : {'2916': {'Battery': {}, 'Ep': {'01': {'1000': {}, '0b05': {}, '0001': {}, '0009': {}, '0003': {}, '0000': {}}}, 'ProfileID': 'c05e', 'Type': {}, 'MacCapa': '80', 'Status': '8043', 'RIA': '0', 'Model': {}, 'IEEE': '000b57fffe2c8904', 'Heartbeat': '2', 'ZDeviceID': '0810'}}
2018-02-03 21:49:51.585 (zigate) HB count = 10

ebretteville, 2018-02-03:
Quasi la même chose, firmware ZiGate mis à jour sur la 3.0c domoticz à la dernière beta. Je peux redémarrer domoticz et faire une sauvegarde au format texte des logs si nécessaire.

ebretteville, 2018-02-03:
Domoticz Version: 3.8875
Build Hash: 3674fa59
Compile Date: 2018-02-02 16:03:49
dzVents Version: 2.4.1

Zigate: 3.0c

[logs.txt](https://github.com/sasu-drooz/Domoticz-Zigate/files/1692199/logs.txt)

zaraki673, 2018-02-04:
ok, dc stand bye sur ce device, on reçoit que les trames d appairages mais pas de trames de commandes :(