zigbeefordomoticz/Domoticz-Zigbee #62: domoticz v3.8872 not working

handfreezer, 2018-01-30:
Hello,

   I tried 2.2.0, master, and 1.1.0 with domoticz 8872 : joining process seems cool in log, but no device in domoticz.
  I'm on Rpi3.
 I had to fall back to domoticz v3.8805 to get 1.1.0 working, but no success with master.

zaraki673, 2018-01-30:
yes to make it work your device must send data
ex :
for door sensor, you need to open or clse it one time, then you should have
a device type switch in domoticz insteed of door (this is one of the bug)

handfreezer, 2018-01-30:
I'm sorry, even if I don't remember well, I think I have recorded a lot of aqara temp/hum/pressure sensors and seen a lot of trame coming back but only one sensor appeared.

You think I should give a new try? (Do I need to erase EEPROM of my ziagte and register everything again?)

zaraki673, 2018-01-30:
no if you use the last version, it s broken, i think the last good release
was this one
https://github.com/sasu-drooz/Domoticz-Zigate/releases/tag/2.1.0

handfreezer, 2018-01-30:
OK, I'll give a new try this we as I won't be able before...

zaraki673, 2018-02-02:
this should work now ;) with release 2.2.2

zaraki673, 2018-02-02:
need to remove plugin from domoticz db and add it again
don't forget to clear devicelist.txt (or replace with the one from the git)

:D

closed 2018-02-02 by zaraki673

handfreezer, 2018-02-02:
Hello,

Do I need to reset the zigate, or will the plugin add devices on frame received?

zaraki673, 2018-02-02:
no zigate reset,
only plugin in domoticz and files