zigbeefordomoticz/Domoticz-Zigbee #64: Perte du multi Temp/Humi/Baro

cokreyan, 2018-02-04:
Hello,

Sur le sensor lumi.weather (aqara temp/humi/baro), lors de l'inclusion du device, 3 Idx sont créés (1 pour Temp, 1 pour Humi, 1 pour Baro), mais il n'y a plus la création de l'Idx commun Temp+Humi+Baro comme auparavant. Il était bien pratique !

Est-ce volontaire ?
plugin zigate: 2.2.3
domoticz: 3.8875 (derniere beta)

cokreyan, 2018-02-04:
Aussi, le niveau de batterie n'apparait plus sur ces devices.

zaraki673, 2018-02-04:
oui volontaire, pas eu le temps de regarder comment gérer cette exception (je verais si possible de la ré-ajouter, je penses que oui)

pour le niveau de batterie, il devrait bien s'ajouter, mais la trame d'infos n'est pas envoyés très souvent ;)

cokreyan, 2018-02-04:
Effectivement, le niveau de batterie est bien apparu depuis mon post ! :)
Merci pour les infos.

commit 65df5717 (2018-03-21): ajout capteut temp+humi fusionner ds domoticz

fix #64
closed 2018-03-21 by zaraki673