zigbeefordomoticz/Domoticz-Zigbee #65: Mswitch error

zaraki673, 2018-02-06:
par jk77 » 05 févr. 2018, 17:04

Bonjour,

tout d'abord, un grand merci pour ce plugin !!

Juste un bug à signaler avec un switch Xiaomi v1, un MSwitch donc: le double, triple et quadruple click ne fonctionnent plus sur les versions > 2.1.0 

en 2.1.0:
2018-02-05 22:30:07.192 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception General: On/Off : 00
2018-02-05 22:30:07.446 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception General: On/Off : 01
2018-02-05 22:30:12.657 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception General: On/Off : 02
2018-02-05 22:30:15.924 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception General: On/Off : 02
2018-02-05 22:30:52.235 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception General: On/Off : 03
2018-02-05 22:30:58.083 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception General: On/Off : 04

en 2.2.3:
2018-02-05 22:17:17.475 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception General: On/Off : 00
2018-02-05 22:17:17.728 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception General: On/Off : 01
2018-02-05 22:17:31.557 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception heartbeat - Message attribut inconnu : 4bfda301000680000020000102
2018-02-05 22:19:40.178 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception heartbeat - Message attribut inconnu : 4cfda301000680000020000103
2018-02-05 22:19:45.275 (Zigate) ReadCluster (8102) - ClusterId=0006 - reception heartbeat - Message attribut inconnu : 4dfda301000680000020000104

Sinon... tout roule ! Merci encore !

rollyfx, 2018-03-14:
Bonjour,
Je viens de regarder concernant ce problème que j'ai moi aussi, j'ai localisé un soucis sur cette ligne : 
https://github.com/sasu-drooz/Domoticz-Zigate/blob/cc129607d797d38d3e86f02f60fc4c08419894ed/plugin.py#L1100

En fait la variable MsgAttrID lorsqu'on ne fait pas du double click est 8000 et pas 0000 .

En changeant la ligne 1100 par 
`if MsgAttrID=="0000" or MsgAttrID=="8000":`

le switch fonctionnent correctement.
Le truc c'est que je ne sais pas à quoi correspond cette variable et ne trouve pas de doc pour m'aider. 
Je sais pas si c'est le firmware qui devrait retourner 0000 dans le cas d'un double click ou le plugin domoticz qui ne prend pas en compte la valeur 8000 ^^

zaraki673, 2018-03-20:
merci ;)

closed 2018-03-20 by zaraki673