zigbeefordomoticz/Domoticz-Zigbee #67: prb creation device

zaraki673, 2018-02-06:
par domotest7766 » 06 févr. 2018, 20:39

Bonjour,

De mon coté impossible d'ajouter de nouveaux équipements (j'en ai deja ajouté 12 sans pb). J'ai testé avec un device XIAOMI et une ampoule IKEA. Toujours la meme erreur. J'ai la version 2.2.3 et 3.0b pour le firmware.

2018-02-06 20:27:34.472 (Zigbee) GetType - check Type for Cluster : 0006
2018-02-06 20:27:34.472 (Zigbee) GetType - Type will be set to : LvlControl////Switch
2018-02-06 20:27:34.473 (Zigbee) GetType - check Type for Cluster : 0004
2018-02-06 20:27:34.473 (Zigbee) GetType - Type is now set to : LvlControl/Switch/
2018-02-06 20:27:34.473 (Zigbee) CreateDomoDevice - Device ID : 9851 Device EP : 01 Type : LvlControl
2018-02-06 20:27:34.473 (Zigbee) Creating device 'LvlControl - 9851'.
2018-02-06 20:27:34.474 Error: (Zigbee) Device creation failed, Hardware/Unit combination (23:19) already exists in Domoticz.
2018-02-06 20:27:34.474 (Zigbee) CreateDomoDevice - Device ID : 9851 Device EP : 01 Type : Switch
2018-02-06 20:27:34.474 (Zigbee) Creating device 'Switch - 9851'.
2018-02-06 20:27:34.474 Error: (Zigbee) Device creation failed, Hardware/Unit combination (23:19) already exists in Domoticz.
2018-02-06 20:27:34.475 (Zigbee) CreateDomoDevice - Device ID : 9851 Device EP : 01 Type :

buragohf, 2018-03-16:
Hello, i'm getting the same kind of issue. I'm trying to use Xiaomi Aqara temerature sensor.
I succesfully connected 5 sensors but for the others, i receive the following error in domoticz log
"(Zigate) Device creation failed, Hardware/Unit combination (3:16) already exists in Domoticz."
Did i make something wrong?

zaraki673, 2018-03-20:
how many devices have you in domoticz for this plugin ?

buragohf, 2018-03-20:
Hi and thank you for tour Quick reply. I have only 5 devices already
connected

zaraki673, 2018-03-20:
Hi, i see my mistake, i will push new version soon ;)

zaraki673, 2018-03-20:
c'est réglé ;)

it's good now ;)

v2.2.5

closed 2018-03-20 by zaraki673