zigbeefordomoticz/Domoticz-Zigbee #72: Problème avec les switches

Miloune, 2018-03-22:
Je viens de ré-associer mon, périphérique à la zigate, il fonctionnait jusqu'au moment ou je met un "Off delay" de quelques secondes afin que le switch se set à OFF.

Voici les logs : 
```
2018-03-22 19:23:04.690 (Zigate) ZigateDecode - decodind data : 018102120210021ef4103f9d0211021002168002100210200210021102126f03
2018-03-22 19:23:04.690 (Zigate) ZigateRead - decoded data : 018102000ef4103f9d010006800000200001026f03
2018-03-22 19:23:04.690 (Zigate) ZigateRead - Message Type : 8102, Data : 103f9d01000680000020000102, RSSI : 6f, Length : 000e, Checksum : f4
2018-03-22 19:23:04.690 (Zigate) ZigateRead - MsgType 8102 - reception data : 018102000ef4103f9d010006800000200001026f03 ClusterID : 0006 Attribut ID : 8000 Src Addr : 3f9d Scr Ep: 01
2018-03-22 19:23:04.690 (Zigate) MajDomoDevice - Device ID : 3f9d - Device EP : 01 - Type : Switch - Value : 02
2018-03-22 19:23:04.690 (Zigate) onMessage - effacement de la trame suite a une erreur de decodage : 018102120210021ef4103f9d0211021002168002100210200210021102126f03
```

Merci d'avance !

zaraki673, 2018-03-22:
je regarde ca, j ai jamais essayer cette fonctionnalité :p

Miloune, 2018-03-22:
Ça marche merci beaucoup.
A savoir que le problème se reproduit aussi avec le détecteur de mouvement/présence Xiaomi.

zaraki673, 2018-03-22:
pas besoin de le mettre sur les detecteurs de mouvements, ils repassent
auto apres 5sec a off ;)

Miloune, 2018-03-22:
Hmm, chez moi ce n'est pas le cas, il reçoit toujours 1 comme valeur pour
dire qu'il est activé et ne passe jamais à 0. Du coup la date du "last
seen"  est celle de l'ajout du device...

zaraki673, 2018-03-22:
> issuecomment-375456074
> MgKJMG2LDuykSEYTks5thBENgaJpZM4S3x6h

tu es sur quelle version du plugin ?

je viens de pousser une version 2.3.0, les motions fonctionnent très bien
dedans, j ai corrigé une erreur sur l envoi des commandes de ce type de
switch depuis domoticz
et si tu as une erreur tu auras le ligne exact et le message d'erreur

par contre ton erreur est normal (si on peu dire) c'est un bug domoticz, ca
changes les informations contenu dans les options du devices en effaçant le
dictionnaire utilisé par le plugins :(
du coup je peux rien y faire (pas grand choses plutot, j ai mis un patch
qui réactualise ces informations qd la zigate reçois un nouveau status pour
le device, je peux rien faire de plus malheureusement :( )

Miloune, 2018-03-23:
Salut,

J'utilisais la 2.2.4, j'ai fais la MAJ ce matin, j'ai supprimé + reconnecté mon motion, et j'ai toujours le problème, il ne passe jamais à OFF.

Voici un screenshot depuis que je l'ai ré-associé : 
[image]

zaraki673, 2018-03-23:
c est bien un xiaomi ton motion ?

Miloune, 2018-03-23:
Normalement oui, voici une photo : 
[image]

zaraki673, 2018-03-23:
donc oui, un v2, y a pas de raison que ca fonctionne pas, sauf si il n est
pas vu etant que motion au moment de l association avec la domoticz :(

Miloune, 2018-04-03:
Après avoir update ma zigate a la dernière version, reset tous les périphériques, et tout ré-associer, je n'ai plus ce problème

closed 2018-04-03 by Miloune