zigbeefordomoticz/Domoticz-Zigbee #73: aqara swicth

dondipietro, 2018-03-30:
Quand on appuie plusiquers fois sur un interupteur aqara, j'ai cette erreur et je suis obligé de relancer le plugin car l'erreur boucle 

2018-03-30 20:58:58.223 (zigate1) ZigateRead - Message Type : 8102, Data : 336a3601000680000020000103, RSSI : 7b, Length : 000e, Checksum : 3c
2018-03-30 20:58:58.223 (zigate1) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000e3c336a36010006800000200001037b03
2018-03-30 20:58:58.223 (zigate1) Decode8102 - reception data : 03 ClusterID : 0006 Attribut ID : 8000 Src Addr : 6a36 Scr Ep: 01
2018-03-30 20:58:58.223 (zigate1) MajDomoDevice - Device ID : 6a36 - Device EP : 01 - Type : 0006 - Value : 03
2018-03-30 20:58:58.224 Error: (zigate1) 'onMessage' failed 'UnboundLocalError':'local variable 'state' referenced before assignment'.
2018-03-30 20:58:58.224 Error: (zigate1) ----> Line 287 in /home/pi/domoticz/plugins/zigate/plugin.py, function onMessage
2018-03-30 20:58:58.224 Error: (zigate1) ----> Line 115 in /home/pi/domoticz/plugins/zigate/plugin.py, function onMessage
2018-03-30 20:58:58.224 Error: (zigate1) ----> Line 352 in /home/pi/domoticz/plugins/zigate/plugin.py, function ZigateDecode
2018-03-30 20:58:58.224 Error: (zigate1) ----> Line 596 in /home/pi/domoticz/plugins/zigate/plugin.py, function ZigateRead
2018-03-30 20:58:58.224 Error: (zigate1) ----> Line 757 in /home/pi/domoticz/plugins/zigate/plugin.py, function Decode8102
2018-03-30 20:58:58.224 Error: (zigate1) ----> Line 1144 in /home/pi/domoticz/plugins/zigate/plugin.py, function ReadCluster
2018-03-30 20:58:58.224 Error: (zigate1) ----> Line 928 in /home/pi/domoticz/plugins/zigate/plugin.py, function MajDomoDevice
2018-03-30 20:58:58.224 (zigate1) Calling message handler 'onMessage'.
2018-03-30 20:58:58.224 (zigate1) onMessage called
2018-03-30 20:58:58.224 (zigate1) ZigateDecode - decodind data : 018102120210021e3c336a360211021002168002100210200210021102137b03
2018-03-30 20:58:58.225 (zigate1) ZigateRead - decoded data : 018102000e3c336a36010006800000200001037b03
2018-03-30 20:58:58.225 (zigate1) ZigateRead - Message Type : 8102, Data : 336a3601000680000020000103, RSSI : 7b, Length : 000e, Checksum : 3c
2018-03-30 20:58:58.225 (zigate1) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000e3c336a36010006800000200001037b03
2018-03-30 20:58:58.225 (zigate1) Decode8102 - reception data : 03 ClusterID : 0006 Attribut ID : 8000 Src Addr : 6a36 Scr Ep: 01
2018-03-30 20:58:58.225 (zigate1) MajDomoDevice - Device ID : 6a36 - Device EP : 01 - Type : 0006 - Value : 03
2018-03-30 20:58:58.226 Error: (zigate1) 'onMessage' failed 'UnboundLocalError':'local variable 'state' referenced before assignment'.
2018-03-30 20:58:58.226 Error: (zigate1) ----> Line 287 in /home/pi/domoticz/plugins/zigate/plugin.py, function onMessage
2018-03-30 20:58:58.226 Error: (zigate1) ----> Line 115 in /home/pi/domoticz/plugins/zigate/plugin.py, function onMessage
2018-03-30 20:58:58.226 Error: (zigate1) ----> Line 352 in /home/pi/domoticz/plugins/zigate/plugin.py, function ZigateDecode
2018-03-30 20:58:58.226 Error: (zigate1) ----> Line 596 in /home/pi/domoticz/plugins/zigate/plugin.py, function ZigateRead
2018-03-30 20:58:58.226 Error: (zigate1) ----> Line 757 in /home/pi/domoticz/plugins/zigate/plugin.py, function Decode8102
2018-03-30 20:58:58.226 Error: (zigate1) ----> Line 1144 in /home/pi/domoticz/plugins/zigate/plugin.py, function ReadCluster
2018-03-30 20:58:58.226 Error: (zigate1) ----> Line 928 in /home/pi/domoticz/plugins/zigate/plugin.py, function MajDomoDevice
2018-03-30 20:58:58.226 (zigate1) Calling message handler 'onMessage'.
2018-03-30 20:58:58.226 (zigate1) onMessage called
2018-03-30 20:58:58.226 (zigate1) ZigateDecode - decodind data : 018102120210021e3c336a360211021002168002100210200210021102137b03
2018-03-30 20:58:58.227 (zigate1) ZigateRead - decoded data : 018102000e3c336a36010006800000200001037b03
2018-03-30 20:58:58.227 (zigate1) ZigateRead - Message Type : 8102, Data : 336a3601000680000020000103, RSSI : 7b, Length : 000e, Checksum : 3c
2018-03-30 20:58:58.227 (zigate1) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000e3c336a36010006800000200001037b03
2018-03-30 20:58:58.227 (zigate1) Decode8102 - reception data : 03 ClusterID : 0006 Attribut ID : 8000 Src Addr : 6a36 Scr Ep: 01
2018-03-30 20:58:58.227 (zigate1) MajDomoDevice - Device ID : 6a36 - Device EP : 01 - Type : 0006 - Value : 03
2018-03-30 20:58:58.228 Error: (zigate1) 'onMessage' failed 'UnboundLocalError':'local variable 'state' referenced before assignment'.
2018-03-30 20:58:58.228 Error: (zigate1) ----> Line 287 in /home/pi/domoticz/plugins/zigate/plugin.py, function onMessage
2018-03-30 20:58:58.228 Error: (zigate1) ----> Line 115 in /home/pi/domoticz/plugins/zigate/plugin.py, function onMessage
2018-03-30 20:58:58.228 Error: (zigate1) ----> Line 352 in /home/pi/domoticz/plugins/zigate/plugin.py, function ZigateDecode
2018-03-30 20:58:58.228 Error: (zigate1) ----> Line 596 in /home/pi/domoticz/plugins/zigate/plugin.py, function ZigateRead
2018-03-30 20:58:58.228 Error: (zigate1) ----> Line 757 in /home/pi/domoticz/plugins/zigate/plugin.py, function Decode8102
2018-03-30 20:58:58.228 Error: (zigate1) ----> Line 1144 in /home/pi/domoticz/plugins/zigate/plugin.py, function ReadCluster
2018-03-30 20:58:58.228 Error: (zigate1) ----> Line 928 in /home/pi/domoticz/plugins/zigate/plugin.py, function MajDomoDevice
2018-03-30 20:58:58.228 (zigate1) Calling message handler 'onMessage'.
2018-03-30 20:58:58.228 (zigate1) onMessage called
2018-03-30 20:58:58.228 (zigate1) ZigateDecode - decodind data : 018102120210021e3c336a360211021002168002100210200210021102137b03
2018-03-30 20:58:58.229 (zigate1) ZigateRead - decoded data : 018102000e3c336a36010006800000200001037b03
2018-03-30 20:58:58.229 (zigate1) ZigateRead - Message Type : 8102, Data : 336a3601000680000020000103, RSSI : 7b, Length : 000e, Checksum : 3c
2018-03-30 20:58:58.229 (zigate1) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000e3c336a36010006800000200001037b03
2018-03-30 20:58:58.229 (zigate1) Decode8102 - reception data : 03 ClusterID : 0006 Attribut ID : 8000 Src Addr : 6a36 Scr Ep: 01
2018-03-30 20:58:58.229 (zigate1) MajDomoDevice - Device ID : 6a36 - Device EP : 01 - Type : 0006 - Value : 03
2018-03-30 20:58:58.230 Error: (zigate1) 'onMessage' failed 'UnboundLocalError':'local variable 'state' referenced before assignment'.
2018-03-30 20:58:58.230 Error: (zigate1) ----> Line 287 in /home/pi/domoticz/plugins/zigate/plugin.py, function onMessage
2018-03-30 20:58:58.230 Error: (zigate1) ----> Line 115 in /home/pi/domoticz/plugins/zigate/plugin.py, function onMessage
2018-03-30 20:58:58.230 Error: (zigate1) ----> Line 352 in /home/pi/domoticz/plugins/zigate/plugin.py, function ZigateDecode
2018-03-30 20:58:58.230 Error: (zigate1) ----> Line 596 in /home/pi/domoticz/plugins/zigate/plugin.py, function ZigateRead
2018-03-30 20:58:58.230 Error: (zigate1) ----> Line 757 in /home/pi/domoticz/plugins/zigate/plugin.py, function Decode8102
2018-03-30 20:58:58.230 Error: (zigate1) ----> Line 1144 in /home/pi/domoticz/plugins/zigate/plugin.py, function ReadCluster
2018-03-30 20:58:58.230 Error: (zigate1) ----> Line 928 in /home/pi/domoticz/plugins/zigate/plugin.py, function MajDomoDevice
2018-03-30 20:58:58.230 (zigate1) Calling message handler 'onMessage'.
2018-03-30 20:58:58.230 (zigate1) onMessage called
2018-03-30 20:58:58.230 (zigate1) ZigateDecode - decodind data : 018102120210021e3c336a360211021002168002100210200210021102137b03
2018-03-30 20:58:58.231 (zigate1) ZigateRead - decoded data : 018102000e3c336a36010006800000200001037b03
2018-03-30 20:58:58.231 (zigate1) ZigateRead - Message Type : 8102, Data : 336a3601000680000020000103, RSSI : 7b, Length : 000e, Checksum : 3c
2018-03-30 20:58:58.231 (zigate1) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000e3c336a36010006800000200001037b03
2018-03-30 20:58:58.231 (zigate1) Decode8102 - reception data : 03 ClusterID : 0006 Attribut ID : 8000 Src Addr : 6a36 Scr Ep: 01
2018-03-30 20:58:58.231 (zigate1) MajDomoDevice - Device ID : 6a36 - Device EP : 01 - Type : 0006 - Value : 03
[... 7 lines cut ...]
2018-03-30 20:58:58.232 Error: (zigate1) ----> Line 928 in /home/pi/domoticz/plugins/zigate/plugin.py, function MajDomoDevice
2018-03-30 20:58:58.282 (zigate1) Received 18 bytes of data:
[... cut ...]
2018-03-30 20:58:58.284 Error: (zigate1) ----> Line 928 in /home/pi/domoticz/plugins/zigate/plugin.py, function MajDomoDevice
2018-03-30 20:59:01.783 (zigate1) Stop directive received.
2018-03-30 20:59:01.791 (zigate1) Disconnect directive received for '/dev/ttyUSB0'.
2018-03-30 20:59:01.840 (zigate1) Exiting work loop.
2018-03-30 20:59:06.787 Error: (zigate1) Plugin did not stop after 5 seconds, flushing event queue...
2018-03-30 20:59:06.787 (zigate1) Stopping threads.
2018-03-30 20:59:06.787 (zigate1) Stopped.

deennoo, 2018-04-03:
Corrigé dans une version que j'ai envoyer au dev.

la demarche : 
Dans le fichier DeviceConf.txt : Modifier la ligne : 7  Switch Pour MSwitch
Sauvegarder

Dans domoticz effacer le device, et refaire une inclusion : et voila il apparait en selector avec 1 click, 2 click, 3 click, 4 click

dondipietro, 2018-04-03:
Bonjour,
Cette solution marche mais donne parfois cette erreur : 

2018-04-03 18:23:14.152 (zigate1) Update 0:'00' (zigate1 - MSwitch1)
2018-04-03 18:23:16.110 (zigate1) Update 1:'10' (zigate1 - MSwitch1)
2018-04-03 18:23:16.120 (zigate1) Update 1:'On' (zigate1 - Motion - IR1)
2018-04-03 18:23:27.563 (zigate1) Update 0:'00' (zigate1 - MSwitch1)
**# 2018-04-03 18:23:27.564 Error: (CDevice_update) zigate1 - MSwitch1: Failed to parse parameters: 'nValue', 'sValue', 'Image', 'SignalLevel', 'BatteryLevel', 'Options', 'TimedOut', 'Name', 'TypeName', 'Type', 'Subtype', 'Switchtype' or 'Used' expected.
2018-04-03 18:23:27.564 Error: (zigate1) 'CDevice_update' failed 'OverflowError':'Python int too large to convert to C long'.**
2018-04-03 18:23:27.564 (zigate1) Update 1780301800000054080000045:'0' (zigate1 - MSwitch1)
2018-04-03 18:23:50.350 (zigate1) Update 3:'30' (zigate1 - MSwitch1)
2018-04-03 18:23:55.568 (zigate1) Update 0:'Off' (zigate1 - Motion - IR1)
2018-04-03 18:23:57.577 (zigate1) Devices exist in DB. Unit=44
2018-04-03 18:23:57.577 (zigate1) CurrentnValue = 3
2018-04-03 18:23:57.577 (zigate1) CurrentsValue = 30
2018-04-03 18:23:57.577 (zigate1) BatteryLvl = 152
2018-04-03 18:24:00.967 (zigate1) Update 2:'20' (zigate1 - MSwitch1)
2018-04-03 18:24:12.930 (zigate1) Update 3:'30' (zigate1 - MSwitch1)
2018-04-03 18:24:17.232 (zigate1) Devices exist in DB. Unit=44
2018-04-03 18:24:17.232 (zigate1) CurrentnValue = 3
2018-04-03 18:24:17.233 (zigate1) CurrentsValue = 30
2018-04-03 18:24:17.233 (zigate1) BatteryLvl = 152
2018-04-03 18:24:17.252 (zigate1) Update 1:'On' (zigate1 - Motion - IR1)
2018-04-03 18:24:21.873 (zigate1) Update 0:'00' (zigate1 - MSwitch1)

deennoo, 2018-04-04:
Tu parles du bouton rond ou le carré ?

dondipietro, 2018-04-04:
Celui-ci https://ae01.alicdn.com/kf/HTB1t4sJRFXXXXadXpXXq6xXFXXX6/Xiaomi-AQara-Smart-Multi-Functional-Intelligent-Wireless-Switch-Key-Built-In-Gyro-Function-Work-With-Android.jpg_640x640.jpg

Carré avec une partie mobile ronde

closed 2018-04-12 by zaraki673