zigbeefordomoticz/Domoticz-Zigbee #74: Heartbeat update device

deennoo, 2018-04-02:
Domoticz : Version: 3.9139 on rpi debian strech
Zigate Fw :3.0d
Plugin version : 2.3.3
Python 3.4.4

On various device like 86sw1/86sw2/lumi.switch.aq there is an heartbeat every 50 minutes (don't know if this come from the plugin or device).

This heartbeat trigger device and event attach to them.

Some logs : 

```
2018-04-02 10:37:41 | 3 Click
-- | --
2018-04-02 09:47:27 | 3 Click
2018-04-02 08:57:10 | 3 Click
2018-04-02 08:06:52 | 3 Click
2018-04-02 07:16:36 | 3 Click
2018-04-02 06:26:18 | 3 Click
2018-04-02 05:36:00 | 3 Click
2018-04-02 04:45:43 | 3 Click
2018-04-02 03:55:26 | 3 Click
2018-04-02 03:05:08 | 3 Click
2018-04-02 02:14:51 | 3 Click
2018-04-02 01:24:33 | 3 Click
2018-04-02 00:34:14 | 3 Click
2018-04-01 23:43:58 | 3 Click
2018-04-01 22:53:39 | 3 Click
2018-04-01 22:03:24 | 3 Click
2018-04-01 21:13:04 | 3 Click
2018-04-01 20:22:48 | 3 Click
2018-04-01 19:32:30 | 3 Click
2018-04-01 18:42:13 | 3 Click
2018-04-01 17:51:56 | 3 Click
2018-04-01 17:01:39 | 3 Click
2018-04-01 16:11:24 | 3 Click
2018-04-01 15:21:06 | 3 Click
2018-04-01 13:38:47 | 3 Click
```

deennoo, 2018-04-03:
i catch some log : 

`{}}}
2018-04-03 10:03:02.934 (Zigate USB) HB count = 118
2018-04-03 10:03:08.862 (Zigate USB) Calling message handler 'onMessage'.
2018-04-03 10:03:08.862 (Zigate USB) onMessage called
2018-04-03 10:03:08.862 (Zigate USB) Processing 'onMessageCallback' message
2018-04-03 10:03:08.862 (Zigate USB) Calling message handler 'onMessage'.
2018-04-03 10:03:08.862 (Zigate USB) onMessage called
2018-04-03 10:03:08.862 (Zigate USB) ZigateDecode - decodind data : 0181021202102811896f62021102100210ff021102104202101b021121f9021b02132813021421a813021521670210021624021e0210021002100210021a21021002100210a803
2018-04-03 10:03:08.863 (Zigate USB) ZigateRead - decoded data : 018102002811896f62010000ff010042001b0121f90b0328130421a8130521670006240e000000000a21000000a803
2018-04-03 10:03:08.863 (Zigate USB) ZigateRead - Message Type : 8102, Data : 896f62010000ff010042001b0121f90b0328130421a8130521670006240e000000000a21000000, RSSI : a8, Length : 0028, Checksum : 11
2018-04-03 10:03:08.863 (Zigate USB) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102002811896f62010000ff010042001b0121f90b0328130421a8130521670006240e000000000a21000000a803
2018-04-03 10:03:08.863 (Zigate USB) Decode8102 - reception data : 0121f90b0328130421a8130521670006240e000000000a21000000 ClusterID : 0000 Attribut ID : ff01 Src Addr : 6f62 Scr Ep: 01
2018-04-03 10:03:08.863 (Zigate USB) ReadCluster (8102) - ClusterId=0000 - MsgAttrID=ff01 - reception batteryLVL : 93 pour le device addr : 6f62
2018-04-03 10:03:08.863 (Zigate USB) Devices exist in DB. Unit=1
2018-04-03 10:03:08.864 (Zigate USB) CurrentnValue = 0
2018-04-03 10:03:08.864 (Zigate USB) CurrentsValue = On
2018-04-03 10:03:08.864 (Zigate USB) BatteryLvl = 93
2018-04-03 10:03:08.864 (Zigate USB - Switch - 6f62) Updating device from 0:'On' to have values 0:'On'.
2018-04-03 10:03:08.881 (Zigate USB) ReadCluster (8102) - ClusterId=0000 - MsgAttrID=ff01 - reception batteryLVL : erreur de lecture pour le device addr : 6f62
2018-04-03 10:03:08.886 EventSystem: Script event triggered: Plaf Loisir
2018-04-03 10:03:08.893 (Zigate USB) Pushing 'onCommandCallback' on to queue
2018-04-03 10:03:08.894 (Zigate USB) Pushing 'onCommandCallback' on to queue
2018-04-03 10:03:08.931 (Zigate USB) Processing 'onCommandCallback' message
2018-04-03 10:03:08.931 (Zigate USB) Calling message handler 'onCommand'.
2018-04-03 10:03:08.931 (Zigate USB) onCommand called for Unit 2: Parameter 'Off', Level: 0
2018-04-03 10:03:08.933 (Zigate USB) sendZigateCmd - length is : 0006
2018-04-03 10:03:08.933 (Zigate USB) getChecksum - Checksum : 0xf5
2018-04-03 10:03:08.933 (Zigate USB) ZigateEncode - Encodind data : 0092
2018-04-03 10:03:08.933 (Zigate USB) Transcode in : 0092 / out :021092
2018-04-03 10:03:08.933 (Zigate USB) ZigateEncode - Encodind data : 0006
2018-04-03 10:03:08.933 (Zigate USB) Transcode in : 0006 / out :02100216
2018-04-03 10:03:08.933 (Zigate USB) ZigateEncode - Encodind data : f5
2018-04-03 10:03:08.934 (Zigate USB) Transcode in : f5 / out :f5
2018-04-03 10:03:08.934 (Zigate USB) ZigateEncode - Encodind data : 023358010900
2018-04-03 10:03:08.934 (Zigate USB) Transcode in : 023358010900 / out :02123358021102190210
2018-04-03 10:03:08.934 (Zigate USB) sendZigateCmd - Comand send : 0102109202100216f50212335802110219021003
2018-04-03 10:03:08.934 (Zigate USB) Pushing 'WriteDirective' on to queue
2018-04-03 10:03:08.934 (Zigate USB - Plug Salus- 3358) Updating device from 1:'On' to have values 0:'Off'.
2018-04-03 10:03:08.947 (Zigate USB) Update 0:'Off' (Zigate USB - Plug Salus- 3358)
2018-04-03 10:03:08.947 (Zigate USB) Processing 'onCommandCallback' message
2018-04-03 10:03:08.947 (Zigate USB) Calling message handler 'onCommand'.
2018-04-03 10:03:08.947 (Zigate USB) onCommand called for Unit 4: Parameter 'Off', Level: 0
2018-04-03 10:03:08.949 (Zigate USB) sendZigateCmd - length is : 0006
2018-04-03 10:03:08.949 (Zigate USB) getChecksum - Checksum : 0x90
2018-04-03 10:03:08.949 (Zigate USB) ZigateEncode - Encodind data : 0092
2018-04-03 10:03:08.949 (Zigate USB) Transcode in : 0092 / out :021092
2018-04-03 10:03:08.949 (Zigate USB) ZigateEncode - Encodind data : 0006
2018-04-03 10:03:08.949 (Zigate USB) Transcode in : 0006 / out :02100216
2018-04-03 10:03:08.949 (Zigate USB) ZigateEncode - Encodind data : 90
2018-04-03 10:03:08.949 (Zigate USB) Transcode in : 90 / out :90
2018-04-03 10:03:08.950 (Zigate USB) ZigateEncode - Encodind data : 02fcfa010100
2018-04-03 10:03:08.950 (Zigate USB) Transcode in : 02fcfa010100 / out :0212fcfa021102110210
2018-04-03 10:03:08.950 (Zigate USB) sendZigateCmd - Comand send : 0102109202100216900212fcfa02110211021003
2018-04-03 10:03:08.950 (Zigate USB) Pushing 'WriteDirective' on to queue
2018-04-03 10:03:08.950 (Zigate USB - Switch - fcfa) Updating device from 1:'On' to have values 0:'Off'.
2018-04-03 10:03:08.965 (Zigate USB) Update 0:'Off' (Zigate USB - Switch - fcfa)
2018-04-03 10:03:08.965 (Zigate USB) Processing 'WriteDirective' message
2018-04-03 10:03:08.965 (Zigate USB) Sending 20 bytes of data
2018-04-03 10:03:08.965 (Zigate USB) 01 02 10 92 02 10 02 16 f5 02 12 33 58 02 11 02 19 02 10 03 ...�....�..3X.......
2018-04-03 10:03:08.965 (Zigate USB) Processing 'WriteDirective' message
2018-04-03 10:03:08.965 (Zigate USB) Sending 20 bytes of data
2018-04-03 10:03:08.965 (Zigate USB) 01 02 10 92 02 10 02 16 90 02 12 fc fa 02 11 02 11 02 10 03 ...�....�..��.......
[... cut ...]
2018-04-03 10:03:09.143 (Zigate USB) ZigateRead - MsgType 8000 - reception status : 018000000580009700920003
2018-04-03 10:03:09.143 (Zigate USB) Decode8000 - Reception status : Success, SQN : 92, Message :
2018-04-03 10:03:09.886 (Yeelight) Lighting Limitless/Applamp (Plafonnier Loisir)
2018-04-03 10:03:12.908 (Zigate USB) Pushing 'onHeartbeatCallback' on to queue
2018-04-03 10:03:12.951 (Zigate USB) Processing 'onHeartbeatCallback' message
2018-04-03 10:03:12.951 (Zigate USB) Calling message handler 'onHeartbeat'.`

pookiege, 2018-04-03:
I can confirm that every ~50 minutes the device sends his battery level, triggering an event in Domoticz. A first attempt was to simply comment the battery update in the code, solving (badly) my problem.

deennoo, 2018-04-03:
Another one can be : 

Create à Voltage device not link to the Switch

The last one can be to search how is this manage on other plugin.

zaraki673, 2018-04-12:
ok, voir pour que la batterie ne soit mis à jour que si nvalue et svalue sont mis à jour

je regarde ça ;)

pipiche38, 2018-04-13:
In advance sorry to use this post .

@deennoo I have open and issue on a Zigate communication with Profalux rolling shutters. Do you have such experience ?

deennoo, 2018-04-13:
@pipiche38 nope i don't have

commit 26ffa885 (2018-04-13): fix #74
closed 2018-04-14 by zaraki673