zigbeefordomoticz/Domoticz-Zigbee #76: Certaines trames recues ne sont pas décodées aqara door sensor

dondipietro, 2018-04-02:
Bonjour,
Voici le log, afin de mieux voir les données reçues j'ai modifié le programme 
def onMessage(self, Connection, Data):
		#Domoticz.Log("onMessage called")
		global Tmprcv
		global ReqRcv
		Tmprcv=binascii.hexlify(Data).decode('utf-8')
		**Domoticz.Log("onMessage called "+str(Tmprcv))**

Dans le log certaines trame du switch aqra door sensor se bien prise en compte d'autres non 

short adress of (zigate1 - Door - 42) : E504 => E50214

2018-04-02 14:16:06.318 (zigate1) Update 1:'Open' (zigate1 - Door - 41)
2018-04-02 14:16:46.491 (zigate1) onMessage called 018102120210021e4fd9e64302110210021602100210021010021002110210a803
2018-04-02 14:16:46.548 (zigate1) Update 0:'Closed' (zigate1 - Door - 41)
2018-04-02 14:16:47.501 (zigate1) onMessage called 018102120210021f7ab013ee02110214021202100210021029021002120218ff6003018102120210021fc7b113ee021102140215021002100210210210021211556003018102120210021f59b213ee02110214021302100210021029021002120213d46003018102120210021e66b213ee02110214021302101402102802100211ff6003018102120210021fe8b213ee0211021402130210100210290210021226506003
2018-04-02 14:16:47.536 (zigate1) Update 0:'23.0;44.4;1' (ZG1 - TH5 Cuisine)
2018-04-02 14:16:47.556 (zigate1) Update 0:'23.0' (zigate1 - Temp - 13ee)
2018-04-02 14:16:48.612 (zigate1) onMessage called 018102120210021e73dae643021102100216021002100210100210021102119603
2018-04-02 14:16:50.401 (zigate1) onMessage called 018102120210021e7cdbe643021102100216021002100210100210021102109903
2018-04-02 14:16:52.558 (zigate1) onMessage called 018102120210021e4bdce64302110210021602100210021010021002110211a803
2018-04-02 14:16:52.610 (zigate1) Update 1:'Open' (zigate1 - Door - 41)
2018-04-02 14:16:54.866 (zigate1) onMessage called 018102120210021e4bdde643021102
2018-04-02 14:16:54.866 (zigate1) Else onMessage
2018-04-02 14:16:54.868 (zigate1) onMessage called 10021602100210021010021002110210a803
2018-04-02 14:16:54.914 (zigate1) Update 0:'Closed' (zigate1 - Door - 41)
2018-04-02 14:16:55.816 (zigate1) onMessage called 018102120210021e49dee64302110210021602100210021010021002110211a803
2018-04-02 14:16:55.864 (zigate1) Update 1:'Open' (zigate1 - Door - 41)
2018-04-02 14:17:05.946 (zigate1) onMessage called 018102120210021fee84549f021102140212021002100210290210021202184c
2018-04-02 14:17:05.946 (zigate1) Else onMessage
2018-04-02 14:17:05.999 (zigate1) onMessage called 4503018102120210021f4085549f021102140215021002100210210210021212f64503018102120210021f7f86549f02110214021302100210021029021002120213d54503018102120210021e4186549f02110214021302101402102802100211ff4503018102120210021fcd86549f0211021402130210100210290210021226524503
2018-04-02 14:17:06.035 (zigate1) Update 0:'21.2;48.8;1' (ZG1 - TH7 Veranda)
2018-04-02 14:17:06.056 (zigate1) Update 0:'21.2' (zigate1 - Temp - 549f)
2018-04-02 14:17:06.457 (zigate1) onMessage called 018102120210021ef426e5021402110210021602100210021010021002110210a803
2018-04-02 14:17:06.488 (zigate1) Update 0:'21.2;48.5;1' (ZG1 - TH7 Veranda)
2018-04-02 14:17:06.510 (zigate1) Update 0:'1.3156674017467442e+155;2.7858011880155747e+133' (zigate1 - Baro - 549f)
**2018-04-02 14:17:20.603 (zigate1) onMessage called 018102120210021ef227e5021402110210021602100210021010021002110211ae03**
2018-04-02 14:17:41.772 (zigate1) onMessage called 018102120210021ef728e5021402110210021602100210021010021002
2018-04-02 14:17:41.772 (zigate1) Else onMessage
2018-04-02 14:17:41.772 (zigate1) onMessage called 110210a503
2018-04-02 14:17:41.812 (zigate1) Update 0:'Closed' (zigate1 - Door - 42)
2018-04-02 14:17:47.734 (zigate1) onMessage called 018102120210021ef929e5021402110210021602100210021010021002110211ab03
2018-04-02 14:17:47.772 (zigate1) Update 1:'Open' (zigate1 - Door - 42)

La trame de 2018-04-02 14:17:20.603  ne semble pas avoir été lue par le plugin.
Alors que 30 sencondes plus tard à 2018-04-02 14:17:41.772 celà a fonctionné.

dondipietro, 2018-04-02:
J'ai décodé la trame à l'aide du plugin modifié à la main : 
decoded = ZigateDecode("**018102120210021ef227e5021402110210021602100210021010021002110211ae03**")
ZigateDecode - decodind data : 018102120210021ef227e5021402110210021602100210021010021002110211ae03
decoded
Out[25]: '018102000ef227e50401000600000010000101ae03'

Decode8102(decoded[12:])
Decode8102 - reception data : 01ae03 ClusterID : 0006 Attribut ID : 0000 Src Addr : e504 Scr Ep: 01

La trame me semble donc correcte. Je ne comprends pas pourquoi le plugin ne génère pas un appel a domoticz suite à la reception de cette trame.

dondipietro, 2018-04-02:
A nouveau testé mais avec le full debug ci-dessous 
A cette ligne ci : 
2018-04-02 16:44:41.859 (zigate1) ZigateDecode - decodind data : 018102120210021fd99fd2d802110214021302100210021029021002120213d23c03018102120210021ee09fd2d802110214021302101402102802100211ff3c03018102120210021f02179fd2d80211021402130210100210290210021226393c03018102120210021eb444e50214021102100216021002100210100210021102108a03

En fait cette ligne est composée de 4 messages collés
018102120210021fd99fd2d802110214021302100210021029021002120213d23c03
018102120210021ee09fd2d802110214021302101402102802100211ff3c03
018102120210021f02179fd2d80211021402130210100210290210021226393c03
018102120210021eb444e50214021102100216021002100210100210021102108a03

Le décodage via python de ces 4 lignes donne ceci

Data="018102120210021fd99fd2d802110214021302100210021029021002120213d23c03"
MsgData=Decode8102(ZigateDecode(Data)[12:])
ZigateDecode - decodind data : 018102120210021fd99fd2d802110214021302100210021029021002120213d23c03
Decode8102 - reception data : 03d23c03 ClusterID : 0403 Attribut ID : 0000 Src Addr : d2d8 Scr Ep: 01

Data="018102120210021ee09fd2d802110214021302101402102802100211ff3c03"
MsgData=Decode8102(ZigateDecode(Data)[12:])
ZigateDecode - decodind data : 018102120210021ee09fd2d802110214021302101402102802100211ff3c03
Decode8102 - reception data : ff3c03 ClusterID : 0403 Attribut ID : 0014 Src Addr : d2d8 Scr Ep: 01

Data="018102120210021f02179fd2d80211021402130210100210290210021226393c03"
MsgData=Decode8102(ZigateDecode(Data)[12:])
ZigateDecode - decodind data : 018102120210021f02179fd2d80211021402130210100210290210021226393c03
Decode8102 - reception data : 26393c03 ClusterID : 0403 Attribut ID : 0010 Src Addr : d2d8 Scr Ep: 01

Data="018102120210021eb444e50214021102100216021002100210100210021102108a03"
MsgData=Decode8102(ZigateDecode(Data)[12:])
ZigateDecode - decodind data : 018102120210021eb444e50214021102100216021002100210100210021102108a03
Decode8102 - reception data : 008a03 ClusterID : 0006 Attribut ID : 0000 Src Addr : e504 Scr Ep: 01

Le dernier morceau de la trame est bien le door sensor 42. 

Je pense que quand au cours d'une seule lecture il y a des trames humidité température et une trame door sensor, la trame door sensor n'est pas décodée et le status open ou close n'est pas renvoyé à domoticz.

Voici le full log ci dessous au cas ou. La trame door sensor a marché plusieurs fois sauf justement à la ligne 2018-04-02 16:44:41.859

`Domoticz V3.9165
 Dashboard
 Floorplan
 Switches
 Scenes
 Temperature
 Weather
 Utility
 Setup 
All
Status
Error
Filter: 
 X
2018-04-02 16:44:06.830 (zigate1) Pushing 'onHeartbeatCallback' on to queue
2018-04-02 16:44:06.840 (zigate1) Processing 'onHeartbeatCallback' message
2018-04-02 16:44:06.840 (zigate1) Calling message handler 'onHeartbeat'.
2018-04-02 16:44:06.841 (zigate1) ListOfDevices : {'edff': {'ProfileID': {}, 'MacCapa': '80', 'Heartbeat': '25908', 'ZDeviceID': {}, 'Status': 'inDB', 'RIA': '656', 'IEEE': '00158d0001713d21', 'Ep': {'01': {'0000': {}, '0402': 25.6, '0403': 2.792411232566359e+133, '0405': 40.1}}, 'Battery': {}, 'Type': 'Temp/Humi/Baro', 'Model': 'lumi.weather'}, '13ee': {'ProfileID': {}, 'MacCapa': '80', 'Heartbeat': '25690', 'ZDeviceID': {}, 'Model': 'lumi.weather', 'RIA': '587', 'IEEE': '00158d0001a4b95e', 'Battery': {}, 'Status': 'inDB', 'Type': 'Temp/Humi/Baro', 'Ep': {'01': {'0000': {}, '0402': 21.9, '0403': 'ff', '0405': 46.7}}}, '7c37': {'ProfileID': {}, 'MacCapa': '80', 'Heartbeat': '25657', 'ZDeviceID': {}, 'Model': 'lumi.weather', 'RIA': '603', 'Battery': 91, 'IEEE': '00158d000171dfa0', 'Status': 'inDB', 'Type': 'Temp/Humi/Baro', 'Ep': {'01': {'0000': {}, '0402': 23.8, '0403': 1.3903084681141902e+220, '0405': 43.5}}}, '549f': {'ProfileID': {}, 'RIA': '550', 'Heartbeat': '25366', 'ZDeviceID': {}, 'Status': 'inDB', 'MacCapa': '80', 'IEEE': '00158d0001a4ba68', 'Ep': {'01': {'0000': {}, '0402': 21.2, '0403': 2.7946182776428787e+133, '0405': 49.1}}, 'Battery': {}, 'Type': 'Temp/Humi/Baro', 'Model': 'lumi.weather'}, '70bc': {'ProfileID': {}, 'RIA': '586', 'Heartbeat': '25627', 'ZDeviceID': {}, 'Model': 'lumi.weather', 'MacCapa': '80', 'IEEE': '00158d0001a35ced', 'Battery': 92, 'Status': 'inDB', 'Type': 'Temp/Humi/Baro', 'Ep': {'01': {'0000': {}, '0402': 21.4, '0403': 'ff', '0405': 49.4}}}, '52b6': {'ProfileID': {}, 'RIA': '588', 'Heartbeat': '25606', 'ZDeviceID': {}, 'Model': 'lumi.weather', 'MacCapa': '80', 'IEEE': '00158d0001a230fc', 'Battery': 90, 'Status': 'inDB', 'Type': 'Temp/Humi/Baro', 'Ep': {'01': {'0000': {}, '0402': 18.4, '0403': 'ff', '0405': 55.6}}}, 'cd75': {'ProfileID': {}, 'RIA': '574', 'Heartbeat': '25694', 'ZDeviceID': {}, 'Model': 'lumi.weather', 'MacCapa': '80', 'Battery': {}, 'IEEE': '00158d000171f2a2', 'Status': 'inDB', 'Type': 'Temp/Humi/Baro', 'Ep': {'01': {'0000': {}, '0402': 18.5, '0403': 2.78606736430621e+133, '0405': 56.9}}}, 'd2d8': {'ProfileID': {}, 'MacCapa': '80', 'Heartbeat': '25709', 'ZDeviceID': {}, 'Model': 'lumi.weather', 'RIA': '563', 'IEEE': '00158d00017200e6', 'Battery': 90, 'Status': 'inDB', 'Type': 'Temp/Humi/Baro', 'Ep': {'01': {'0000': {}, '0402': 18.9, '0403': 'ff', '0405': 56.1}}}, '2d53': {'ProfileID': {}, 'MacCapa': '80', 'Heartbeat': '25922', 'ZDeviceID': {}, 'Model': 'lumi.weather', 'RIA': '717', 'IEEE': '00158d0001a4a490', 'Battery': 93, 'Status': 'inDB', 'Type': 'Temp/Humi/Baro', 'Ep': {'01': {'0000': {}, '0402': 19.9, '0403': 'ff', '0405': 53.1}}}, '0862': {'ProfileID': {}, 'RIA': '599', 'Heartbeat': '25936 [... cut ...] fileID': 'c05e', 'Ep': {'03': {'0005': {}, '1000': {}, '0b04': {}, 'fc0f': {}, '0004': {}, '0000': {}, '0003': {}, '0006': {}}}, 'Heartbeat': '5', 'Battery': {}, 'RIA': '0'}, '2cb4': {'ZDeviceID': '0010', 'Model': 'plug.osram', 'MacCapa': '8e', 'Type': 'Switch', 'IEEE': '84182600000ca957', 'Status': 'inDB', 'ProfileID': 'c05e', 'Ep': {'03': {'0005': {}, '1000': {}, '0b04': {}, 'fc0f': {}, '0004': {}, '0000': {}, '0003': {}, '0006': {}}}, 'Heartbea
2018-04-02 16:44:46.892 (zigate1) HB count = 4
`

dondipietro, 2018-04-03:
Voici un correctif qui sembke fonctionner en cas de reception de plusieurs magges en même temps : 
`	def onMessage(self, Connection, Data):
		#print("onMessage called")
		global Tmprcv
		global ReqRcv
		Tmprcv=binascii.hexlify(Data).decode('utf-8')
		if Tmprcv.find('03') != -1 and len(ReqRcv+Tmprcv[:Tmprcv.find('03')+2])%2==0 :### fin de messages detecter dans Data
			ReqRcv+=Tmprcv#[:Tmprcv.find('03')+2] #
			#try :
			while (ReqRcv.find("0301") != -1): #Tant qu'il reste des messages colles
				ZigateDecode(self, ReqRcv[:ReqRcv.find("0301")+2])#demande de decodage de la trame recu
				ReqRcv=ReqRcv[ReqRcv.find("0301")+2:]  
				time.sleep(1)             
			if ReqRcv.find('03') != -1 : #est ce qu il reste une derniere trame complete ?
				ZigateDecode(self, ReqRcv[:ReqRcv.find("03")+2]) #demande de decodage de la trame recu
				ReqRcv=ReqRcv[ReqRcv.find('03')+2:] 
				#print("end:", ReqRcv)
			#else : 
				#print("end:", ReqRcv)
		#except :
			#	print("onMessage - effacement de la trame suite a une erreur de decodage : " + ReqRcv)
			#	ReqRcv = Tmprcv[Tmprcv.find('03')+2:]  # efface le tampon en cas d erreur
		else : # while end of data is receive
			ReqRcv+=Tmprcv
		return

`

zaraki673, 2018-04-12:
bonjour,
ton correctif est ajouté au code,
merci ;)

closed 2018-04-12 by zaraki673