zigbeefordomoticz/Domoticz-Zigbee #78: Error: (zigate) 'onHeartbeat' failed 'KeyError'

james013, 2018-04-08:
par contre depuis la dernière version j'ai ces messages toutes les 10s et au bout d'un moment plus de mise à jour des devices, obligé de desactiver/réactiver le plugin : 

```
2018-04-08 12:25:03.257 Error: (zigate) 'onHeartbeat' failed 'KeyError'.
2018-04-08 12:25:03.257 Error: (zigate) ----> Line 299 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:03.257 Error: (zigate) ----> Line 183 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:13.265 Error: (zigate) 'onHeartbeat' failed 'KeyError'.
2018-04-08 12:25:13.265 Error: (zigate) ----> Line 299 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:13.266 Error: (zigate) ----> Line 183 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:23.272 Error: (zigate) 'onHeartbeat' failed 'KeyError'.
2018-04-08 12:25:23.272 Error: (zigate) ----> Line 299 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:23.272 Error: (zigate) ----> Line 183 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:33.330 Error: (zigate) 'onHeartbeat' failed 'KeyError'.
2018-04-08 12:25:33.330 Error: (zigate) ----> Line 299 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:33.330 Error: (zigate) ----> Line 183 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:43.338 Error: (zigate) 'onHeartbeat' failed 'KeyError'.
2018-04-08 12:25:43.339 Error: (zigate) ----> Line 299 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:43.339 Error: (zigate) ----> Line 183 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:53.391 Error: (zigate) 'onHeartbeat' failed 'KeyError'.
2018-04-08 12:25:53.391 Error: (zigate) ----> Line 299 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:25:53.391 Error: (zigate) ----> Line 183 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:26:03.388 Error: (zigate) 'onHeartbeat' failed 'KeyError'.
2018-04-08 12:26:03.389 Error: (zigate) ----> Line 299 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onHeartbeat
2018-04-08 12:26:03.389 Error: (zigate) ----> Line 183 in /usr/local/domoticz/var/plugins/Domoticz-

```

zaraki673, 2018-04-12:
j ai réactivé la suppression des trames en cas d erreur de lecture, faudra trouver la trame qui pose soucis ;)

Smanar, 2018-08-21:
J'ai le meme probleme.
En fait ce probleme apparait a chaque lancement du pluggin, mais il y a autre chose qui n'apparait qu'au premier lancement
```
 2018-08-21 19:24:27.220 Error: (Zigate) 'onStart' failed 'KeyError'.
2018-08-21 19:24:27.220 Error: (Zigate) ----> Line 488 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onStart
2018-08-21 19:24:27.220 Error: (Zigate) ----> Line 77 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onStart 
```
Le probleme est sur la ligne
`self.ListOfDevices[ID]=eval(Devices[x].Options['Zigate'])`

Par contre impossible de mettre en place un debogage sur cette partie, j'ai rien qui apparit dans les logs a part l'erreur.

pipiche38, 2018-08-21:
Utilises-tu la dernière version, car en ligne 77 il n'y a pas du tout cette ligne de code !

Peux-tu préciser version de Domoticz et t'assurer que c'est bien la dernière version du plugin.

Smanar, 2018-08-21:
Yep, c'est la version developpement bidouille (thorgal), sinon tout est a jours.
Mais la j'ai laisse tomber, comme les lignes de debug ne marchent pas ici( enfin je croyais), j'ai voulu sauver le debug dans un fichier > bloquage de domoticz > plantage > redemarrage impossible > recherche sur forum > encore bug > reinstallation.

Donc j'ai plus le bug ^^, mais je pense pouvoir le provoquer en editant les "devices", boutton "modifier", 

La je viens de comprendre comment fonctionne le log et qu'il n'affiche pas tout, la prochaine fois que j'ai le bug, je regarderais mieux. Mais en fait c'est la lecture des devices de domoticz qui provoquait des keyError un peu partout dans ton plugin eg
`Dtypename=DOptions['TypeName']`
Et on se retrouve dans un cas sans TypeName.

pipiche38, 2018-08-22:
@Smanar c'est bon j'ai recollé les morceaux ;-)

Comme tu l'as identifié,  c'est dans la boucle d'initialisation des Devices, et vraisemblablement on attend des valeurs 'Zigate' stockées du coté de Domoticz qui ne sont pas au rendez-vous :-(
```

                for x in Devices : # initialise listeofdevices avec les devices en bases domoticz
                        ID = Devices[x].DeviceID
                        self.ListOfDevices[ID]={}
                        self.ListOfDevices[ID]=eval(Devices[x].Options['Zigate'])

```

On devrait s'assurer qu'il y a bien une valeur récupérée. Et si rien, ac ce moment il ne faut pas initialiser ce Devices[x] car il est douteux coté Domoticz !

Dans la TODOs il y a notamment un check de consistance des bases de données au démarrage. En effet on a : 
- Base Domoticz qui sert justement à initialiser Devices[]
- Base du Plugin qui sert à initialiser self.ListOfDevices{}
- Liste des Devices dans la Zigate que l'on obtient via la commande 0x0015/0x8015

pipiche38, 2018-08-22:
En attendant, voila ce que je propose:
```

                for x in Devices : # initialise listeofdevices avec les devices en bases domoticz
                        ID = Devices[x].DeviceID
                        try:
                                self.ListOfDevices[ID]=eval(Devices[x].Options['Zigate'])
                        except:
                                Domoticz.Error("Error loading Device " +str(Devices[x] + " not loaded int Zigate Plugin!" )
                        else :
                                self.ListOfDevices[ID]={}
```

pipiche38, 2018-09-18:
This has introduced a problem, which is that the loading of the Domoticz database is reset in the else statement of the try/except encapsulation

pipiche38, 2018-09-18:
Fix in 3.0.3 ( 63f31912c8bfe753326e6a37618432ed088f1b6a )

closed 2018-09-18 by pipiche38