zigbeefordomoticz/Domoticz-Zigbee #79: Device THB  pour capteur de température humidité pression atmosphérique Xiaomi Aqara

lboue, 2018-04-10:
Bonjour,

Serait-il possible de remonter le capteur de température humidité pression atmosphérique Xiaomi Aqara dans un seul device THB  (qui est prévu pour ça) plutôt que de multiples devices ?

### Actuellement 
Type				Sous type
Temp + Humidity 	THGN122
General				Barometer

### Cible
Mais plutôt dans un triple device THB ?
Type					Sous type
Temp + Humidity + Baro	THB1 

Cordialement,
Ludovic

lboue, 2018-04-10:
I forked the repo and change the code to add THB device. Now I have this error:

```
Warning: Expecting svalue with at least 4 elements separated by semicolon, 3 elements received ("23.6;55.3;1"), notification not sent
```

**Full message for THB** 

```
2018-04-10 22:12:05.646  (Zigate) CurrentnValue = 0
2018-04-10 22:12:05.646  (Zigate) CurrentsValue = 23.6;55.3;1
2018-04-10 22:12:05.646  (Zigate) BatteryLvl = 90
2018-04-10 22:12:05.647  (Zigate - Temp+Hum+Baro - 071a) Updating device from 0:'23.6;55.3;1' to have values 0:'23.6;55.3;1'.
2018-04-10 22:12:05.653  Warning: Expecting svalue with at least 4 elements separated by semicolon, 3 elements received ("23.6;55.3;1"), notification not sent (Hardware: 15 - Zigate, ID: 071a, Unit: 10, Type: 54 - Temp + Humidity + Baro, SubType: 1 - THB1 - BTHR918, BTHGN129)
2018-04-10 22:12:05.654  (Zigate) Devices exist in DB. Unit=11
2018-04-10 22:12:05.654  (Zigate) CurrentnValue = 0
2018-04-10 22:12:05.654  (Zigate) CurrentsValue = 23.6;55.3;1
2018-04-10 22:12:05.655  (Zigate) BatteryLvl = 90

```

**Full message for the device list** 
```
2018-04-10 22:12:05.642  (Zigate) onMessage called
2018-04-10 22:12:05.643  (Zigate) ZigateDecode - decodind data : 018102120210322a2402171a021102100210ff0211021042021025021121a9021b021421a8130215210215021002162402110210021002100210642937021965219715662bc98202110210021a2102100210bd03
2018-04-10 22:12:05.644  (Zigate) ZigateRead - decoded data : 01810200322a24071a010000ff01004200250121a90b0421a81305210500062401000000006429370965219715662bc98201000a210000bd03
2018-04-10 22:12:05.644  (Zigate) ZigateRead - Message Type : 8102, Data : 24071a010000ff01004200250121a90b0421a81305210500062401000000006429370965219715662bc98201000a210000, RSSI : bd, Length : 0032, Checksum : 2a
2018-04-10 22:12:05.645  (Zigate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 01810200322a24071a010000ff01004200250121a90b0421a81305210500062401000000006429370965219715662bc98201000a210000bd03
2018-04-10 22:12:05.645  (Zigate) Decode8102 - reception data : 0121a90b0421a81305210500062401000000006429370965219715662bc98201000a210000 ClusterID : 0000 Attribut ID : ff01 Src Addr : 071a Scr Ep: 01
2018-04-10 22:12:05.646  (Zigate) ReadCluster (8102) - ClusterId=0000 - MsgAttrID=ff01 - reception batteryLVL : 90 pour le device addr : 071a
2018-04-10 22:12:05.646  (Zigate) Devices exist in DB. Unit=10
2018-04-10 22:12:05.646  (Zigate) CurrentnValue = 0
2018-04-10 22:12:05.646  (Zigate) CurrentsValue = 23.6;55.3;1
2018-04-10 22:12:05.646  (Zigate) BatteryLvl = 90
2018-04-10 22:12:05.647  (Zigate - Temp+Hum+Baro - 071a) Updating device from 0:'23.6;55.3;1' to have values 0:'23.6;55.3;1'.
2018-04-10 22:12:05.653  Warning: Expecting svalue with at least 4 elements separated by semicolon, 3 elements received ("23.6;55.3;1"), notification not sent (Hardware: 15 - Zigate, ID: 071a, Unit: 10, Type: 54 - Temp + Humidity + Baro, SubType: 1 - THB1 - BTHR918, BTHGN129)
2018-04-10 22:12:05.654  (Zigate) Devices exist in DB. Unit=11
2018-04-10 22:12:05.654  (Zigate) CurrentnValue = 0
2018-04-10 22:12:05.654  (Zigate) CurrentsValue = 23.6;55.3;1
2018-04-10 22:12:05.655  (Zigate) BatteryLvl = 90
2018-04-10 22:12:05.655  (Zigate - Temp+Hum - 071a) Updating device from 0:'23.6;55.3;1' to have values 0:'23.6;55.3;1'.
2018-04-10 22:12:05.657  (Zigate) Devices exist in DB. Unit=12
2018-04-10 22:12:05.658  (Zigate) CurrentnValue = 0
2018-04-10 22:12:05.658  (Zigate) CurrentsValue = 23.6
2018-04-10 22:12:05.658  (Zigate) BatteryLvl = 90
2018-04-10 22:12:05.658  (Zigate - Temp - 071a) Updating device from 0:'23.6' to have values 0:'23.6'.
2018-04-10 22:12:05.661  (Zigate) Devices exist in DB. Unit=13
2018-04-10 22:12:05.661  (Zigate) CurrentnValue = 55
2018-04-10 22:12:05.661  (Zigate) CurrentsValue = 0
2018-04-10 22:12:05.662  (Zigate) BatteryLvl = 90
2018-04-10 22:12:05.662  (Zigate - Humi - 071a) Updating device from 55:'0' to have values 55:'0'.
2018-04-10 22:12:05.664  (Zigate) Devices exist in DB. Unit=14
2018-04-10 22:12:05.664  (Zigate) CurrentnValue = 0
2018-04-10 22:12:05.665  (Zigate) CurrentsValue = 990.0;9.9
2018-04-10 22:12:05.665  (Zigate) BatteryLvl = 90
2018-04-10 22:12:05.665  (Zigate - Baro - 071a) Updating device from 0:'990.0;9.9' to have values 0:'990.0;9.9'.
2018-04-10 22:12:05.672  (Zigate) ReadCluster (8102) - ClusterId=0000 - MsgAttrID=ff01 - reception batteryLVL : erreur de lecture pour le device addr : 071a
2018-04-10 22:12:14.470  (Zigate) Pushing 'onHeartbeatCallback' on to queue
2018-04-10 22:12:14.488  (Zigate) Processing 'onHeartbeatCallback' message
```

lboue, 2018-04-10:
Le problème semble corrigé désormais. Je n'ai plus l'erreur:
```
2018-04-10 23:33:36.794  (Zigate) MajDomoDevice - Device ID : 2791 - Device EP : 01 - Type : 0403 - Value : 9.9
2018-04-10 23:33:36.795  (Zigate) MajDomoDevice baro CurrentsValue : 989.9;9.9
```

zaraki673, 2018-04-12:
bonjour merci, j ai ajouté ton code ;)

zaraki673, 2018-04-12:
#0ded3a1

closed 2018-04-12 by zaraki673

lboue, 2018-04-21:
Je n'avais pas terminé mes tests car j'étais occupé entre temps. J'ai ajouté un nouveau PR : #82 pour corriger.

Merci 😄