zigbeefordomoticz/Domoticz-Zigbee #80: What should I expect when pairing Profalux rolling shutter

pipiche38, 2018-04-13:
What should I expect when pairing Profalux rolling shutter ?

I have pair one shutter. When I Disable/enable the zigate hardware I got the shutter moving up/dow quickly, but then nothing .

I'm not able to see anything in the devices list.

Nothing special from the logs

zaraki673, 2018-04-13:
hi, 

sorry, but nothing yet, i don't support it yet

but soon, now i know someone need it :D

zaraki673, 2018-04-14:
ca devrait etre bon, j attends ton retour ;)

pipiche38, 2018-04-14:
Let me know if I should do it in French ? or if English is ok ?

pipiche38, 2018-04-14:
I have the confirmation from Profalux that this is based on HA 1.2 version , which is compatible with Zigate from the Zigate documentation.

When I pair a shutter, I don't see devices coming to the Devices list.
If I try to create Manual a Switch I don't see it either.

Attached is the log file of the following

(1) Start domoticz server with Zigate Plugin already enabled with Full debug mode.
(2) Try a shutter pairing 
(3) Shutdown the Domoticz server
     -> This create a move of the shutter , which means that there is some communication between the 2

[domoticz.log](https://github.com/sasu-drooz/Domoticz-Zigate/files/1910145/domoticz.log)

Last:
- The domoticz server is running on a Synology NAS.

pipiche38, 2018-04-14:
Don't know exactly what's happen, but I restarted the Domoticz server, and the shutter was listed in the not used Devices ;-)

pipiche38, 2018-04-14:
There is one problem detected which is that there is no update to the percentage opening. I'm here attaching a screenshot.

Basically if you open the shutter with the On/off, then the % is not updated.
If you open/close the shutter with the shutter remote command, there is no update either.

[image]

pipiche38, 2018-04-14:
Found this error message in the logs
2018-04-14 12:08:58.900  Error: (Zigate) 'onMessage' failed 'ValueError':'invalid literal for int() with base 16: '''.
2018-04-14 12:08:58.900  Error: (Zigate) ----> Line 307 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-04-14 12:08:58.900  Error: (Zigate) ----> Line 121 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-04-14 12:08:58.900  Error: (Zigate) ----> Line 372 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function ZigateDecode
2018-04-14 12:08:58.900  Error: (Zigate) ----> Line 442 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function ZigateRead
2018-04-14 12:08:58.900  Error: (Zigate) ----> Line 680 in /usr/local/domoticz/var/plugins/Domoticz-Zigate/plugin.py, function Decode8000

zaraki673, 2018-04-14:
je ne sais pas du tout, si la centrale est un coordinateur, c est normal, de ce que j'ai compris on ne peut associer qu a un seul coordinateur (et je sais pas si deux coordinateur peuvent dialoguer )

pipiche38, 2018-04-16:
Lorsqu'on positionne les volets à 50% ou toutes autres valeures que 0% (Closed) et (100%) Ouvert, cette information ne remonte pas coté Domoticz . Je pense que c'e sle meme problème que ce que j'avais remonté un peu avant sur le On/off.

zaraki673, 2018-04-16:
quand tu dis tu positionnes, c est depuis domoticz ou depuis la télécommande ?
si depuis la télécommande, elle envoie que on/off et toggle comme ordre il me semble, sinon je veux bien des trames que tu reçois de ta télécommande qd tu définis un %
je penses qu il faudrait en fait que j interroges le volet pour connaitre ça position toutes les X minutes, si c'est possible d'avoir un % en retour ... (ou une valeur interprétable en %)

zaraki673, 2018-04-16:
ah et actuellement je n'utilises que le cluster lvlcontrol pour communiquer avec le volet, il y a aussi le cluster on/off (je sais pas comment ça fonctionne du coup je l utilise pas) c'est sur ce cluster que la télécommande dois communiquer
si tu peux me dire ce que tu reçois quand tu appuies sur la télécommande (et ce que se passe coté volet) ça pourras m'aider :D 

merci ;)

pipiche38, 2018-04-16:
En fait, il faut refaire la config de la commande Zoe (centrale) pour qu’elle intègre elle aussi le réseau coordonnée par la Zigate …

pipiche38, 2018-04-16:
Ci-joint un extrait des logs domoticz avec le plugin Zigate en Full debug

- 16:24 j'ai fait une descente du volet (via la télécomande)

- 16:26 j'ai fait une descente du volets ( à 50% ), puis je l'ai reouvert à 100%
[domoticz.log](https://github.com/sasu-drooz/Domoticz-Zigate/files/1915730/domoticz.log)

zaraki673, 2018-04-16:
je vois bien tes commandes de 16:26, mais pas de trames de ta teleco à 16h24 (y a pas mal de trames de decouverte de devices mais pas de commande d'ordre :( )

pipiche38, 2018-04-17:
Real voudrait dire que la télécommande pilote directement le volet, ce qui me parait plausible.

Ce soir j’essaie avec la commande ZOE. Et je reviens vers toi.

pipiche38, 2018-04-17:
> thread [2].
> Links:

Salut,

J'ai échangé avec la personne qui a développé la Zigate et il confirme 
que les télécommandes pilotes les volets en direct. Donc normal que la 
Zigate ne voit rien.

Du coup à part faire du polling pour récupérer l'info de façon régulière 
il n'y pas de solution :-(

zaraki673, 2018-04-17:
> issuecomment-381768256
> auth/AH6FWivad072np_2vSX2BX_5PEtMcd0qks5tpRjPgaJpZM4TS60u

Bon ben on va mettre en place une fonction qui recupere l état dans le
heartbeat :D (un peu comme le raz des motion :P )

faut juste trouver quelle commande renvois l'état :P

pipiche38, 2018-05-22:
Peut-etre qu'il s'agit du cluster Attribute ?

pipiche38, 2018-05-22:
D'après la documentation Profalux et le message de Fred (concepteur de la Zigate ). J'essaie d'investiguer de mon coté , mais j'ai du mal à comprendre ce que fait la fonction onHearbeat(self) actuelle.

7.6 Cluster Level control 
7.6.1 Les attributs 
 
Nom de l'attribut                    Commentaire Current
level                                Indique le niveau courant de l’élément (valeur entre 0 et 255).
On level                             Indique le niveau à atteindre pour une commande ON
 
 
 
Normalement, mais si vous effectuez une requête "Read Attribute"
* sur le Cluster 0x0008 avec l'attribut 0x0000 --> vous avez le niveau courant de l'élément (entre 0 et 255)
* sur le cluster 0x0008 avec l'attribut 0x0011 --> vous avezle niveau à atteindre pour une commande ON

Vincent4200, 2018-05-25:
Bonjour pipiche38 et zaraki673 , je vous suis attentivement car possédant des volets profalux j'essaye de les intégrer également à domoticz, j'ai bien réussi à ajouter le volet coordinateur sur la zigate mais je rame un peu plus pour ajouter les autres, quelle est la bonne méthode pour l'ajout des autres volets? Ouverture du réseau sur le coordinateur et sur la zigate puis ajout des volets supplémentaires ou ouverture simplement du réseau zigate et ajout des autres volets? J'ai fais plusieurs tests le volet répond bien à mes actions mais ils ne remontent pas dans domoticz et je n'ai pas l'impression de les voir dans les logs. Merci.

pipiche38, 2018-05-25:
Je suppose que quand tu dis 

>  mais ils ne remontent pas dans domoticz

Tu entends par cela , que si tu fais une manip directement avec ta télécomande, il n'y a pas de changemet d'état du coté de Domoticz . C'est effectivement ce que j'avais remonté, et il faudrait comme il est suggéré plus haut d'implémenter dans le onHeartbeat du polling d'information en provenance des volets.

Vincent4200, 2018-05-25:
Merci de ta réponse, en effet je ne vois pas l'état du volet quand il est manipulé par la télécommande mais j'avais vu que tu parlais déjà de ce problème. Mon problème est de pouvoir relier tous mes volets sur la zigate, j'ai bien réussi à relier celui qui est coordinateur de l 'installation mais j'ai du mal avec les suivants et je voulais savoir quel procédure avais tu effectué pour relier chaque volet? reset obligatoire de chaque télécommande? ou peut on les ajouter en ouvrant le réseau de la zigate et/ou de la télécommande coordinatrice? Merci

pipiche38, 2018-05-25:
Je n'avais pas pércuté qu'il y avait un volet coordinateur. 

Dans mon cas, je ne crois pas avoir fait un reset de chaque télécommande.

La Zigate est en mode "Permit Join" pendant environ 255 secondes. c'est le paramètre qui "Permit join time on start (between 0 et 255) " lorsque tu ajoutes le HW Zigate. Si je ne me trompe pas faire un disable/enable du HW remet le compteur à 0 ;-)

Vincent4200, 2018-05-25:
ok et quand la zigate est en mode permit join tu fais une manipulation particulière sur les télécommandes des volets ? Merci de ta réactivité.

pipiche38, 2018-05-25:
Oui, il faut faire l'appairage du volet

    Retourner votre télécommande
    Appuyer 1 fois sur R
    Appuyer ensuite sur la flèche du haut

Il me semble que ca suffisait... Sinon la prcédure entière telle que définie 
http://zigate.fr/2018/02/03/association-des-volets-profalux-a-la-zigate/

Sinon, je me suis aussi aidé de la doc. profalux

Pas mal d'information ici: http://www.profalux-pro.com/download/1.%20Notices,%20Plans,%20Technique/1.%20Volets%20roulants/3.%20Moteurs%20Commandes%20et%20Accessoires/3.%20EMETTEUR%20TELECOMMANDES/Telecommande%20ZOE%20NB046.pdf

Après si tu as une commande Zoe elle ne pilotera plus aucun volets. Il va falloir la faire rentrer dans le réseau
http://www.profalux-pro.com/notice/la-telecommande-generale-zoe-n-a-pas-trouve-d-installation-ndep012/

pipiche38, 2018-07-04:
@zaraki673 comment puis-je aider pour mettre en place la fonction heartbit et ainsi recupérer les informations de Level du volet ?

pipiche38, 2018-08-01:
Will be addressed by #105

closed 2018-08-01 by pipiche38