zigbeefordomoticz/Domoticz-Zigbee #81: (Zigate) 'onMessage' failed 'KeyError'.

Miloune, 2018-04-14:
Bonjour,

J'ai mis à jour le plugin ce matin et j'ai une nouvelle erreur, je ne sais pas trop à quoi c'est du ni quoi faire ?

```
2018-04-14 17:24:37.272 (Zigate) Calling message handler 'onMessage'.
2018-04-14 17:24:37.272 (Zigate) onMessage called
2018-04-14 17:24:37.272 (Zigate) ZigateDecode - decodind data : 0181021202103
2018-04-14 17:24:37.272 (Zigate) ZigateRead - decoded data : 01810200
2018-04-14 17:24:37.272 (Zigate) ZigateRead - Message Type : 8102, Data : , RSSI : 02, Length : 00, Checksum :
2018-04-14 17:24:37.272 (Zigate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 01810200
2018-04-14 17:24:37.272 (Zigate) Decode8102 - reception data : ClusterID : Attribut ID : Src Addr : Scr Ep:
2018-04-14 17:24:37.272 Error: (Zigate) 'onMessage' failed 'KeyError'.
2018-04-14 17:24:37.272 Error: (Zigate) ----> Line 307 in /root/domoticz/plugins/domoticz-zigate/plugin.py, function onMessage
2018-04-14 17:24:37.272 Error: (Zigate) ----> Line 121 in /root/domoticz/plugins/domoticz-zigate/plugin.py, function onMessage
2018-04-14 17:24:37.272 Error: (Zigate) ----> Line 372 in /root/domoticz/plugins/domoticz-zigate/plugin.py, function ZigateDecode
2018-04-14 17:24:37.272 Error: (Zigate) ----> Line 616 in /root/domoticz/plugins/domoticz-zigate/plugin.py, function ZigateRead
2018-04-14 17:24:37.272 Error: (Zigate) ----> Line 777 in /root/domoticz/plugins/domoticz-zigate/plugin.py, function Decode8102
2018-04-14 17:24:37.272 Error: (Zigate) ----> Line 1193 in /root/domoticz/plugins/domoticz-zigate/plugin.py, function ReadCluster
```

zaraki673, 2018-04-14:
verifie ton devicelist.txt il doit y avoir une ligne quasi vide (en tout cas sans shortadress), sinon supprime le et relance domoticz

Miloune, 2018-04-15:
Ca m'a l'air bon, je n'ai plus l'erreur, par contre, j'ai un autre souci, je ne sais pas si c'est lié ou non mais j'ai l'impression que mes devices ne se mettent plus à jour autant qu'avant:
[image]

J'ai un script LUA qui utilise le détecteur de mouvement, mais il ne fonctionne plus car mon device ne se met plus à jour côté Domoticz : 
[image]
[image]

Mat2Vence, 2018-04-16:
Même erreur de mon côté. Dans mon DeviceList.txt j'ai seulement mes deux capteurs de températures... rien de plus.
Dans le workaround proposé, il faut supprimé le contenu du DeviceList.txt et refaire l'association?

zaraki673, 2018-04-16:
dans les versions précédente a chaque réception d'une mise a jour de la
batterie, tout le capteur était mis à jour, même si la température n'avait
pas changé
c'est le capteur qui envoie la trames de température ou humidité,
uniquement si celle ci change

on peut verifier en mettant un capteur sur la zigate et un autre sur la
gateway et en les plaçant au même endroit, les deux graphs devrait être a
peu près identiques (ce qui est le cas chez moi)
 (zigate en haut - gateway en bas)
​

Mat2Vence, 2018-04-16:
Je vais faire un essai en comparant avec le gateway.
Par contre je crois que j'ai fait une boulette. J'ai supprimé mon capteur dans Domoticz, et j'arrive pas à la réassocier via la procédure standard... y'a moyen de se remettre d'aplomb sans avoir à faire un reset de la Zigate?

zaraki673, 2018-04-16:
oui tu edites ton devicelist.txt et pour le device en question tu modifie
le champs status de "inDB" à "" et tu redémarres le plugin ou domoticz. ton
device sera de nouveau creer au premier heartbeat
;)

Mat2Vence, 2018-04-16:
J'ai retrouvé mon capteur.. par contre sur un autre capteur de température que j'instancie avec la Zigate sur une instance de Domoticz, et via le gateway sur une autre instance, ça semble confirmer que je n'ai pas de mise à jour des data pour la version Zigate.
[image]
J'ai soufflé sur le capteur pour faire varier la température.

Mat2Vence, 2018-04-16:
Ouep j'ai ça

2018-04-17 00:26:01.605 (Zigate) Calling message handler 'onMessage'.
2018-04-17 00:26:01.605 (Zigate) onMessage called
2018-04-17 00:26:01.605 (Zigate) ZigateDecode - decodind data : 0180021002100215a402100210021021021003
2018-04-17 00:26:01.605 (Zigate) ZigateRead - decoded data : 0180000005a4000000210003
2018-04-17 00:26:01.605 (Zigate) ZigateRead - Message Type : 8000, Data : 00000021, RSSI : 00, Length : 0005, Checksum : a4
2018-04-17 00:26:01.605 (Zigate) ZigateRead - MsgType 8000 - reception status : 0180000005a4000000210003
2018-04-17 00:26:01.605 (Zigate) Decode8000 - Reception status : Success, SQN : 21, Message :
2018-04-17 00:26:01.605 (Zigate) Processing 'onMessageCallback' message
2018-04-17 00:26:01.605 (Zigate) Calling message handler 'onMessage'.
2018-04-17 00:26:01.605 (Zigate) onMessage called
2018-04-17 00:26:01.605 (Zigate) ZigateDecode - decodind data : 0180021002100215a602100210021023021003
2018-04-17 00:26:01.605 (Zigate) ZigateRead - decoded data : 0180000005a6000000230003
2018-04-17 00:26:01.606 (Zigate) ZigateRead - Message Type : 8000, Data : 00000023, RSSI : 00, Length : 0005, Checksum : a6
2018-04-17 00:26:01.606 (Zigate) ZigateRead - MsgType 8000 - reception status : 0180000005a6000000230003
2018-04-17 00:26:01.606 (Zigate) Decode8000 - Reception status : Success, SQN : 23, Message :
2018-04-17 00:26:01.606 (Zigate) Processing 'onMessageCallback' message
2018-04-17 00:26:01.606 (Zigate) Calling message handler 'onMessage'.
2018-04-17 00:26:01.606 (Zigate) onMessage called

Tu es sur quelle version de Domoticz?
Ensuite j'ai des messages qui laisse entendre qu'il y a de la vie, mais ça bouge pas au niveau Domoticz..

2018-04-17 00:28:20.871 (Zigate) HB count = 13
2018-04-17 00:28:30.839 (Zigate) Pushing 'onHeartbeatCallback' on to queue
2018-04-17 00:28:30.889 (Zigate) Processing 'onHeartbeatCallback' message
2018-04-17 00:28:30.889 (Zigate) Calling message handler 'onHeartbeat'.
2018-04-17 00:28:30.889 (Zigate) ListOfDevices : {'2bcf': {'Status': 'inDB', 'Battery': {}, 'Heartbeat': '15', 'MacCapa': '80', 'Ep': {'01': {'0405': 65.7, '0403': 977.5, '0402': 22.4, '0000': {}}}, 'Model': 'lumi.weather', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d0001d5c563', 'Type': 'Temp/Humi/Baro', 'RIA': '10'}, '4766': {'Model': 'lumi.sensor_ht', 'Battery': 89, 'Heartbeat': '145', 'MacCapa': '80', 'ProfileID': {}, 'Status': 'inDB', 'Ep': {'01': {'0405': 66.0, '0402': 19.7, '0000': {}}}, 'ZDeviceID': {}, 'IEEE': '00158d0001100338', 'Type': 'Temp/Humi', 'RIA': '23'}}
2018-04-17 00:28:30.892 (Zigate) HB count = 14
2018-04-17 00:28:40.841 (Zigate) Pushing 'onHeartbeatCallback' on to queue
2018-04-17 00:28:40.860 (Zigate) Processing 'onHeartbeatCallback' message
2018-04-17 00:28:40.860 (Zigate) Calling message handler 'onHeartbeat'.
2018-04-17 00:28:40.860 (Zigate) ListOfDevices : {'2bcf': {'Status': 'inDB', 'Battery': {}, 'Heartbeat': '16', 'MacCapa': '80', 'Ep': {'01': {'0405': 65.7, '0403': 977.5, '0402': 22.4, '0000': {}}}, 'Model': 'lumi.weather', 'ProfileID': {}, 'ZDeviceID': {}, 'IEEE': '00158d0001d5c563', 'Type': 'Temp/Humi/Baro', 'RIA': '10'}, '4766': {'Model': 'lumi.sensor_ht', 'Battery': 89, 'Heartbeat': '146', 'MacCapa': '80', 'ProfileID': {}, 'Status': 'inDB', 'Ep': {'01': {'0405': 66.0, '0402': 19.7, '0000': {}}}, 'ZDeviceID': {}, 'IEEE': '00158d0001100338', 'Type': 'Temp/Humi', 'RIA': '23'}}

Mat2Vence, 2018-04-16:
Je viens de faire un essai avec la beta V3.9268
Build Hash: 27c7e89f
Build Date: 2018-04-16 20:14:39

pas mieux...

zaraki673, 2018-04-16:
même version de domoticz et plugin et 2.3.5

et "la vie" que tu as est uniquement local a domoticz, y a pas de réception
de trames :( pourtant en soufflant sur le capteur tu devrait bien en avoir
(ou même en appuyant une fois sur le bouton ... )

Mat2Vence, 2018-04-16:
Est-ce que tu aurais un lien qui fonctionne pour downloader ZGWUI? Je vais refaire des tests de bases mais j'ai plus l'install et le lien donné sur le wiki de Zigate est mort.

zaraki673, 2018-04-16:
https://github.com/fairecasoimeme/ZiGate/tree/master/Tools

;)

Mat2Vence, 2018-04-17:
Ouep je suis un boulet... j'ai enlevé mon commentaire après avoir trouver en cherchant un chouilla!
Sinon, j'ai relancé une procédure d'ajout de mes deux sensors dans domoticz... cette fois-ci ça semble fonctionner. Par contre, je ne les retrouve pas dans devicelist.txt? j'ai toujours les anciennes références.

zaraki673, 2018-04-17:
pas possible, si elle remonte ds domoticz elle sont ds le devicelist.txt
(enfin ce fichier est ecrit que tout les 200 HB soit 2000secondes :P, du
coup faut attendre un peu une fois la detection faite
(j ecris pas plus souvent le fichier pour les personnes sur rpi comme moi,
ca evite de flinguer la sd a lui ecrire a tout va dessus)

:D

Mat2Vence, 2018-04-17:
En effet, je les ai à présent.
Cela dit, sur les deux capteurs de température que j'avais configurer hier soir, à la remise en marche tout à l'heure du système, j'en ai qu'un seul qui se met à jour.
Va falloir que je creuse.

SpaWnOL68, 2018-06-09:
Toujours les messages d'erreur chez moi.. j'ai vérifié la devicelist et rien d'anormal à priori.. 
Je ne sais pas si cela est lié mais mes switch aqara fonctionne super mal.. Ca peut fonctionner comme parfois il ne reagit pas... please HELP
Voici mes logs

(Zigate) 'onMessage' failed 'KeyError'.
2018-06-09 09:38:55.416 Error: (Zigate) ----> Line 307 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-06-09 09:38:55.416 Error: (Zigate) ----> Line 121 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-06-09 09:38:55.417 Error: (Zigate) ----> Line 372 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ZigateDecode
2018-06-09 09:38:55.417 Error: (Zigate) ----> Line 616 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ZigateRead
2018-06-09 09:38:55.417 Error: (Zigate) ----> Line 777 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function Decode8102
2018-06-09 09:38:55.417 Error: (Zigate) ----> Line 1190 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ReadCluster
2018-06-09 09:40:13.316 Error: (Zigate) 'onMessage' failed 'IndexError':'string index out of range'.
2018-06-09 09:40:13.316 Error: (Zigate) ----> Line 307 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-06-09 09:40:13.316 Error: (Zigate) ----> Line 121 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-06-09 09:40:13.316 Error: (Zigate) ----> Line 372 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ZigateDecode
2018-06-09 09:40:13.316 Error: (Zigate) ----> Line 616 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ZigateRead
2018-06-09 09:40:13.316 Error: (Zigate) ----> Line 777 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function Decode8102
2018-06-09 09:40:13.316 Error: (Zigate) ----> Line 1252 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ReadCluster

Miloune, 2018-06-09:
> Ca peut fonctionner comme parfois il ne reagit pas... please HELP
Moi aussi, du coup, j'ai fais un CRON qui reboot mon Domotcz a minuit chaque jour, et j'ai aussi activé le data timeout a 15 min.

SpaWnOL68, 2018-06-10:
J'ai programmé le reboot de Domoticz. 
En revanche, j'ai de nouveaux plus rien qui marche avec la Zigate... les switchs ne fonctionnent toujours pas, mon detecteur d'ouverture ne fonctionne plus et mes capteurs de temperature ne sont pas mis à jour. 
Je fais que de reprogrammer en permanence et ca tient 15j.

pipiche38, 2018-08-10:
@SpaWnOL68 est-ce que le problème existe toujours ?

Si oui, as-tu la possibilité de refaire le test avec la dernière version du plugin et également mettre en mode debug 'Python only' afin de voir un peu mieux ce qui se passe dans le code ?

ReadCluster qui est la fonction à l'origine du problème reporté le 9 Juin à quelque peu changé et a été rendu un peu plus solide au erreurs.

pipiche38, 2018-09-10:
Je propose de clore ce ticket.
La version 2.4.0 qui a été releasé recèmenent fixe beaucoup de problème de ce style.

Si le problème reapparaite, merci d'ouvrir un nouveau ticket

closed 2018-09-10 by pipiche38