zigbeefordomoticz/Domoticz-Zigbee #83: Error in code, message in domoticz logs

duck987, 2018-04-22:
I sometimes get this error message in logs:

> Error: (Zigate) 'onCommand' failed 'UnboundLocalError':'local variable 'ClusterSearch' referenced before assignment'.
> Error: (Zigate) ----> Line 300 in /domoticz/plugins/Domoticz-Zigate/plugin.py, function onCommand
> Error: (Zigate) ----> Line 139 in /domoticz/plugins/Domoticz-Zigate/plugin.py, function onCommand
> 

I'm using april 10 version.
Line 139 is:
```
for tmpEp in self.ListOfDevices[Devices[Unit].DeviceID]['Ep'] :
line 139==>    if ClusterSearch in self.ListOfDevices[Devices[Unit].DeviceID]['Ep'][tmpEp] : #switch cluster
        EPout=tmpEp
        EPfound=True
    else :
        EPfound=False
```
Line 300 is:
```
def onCommand(Unit, Command, Level, Hue):
    global _plugin
line 300==>    _plugin.onCommand(Unit, Command, Level, Hue)
```
Please not I'm only using Xiaomi PIR human body movement, window door sensor, temperature.

zaraki673, 2018-04-22:
hi,

if you have edit your device for changing pictures or something else, it 's
a domoticz bug, who clear all plugin key in dbb

you will have to delete the device, and add it again (or satus 's value in
devicelist.txt to nothing (against inDB)

duck987, 2018-05-01:
Hi zaraki673,
I've corrected the error. You'll need to add this variable initialization (on latest version, it's at line 134):
```
                ClusterSearch="" # Add this line to initialize the variable
                if Dtypename=="Switch" or Dtypename=="Plug" or Dtypename=="MSwitch" or Dtypename=="Smoke" or Dtypename=="DSwitch" or Dtypename=="Button" or Dtypename=="DButton":
                        ClusterSearch="0006"
                if Dtypename=="LvlControl" :
                        ClusterSearch="0008"
                if Dtypename=="ColorControl" :
                        ClusterSearch="0300"
                if ClusterSearch == "": # Add these lines to identify the missing entries, it creates an error.txt file with missing entries 
                        with open(Parameters["HomeFolder"]+"error.txt", 'a') as file_error:
                                file_error.write("Not found: Dtypename = " + Dtypename)
                        myfile.close()
```
In my version, I've also added a log to determine which entry is missing. In my install, it is missing an if statement for Dtypename == "Motion"

zaraki673, 2018-05-14:
hi,

thks, will try your fix

pipiche38, 2018-08-01:
Here is what I would suggest as I don't think this make sense to create a file, let simply log the error in the log file
```

                ClusterSearch=""
                if Dtypename=="Switch" or Dtypename=="Plug" or Dtypename=="MSwitch" or Dtypename=="Smoke" or Dtypename=="DSwitch" or Dtypename=="Button" or Dtypename=="DButton":
                        ClusterSearch="0006"
                if Dtypename=="LvlControl" :
                        ClusterSearch="0008"
                if Dtypename=="ColorControl" :
                        ClusterSearch="0300"
                if ClusterSearch == "": #
                        Domoticz.Log('onCommand didn't find any cluster applicable to the current configuration for Dtypename = ' + Dtypename )

```

closed 2018-09-10 by pipiche38