zigbeefordomoticz/Domoticz-Zigbee #84: onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given

pipiche38, 2018-04-23:
I have move from Synology to Raspberry Pi with Domoticz V3.8153 (c)2012-2017 GizMoCuz

and I'm getting those error messages:

Apr 23 08:01:50 pi3.localdomain domoticz[3986]: 2018-04-23 08:01:50.066  (Zigate) Initialized version 2.3.5, author 'zaraki673'
Apr 23 08:01:50 pi3.localdomain domoticz[3986]: 2018-04-23 08:01:50.370  Error: (Zigate) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
Apr 23 08:01:50 pi3.localdomain domoticz[3986]: 2018-04-23 08:01:50.371  Error: (Zigate) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
Apr 23 08:01:50 pi3.localdomain domoticz[3986]: 2018-04-23 08:01:50.471  Error: (Zigate) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
Apr 23 08:01:56 pi3.localdomain domoticz[3986]: 2018-04-23 08:01:56.733  Error: (Yamaha) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
Apr 23 08:01:56 pi3.localdomain domoticz[3986]: 2018-04-23 08:01:56.733  Error: (Yamaha) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.

zaraki673, 2018-04-23:
try domoticz beta ;)

don't work on stable (same for yamaha plugin's ;) )

pipiche38, 2018-04-23:
I do confirm that moving to Beta make it work.

As well as for Yamaha ;-)

closed 2018-04-23 by pipiche38