zigbeefordomoticz/Domoticz-Zigbee #85: The start of the plugin is triggering some move of the rolling shutter

pipiche38, 2018-04-25:
Every time the plugin is started, it triggers a move of the rolling shutter. For instance it happends when you disable/enable the plugin or if you restart domoticz.

I don't know if that is related to the fact that its open the network for pairing devices, but somehow this is impacting the calibration of the shutter, so it would be great if this could be disabled, or at least that we have an option to disable.

For instance we should expect regular reboot of the system (patching ...) while I'm not expecting to add on a regular basic new devices to the Zigate system

zaraki673, 2018-05-14:
Hi,

have you try to set Permit join time to 0 ?

pipiche38, 2018-05-15:
Yes, I did . This happen every-time you enable the Zigate plugin !

pipiche38, 2018-05-15:
Je vais essayer de mettre des temps de pause ( 3 secondes ) entre chaque etape du ZigateConf en mode debug. Comme ca je vais voir rapidement.

Je regarde comment cela ce fait, car je ne connais pas trop python et je reviens avec un feedback.

pipiche38, 2018-05-15:
A priori c'est bien la fonction Disover qui génère cela. Et meme si on l'appelle avec 0. Donc il faudrait que dans le code il y ai un test si Mode2!=0 alors on fait un sendZigateCmd("0049",.... Sinon on ne fait rien.
Ne connaissant pas trop python, je préfère laisser ça au pro. 

Dans mon cas, j'ai mis cette ligne ( 353 ) en commentaire
        #sendZigateCmd("0049","FFFC" + hex(int(Parameters["Mode2"]))[2:4] + "00")

Et je n'ai plus le problème.

pipiche38, 2018-05-15:
if Parameters["Mode2"] != "0":
                sendZigateCmd("0049","FFFC" + hex(int(Parameters["Mode2"]))[2:4] + "00")

zaraki673, 2018-05-26:
merci ;)

closed 2018-05-26 by pipiche38