zigbeefordomoticz/Domoticz-Zigbee #86: Error: (Zigate) 'onMessage' failed 'ValueError':'invalid literal for int() with base 16: '''.

pipiche38, 2018-05-09:
Today after restarting Domoticz I've got those error message in the logs

May 09 21:32:34 pi3 domoticz[19778]: 2018-05-09 21:32:34.349  (Zigate) Entering work loop.
May 09 21:32:34 pi3 domoticz[19778]: 2018-05-09 21:32:34.350  (Zigate) Initialized version 2.3.5, author 'zaraki673'
May 09 21:32:34 pi3 domoticz[19778]: 2018-05-09 21:32:34.807  Error: (Zigate) 'onMessage' failed 'ValueError':'invalid literal for int() with base 16: '''.
May 09 21:32:34 pi3 domoticz[19778]: 2018-05-09 21:32:34.807  Error: (Zigate) ----> Line 307 in /var/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
May 09 21:32:34 pi3 domoticz[19778]: 2018-05-09 21:32:34.807  Error: (Zigate) ----> Line 121 in /var/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
May 09 21:32:34 pi3 domoticz[19778]: 2018-05-09 21:32:34.807  Error: (Zigate) ----> Line 372 in /var/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function ZigateDecode
May 09 21:32:34 pi3 domoticz[19778]: 2018-05-09 21:32:34.807  Error: (Zigate) ----> Line 442 in /var/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function ZigateRead
May 09 21:32:34 pi3 domoticz[19778]: 2018-05-09 21:32:34.807  Error: (Zigate) ----> Line 680 in /var/lib/domoticz/plugins/Domoticz-Zigate/plugin.py, function Decode8000

thiklop, 2018-06-20:
J'ai les mêmes problèmes avec mes capteurs de mouvement/lumière. Je peux faire un copié/collé des logs un peu plus complet au besoin

thiklop, 2018-07-08:
J'ai écrit un patch pour ce problème et d'autres mais je n'arrive pas à le soumettre. Comment puis-je l'envoyer ?

pipiche38, 2018-07-08:
@thiklop je ne sais pas si c'est le plus simple, mais la façon dont je fais est la suivante:
- allez dans le code source (github)
- ouvrir le fichier
- cliquer sur l'icone 'crayon' pour pouvoir entrer en mode edition
- faire les modifs
- Mettre les commentaires (c'est en bas de la page )
- Puis après il faut créer une 'Pull Request' qui permet de soumettre la modif

thiklop, 2018-07-08:
Je vais essayer mais ça va être long ...

thiklop, 2018-07-29:
Corrigé avec le commit de mi-juillet

closed 2018-07-29 by thiklop