zigbeefordomoticz/Domoticz-Zigbee #87: 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'

tcharp38, 2018-05-10:
Hello,
on a fresh new domoticz installation + brand new zigate, I'm unable to make Xaomi temp sensor to be detected. I've found this error in the log.

Any clues ?
Thanks

A more complete log

(Zigate) ListOfDevices : {}
2018-05-10 15:11:56.490  (Zigate) HB count = 0
2018-05-10 15:12:06.490  (Zigate) Calling message handler 'onHeartbeat'.
2018-05-10 15:12:06.490  (Zigate) ListOfDevices : {}
2018-05-10 15:12:06.490  (Zigate) HB count = 1
2018-05-10 15:12:16.491  (Zigate) Calling message handler 'onHeartbeat'.
2018-05-10 15:12:16.491  (Zigate) ListOfDevices : {}
2018-05-10 15:12:16.491  (Zigate) HB count = 2
2018-05-10 15:12:22.366  (Zigate) Received 42 bytes of data:.
2018-05-10 15:12:22.366  (Zigate)     01 81 02 12 02 10 19 db 87 32 42 02 11 02 10 02 10 02 10 02     .ü.....█ç2B.........
2018-05-10 15:12:22.366  (Zigate)     15 02 10 42 02 10 02 1c 6c 75 6d 69 2e 77 65 61 74 68 65 72     ...B....lumi.weather
2018-05-10 15:12:22.366  (Zigate)     b7 03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     À.
2018-05-10 15:12:22.366  (Zigate) Calling message handler 'onMessage'.
2018-05-10 15:12:22.366  Error: (Zigate) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2018-05-10 15:12:25.616  (Zigate) Received 164 bytes of data:.
2018-05-10 15:12:25.616  (Zigate)     01 81 02 12 02 10 02 1f 1e 88 32 42 02 11 02 14 02 12 02 10     .ü.......ê2B........
2018-05-10 15:12:25.616  (Zigate)     02 10 02 10 29 02 10 02 12 02 19 8c c3 03 01 81 02 12 02 10     ....)......î├..ü....
2018-05-10 15:12:25.616  (Zigate)     02 1f c9 89 32 42 02 11 02 14 02 15 02 10 02 10 02 10 21 02     ..╔ë2B............!.
2018-05-10 15:12:25.616  (Zigate)     10 02 12 18 44 c3 03 01 81 02 12 02 10 02 1f 42 8a 32 42 02     ....D├..ü......Bè2B.
2018-05-10 15:12:25.616  (Zigate)     11 02 14 02 13 02 10 02 10 02 10 29 02 10 02 12 02 13 d9 c3     ...........)......┘├
2018-05-10 15:12:25.616  (Zigate)     03 01 81 02 12 02 10 02 1e 70 8a 32 42 02 11 02 14 02 13 02     ..ü......pè2B.......
2018-05-10 15:12:25.616  (Zigate)     10 14 02 10 28 02 10 02 11 ff c3 03 01 81 02 12 02 10 02 1f     ....(.... ├..ü......
2018-05-10 15:12:25.616  (Zigate)     2d 8a 32 42 02 11 02 14 02 13 02 10 10 02 10 29 02 10 02 12     -è2B...........)....
2018-05-10 15:12:25.616  (Zigate)     26 83 c3 03 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     &â├.
2018-05-10 15:12:25.616  (Zigate) Calling message handler 'onMessage'.
2018-05-10 15:12:25.616  Error: (Zigate) 'onMessage' failed 'TypeError':'onMessage() takes 2 positional arguments but 4 were given'.
2018-05-10 15:12:26.491  (Zigate) Calling message handler 'onHeartbeat'.
2018-05-10 15:12:26.491  (Zigate) ListOfDevices : {}
2018-05-10 15:12:26.491  (Zigate) HB count = 3

davidllerena, 2018-06-17:
Hello,

I have no dedicate a lot of time to analize the issue, but...

As a workaround, I have modified the onMessage function adding two extra args:
`def onMessage(self, Connection, Data, Extra, ExtraExtra):`
and 
```
def onMessage(Connection, Data, Extra, ExtraExtra):
        global _plugin
        _plugin.onMessage(Connection, Data, Extra, ExtraExtra)
```

This do not resolve the issue, but Xiaomi sensor appears in devices. Only Barometer sensor show abnormal value, but Temp-Baro show normal value.

Unfortunally Domoticz crashes... :(

pipiche38, 2018-08-09:
Is that issue still alive ?

tcharp38, 2018-08-13:
Hello.
Sorry. I did not find time so far to retry but I'll do it this week.
Back soon

pipiche38, 2018-08-13:
If you still have the issue. Could you : 
(1) Use the Developement branch
(2) Enable Debuging "Python Only"

and then report the logs

pipiche38, 2018-09-10:
Close this for now. 
Please come back if you still have the problem.

closed 2018-09-10 by pipiche38