zigbeefordomoticz/Domoticz-Zigbee #89: Already associated devices not created into Domoticz devices

pipiche38, 2018-05-23:
Suite à une erreur, je n'ai plus mais devices dans Domoticz. J'ai regardé dans la table devicesstatus et effectivement il n'y a plus rien.

Cependant les volets sont bien associés à la Zigate, mais lorsque je cré un HW Zigate ceux-ci ne sont pas créés dans Domoticz

Ci après un extrait des log. On peut voir que les volets sont bien associés. Il sont bien listés, mais rien n'est créé dans Domoticz

May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.470  (Zigate) HearBeat - creating device id : ccb3
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.470  (Zigate) HB count = 126
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.770  (Zigate) Pushing 'ReadEvent' on to queue
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.771  (Zigate) Pushing 'ReadEvent' on to queue
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.771  (Zigate) Processing 'ReadEvent' message
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.771  (Zigate) Received 5 bytes of data
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.771  (Zigate) Pushing 'ReadEvent' on to queue
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.771  (Zigate) Pushing 'onMessageCallback' on to queue
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.771  (Zigate) Processing 'ReadEvent' message
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.771  (Zigate) Pushing 'ReadEvent' on to queue
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.771  (Zigate) Received 3 bytes of data
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.772  (Zigate)     10 02 13 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. ..     ...
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.772  (Zigate) Pushing 'onMessageCallback' on to queue
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.772  (Zigate) Processing 'ReadEvent' message
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.772  (Zigate) Received 2 bytes of data
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.772  (Zigate) Pushing 'onMessageCallback' on to queue
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.772  (Zigate) Processing 'onMessageCallback' message
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.772  (Zigate) Calling message handler 'onMessage'.
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.772  (Zigate) onMessage called
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.772  (Zigate) Processing 'ReadEvent' message
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.772  (Zigate) Received 4 bytes of data
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.773  (Zigate) Pushing 'onMessageCallback' on to queue
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.773  (Zigate) Processing 'onMessageCallback' message
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.773  (Zigate) Calling message handler 'onMessage'.
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.773  (Zigate) onMessage called
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.773  (Zigate) Processing 'onMessageCallback' message
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.773  (Zigate) Calling message handler 'onMessage'.
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.773  (Zigate) onMessage called
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.773  (Zigate) Processing 'onMessageCallback' message
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.773  (Zigate) Calling message handler 'onMessage'.
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.774  (Zigate) onMessage called
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.774  (Zigate) ZigateDecode - decodind data : 0187021102100213f3d00210a603
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.774  (Zigate) ZigateRead - decoded data : 0187010003f3d000a603
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.774  (Zigate) ZigateRead - Message Type : 8701, Data : d000, RSSI : a6, Length : 0003, Checksum : f3
May 24 00:24:55 pi3 domoticz[8146]: 2018-05-24 00:24:55.774  (Zigate) ZigateRead - MsgType 8701 - Reception Router discovery confirm : 0187010003f3d000a603
May 24 00:25:05 pi3 domoticz[8146]: 2018-05-24 00:25:05.454  (Zigate) Pushing 'onHeartbeatCallback' on to queue
May 24 00:25:05 pi3 domoticz[8146]: 2018-05-24 00:25:05.458  (Zigate) Processing 'onHeartbeatCallback' message
May 24 00:25:05 pi3 domoticz[8146]: 2018-05-24 00:25:05.458  (Zigate) Calling message handler 'onHeartbeat'.
May 24 00:25:05 pi3 domoticz[8146]: 2018-05-24 00:25:05.459  (Zigate) onHeartbeat called
May 24 00:25:05 pi3 domoticz[8146]: 2018-05-24 00:25:05.461  (Zigate) ListOfDevices : {'a6e8': {'Ep': {'01': {'0003': {}, '000a': {}, '0000': {}, '0006': {}, '0005': {}, '0004': {}, 'fc20': {}, 'fc21': {}, '0100': {}, '0008': {}, '0015': {}}}, 'Status': 'inDB', 'Heartbeat': '127', 'RIA': '10', 'Battery': {}, 'Model': 'shutter.Profalux', 'MacCapa': '8e', 'IEEE': '20918a0000038f6d', 'Type': {}, 'ProfileID': '0104', 'ZDeviceID': '0200'}, 'b218': {'Ep': {'01': {'fc21': {}, '0003': {}, '0000': {}, '000a': {}, '0006': {}, '0004': {}, '0005': {}, '0015': {}, 'fc20': {}, '0008': {}, '0100': {}}}, 'Status': 'inDB', 'Heartbeat': '127', 'RIA': '10', 'Battery': {}, 'Model': 'shutter.Profalux', 'MacCapa': '8e', 'IEEE': '20918a0000038f67', 'Type': {}, 'ProfileID': '0104', 'ZDeviceID': '0200'}, '1d81': {'Ep': {'01': {'fc21': {}, '0003': {}, '0000': {}, '000a': {}, '0006': {}, '0004': {}, '0005': {}, '0015': {}, 'fc20': {}, '0008': {}, '0100': {}}}, 'Status': 'inDB', 'Heartbeat': '127', 'RIA': '10', 'Battery': {}, 'Model': 'shutter.Profalux', 'MacCapa': '8e', 'IEEE': '20918a0000038f69', 'Type': {}, 'ProfileID': '0104', 'ZDeviceID': '0200'}, '1a19': {'Ep': {'01': {'0006': {}, '0004': {}, '0005': {}, '0015': {}, '0003': {}, '0000': {}, '000a': {}, 'fc21': {}, 'fc20': {}, '0008': {}, '0100': {}}}, 'Status': 'inDB', 'Heartbeat': '127', 'RIA': '10', 'Battery': {}, 'Model': 'shutter.Profalux', 'MacCapa': '8e', 'IEEE': '20918a0000038f6f', 'Type': {}, 'ProfileID': '0104', 'ZDeviceID': '0200'}, '2657': {'Ep': {'01': {'0015': {}, '0008': {}, '0100': {}, '0004': {}, '0005': {}, '0006': {}, 'fc20': {}, '0000': {}, '000a': {}, 'fc21': {}, '0003': {}}}, 'Status': 'inDB', 'Heartbeat': '127', 'RIA': '10', 'Battery': {}, 'Model': 'shutter.Profalux', 'MacCapa': '8e', 'IEEE': '20918a0000038f6b', 'Type': {}, 'ProfileID': '0104', 'ZDeviceID': '0200'}, 'e86c': {'Ep': {'01': {'0015': {}, '0008': {}, '0100': {}, '0004': {}, '0005': {}, '0006': {}, 'fc20': {}, '0000': {}, '000a': {}, 'fc21': {}, '0003': {}}}, 'Status': 'inDB', 'Heartbeat': '127', 'RIA': '10', 'Battery': {}, 'Model': 'shutter.Profalux', 'MacCapa': '8e', 'IEEE': '20918a0000038f65', 'Type': {}, 'ProfileID': '0104', 'ZDeviceID': '0200'}, '550f': {'Ep': {'01': {'000a': {}, '0000': {}, '0003': {}, 'fc21': {}, '0005': {}, '0004': {}, '0006': {}, '0100': {}, '0008': {}, 'fc20': {}, '0015': {}}}, 'Status': 'inDB', 'Heartbeat': '127', 'RIA': '10', 'Battery': {}, 'Model': 'shutter.Profalux', 'MacCapa': '8e', 'IEEE': '20918a0000038f63', 'Type': {}, 'ProfileID': '0104', 'ZDeviceID': '0200'}, '991c': {'Ep': {'01': {'0000': {}, '0003': {}, '0015': {}, '0800': {}, '0300': {}, '0400': {}, '0500': {}, '0600': {}}}, 'Status': 'inDB', 'Heartbeat': '127', 'RIA': '10', 'Battery': {}, 'Model': {}, 'MacCapa': '80', 'IEEE': '20918a0000038f6c', 'Type': 'ColorControl/Lux/Door/', 'ProfileID': '0104', 'ZDeviceID': '0201'}, 'fb20': {'Ep': {'01': {'0000': {}, '0003': {}, '0015': {}, '0800': {}, '0300': {}, '0400': {}, '0500': {}, '0600': {}}}, 'Status': 'inDB', 'Heartbeat': '127', 'RIA': '10', 'Battery': {}, 'Model': {}, 'MacCapa': '', 'IEEE': '20918a0000038f', 'Type': 'ColorControl/Lux/Door/', 'ProfileID': '0104', 'ZDeviceID': '0201'}, '4f82': {'Ep': {'01': {'0300': {}, '0015': {}, '0500': {}, '0900': {}, '0000': {}, '0003': {}, '0400': {}, '0600': {}}}, 'Status': 'inDB', 'Heartbeat': '127', 'RIA': '10', 'Battery': {}, 'Model': {}, 'MacCapa': '80', 'IEEE': '20918a0000038f71', 'Type': 'ColorControl/Door/Lux/', 'ProfileID': '0104', 'ZDeviceID': '0201'}, 'ccb3': {'Ep': {}, 'Status': '0045', 'Heartbeat': '1', 'RIA': '10', 'Battery': {}, 'Model': {}, 'MacCapa': '80', 'IEEE': '20918a0000038f71', 'Type': {}, 'ProfileID': {}, 'ZDeviceID': {}}}
May 24 00:25:05 pi3 domoticz[8146]: 2018-05-24 00:25:05.462  (Zigate) HearBeat - creating device id : ccb3
May 24 00:25:05 pi3 domoticz[8146]: 2018-05-24 00:25:05.463  (Zigate) HB count = 127
[... cut ...] MacCapa': '80', 'IEEE': '20918a0000038f71', 'Type': 'ColorControl/Door/Lux/', 'ProfileID': '0104', 'ZDeviceID': '0201'}, 'ccb3': {'Ep': {}, 'Status': '0045', 'Heartbeat': '1', 'RIA': '10', 'Battery': {}, 'Model': {}, 'MacCapa': '80', 'IEEE': '20918a0000038f71', 'Type': {}, 'ProfileID': {}, 'ZDeviceID': {}}}
May 24 00:25:35 pi3 domoticz[8146]: 2018-05-24 00:25:35.465  (Zigate) HearBeat - creating device id : ccb3
May 24 00:25:35 pi3 domoticz[8146]: 2018-05-24 00:25:35.466  (Zigate) HB count = 130

Nebukad93, 2018-05-24:
Bonjour,

Essaye de remplacer (ou mettre s'il est vide) dans ton fichier DeviceList.txt ceci : ```{}```

pipiche38, 2018-05-24:
Ah oui, j'avais oublié ce fichier ... Merci. Entre temps j'ai utilisé un restore et tout est rentré dans l'ordre.
Mais pour bien comprendre, il contient l'ensemble des devices associés par la Zigate. Donc en le mettant à 0, je les aurais vu réapparaitre dans la liste des Devices ?

Par contre dans ce fichier il y a quand même quelque chose d’étrange, voici les 4 dernières qui ne correspondent à aucun Sensor/Switch que j'ai. J'imagine que c'est la Zigate qui les a associé. Mais sur quelle base . 

991c : {'Ep': {'01': {'0000': {}, '0003': {}, '0015': {}, '0800': {}, '0300': {}, '0400': {}, '0500': {}, '0600': {}}}, 'Status': 'inDB', 'Heartbeat': '603', 'RIA': '10', 'Battery': {}, 'Model': {}, 'MacCapa': '80', 'IEEE': '20918a0000038f6c', 'Type': 'ColorControl/Lux/Door/', 'ProfileID': '0104', 'ZDeviceID': '0201'}
fb20 : {'Ep': {'01': {'0000': {}, '0003': {}, '0015': {}, '0800': {}, '0300': {}, '0400': {}, '0500': {}, '0600': {}}}, 'Status': 'inDB', 'Heartbeat': '603', 'RIA': '10', 'Battery': {}, 'Model': {}, 'MacCapa': '', 'IEEE': '20918a0000038f', 'Type': 'ColorControl/Lux/Door/', 'ProfileID': '0104', 'ZDeviceID': '0201'}
4f82 : {'Ep': {'01': {'0300': {}, '0015': {}, '0500': {}, '0900': {}, '0000': {}, '0003': {}, '0400': {}, '0600': {}}}, 'Status': 'inDB', 'Heartbeat': '603', 'RIA': '10', 'Battery': {}, 'Model': {}, 'MacCapa': '80', 'IEEE': '20918a0000038f71', 'Type': 'ColorControl/Door/Lux/', 'ProfileID': '0104', 'ZDeviceID': '0201'}
ccb3 : {'Ep': {}, 'Status': '0045', 'Heartbeat': '0', 'RIA': '10', 'Battery': {}, 'Model': {}, 'MacCapa': '80', 'IEEE': '20918a0000038f71', 'Type': {}, 'ProfileID': {}, 'ZDeviceID': {}}

closed 2018-07-04 by pipiche38