zigbeefordomoticz/Domoticz-Zigbee #95: Apparition de devices inconnus ...

pipiche38, 2018-07-04:
Comme ci-après dans le screenshot, j'ai des devices Zigate qui apparaissent au fil du temps !!!!

[image]

thiklop, 2018-07-14:
C'est marrant, on dirait que ton device est en mode association continuellement. Le device d'origine continue à fonctionner ou non après ces ajouts ?

pipiche38, 2018-08-03:
A priori ces devices sont sur la Zigate, car j'ai Ouvert l'appairage et ils sont revenus.
Peut-etre que le problème est sur la Zigate elle meme.

Il faut que je la branche sur un PC avec Windows, pour supprimer ces devices

closed 2018-08-03 by pipiche38