zigbeefordomoticz/Domoticz-Zigbee #100: onHeartbeat call several time in the sale minutes/seconds

pipiche38, 2018-07-16:
Don't know if it is expected, but when looking after the logs, I saw onHeartbeat call several time in the same seconds, while my understanding should be every 10s !

Jul 16 14:15:06 pi3 domoticz[22989]: 2018-07-16 14:15:06.784  (Zigate) onHeartbeat called
Jul 16 14:15:30 pi3 domoticz[22989]: 2018-07-16 14:15:30.258  (Zigate) onHeartbeat called
Jul 16 14:15:30 pi3 domoticz[22989]: 2018-07-16 14:15:30.315  (Zigate) onHeartbeat called
Jul 16 14:15:36 pi3 domoticz[22989]: 2018-07-16 14:15:36.784  (Zigate) onHeartbeat called
Jul 16 14:16:00 pi3 domoticz[22989]: 2018-07-16 14:16:00.258  (Zigate) onHeartbeat called
Jul 16 14:16:00 pi3 domoticz[22989]: 2018-07-16 14:16:00.314  (Zigate) onHeartbeat called

thiklop, 2018-07-29:
Il faudrait que sasu-drooz nous explique un peu plus le fonctionnement du plugin mais je me demande si les données du zigate ne sont pas stockées et traitées lors du Hearbeat, ce qui pourrait expliquer qu'on retrouve autant de ligen d'un coup. Ce qui correspondrait à la file d'attente

pipiche38, 2018-07-30:
@thiklop je pense que tu as raison et que le plugin bosse de façon asynchrone et traite donc des messages de la Zigate. 

Par contre, ma compréhension du plugin python et que onHeartbeat et en fait un call fait par domoticz de façon régulière, et ici ce n'est pas régulier du tout 👎 

https://www.domoticz.com/wiki/Developing_a_Python_plugin
Called every 'heartbeat' seconds (default 10) regardless of connection status. Heartbeat interval can be modified by the Heartbeat command. Allows the Plugin to do periodic tasks including request reconnection if the connection has failed. Warning: Setting this interval to greater than 30 seconds  will cause a 'thread seems to have ended unexpectedly' message to be  written to the log file every 30 seconds. The plugin will function  correctly but this message can not be suppressed because it is a  standard warning from Domoticz that a piece of hardware may have stopped  responding.If a plugin wants to heartbeat every 100 seconds it  should be coded with the heartbeat interval set to 10, 20 or 25 seconds  and only take action every 6th, 5th or 4th time the callback is invoked.
--

pipiche38, 2018-07-30:
@thiklop  , a priori en regardant la doc de développement d'un plugin python, 

- c'est onMessage() qui est chargé de processer les messages venant de la Zigate.

- c'est onCommand() qui est chargé de traduire les Commandes venant de Domoticz en commandes Zigate

- et donc onHearbeat() qui va de façon régulière faire des checks et/ou des commandes d'intérogation vers la Zigate

thiklop, 2018-07-30:
Tu as réactivé la sortie vers le log pour voir les "onHearBeat called" ?

pipiche38, 2018-07-30:
Oui, car j'essaie d'implémenter le "Control level" pour récupérer l'etat des volets.

Je pense que je vais finir par y arriver. 
-> Car il faut faire un Read Attribute avec le cluster 0x0008 et la valeur 0x0000
-> Jai ajouté une fonction de decodate 8100 (qui est la réponse attrbute)
-> J'ai ajouter un Cluster 0008 

Mon problème est de retrouver le DeviceID dans onHeartbeat() ainsi que EPin et EPout 

A suivre. Par contre toute piste est welcome ;-)

thiklop, 2018-07-30:
Je vais avoir du mal à t'aider je n'ai pas de volets en zigbee. Mais si tu
veux que je jette un œil quand tu auras un code à proposer, avec plaisir.

pipiche38, 2018-07-30:
Mon problème est la création de la sendZigateCmd()

[https://zigate.fr/wiki/commandes-zigate/](https://zigate.fr/wiki/commandes-zigate/) -> Cluster 0x0100

Voici le code que j'essaie d'ajouter dans le onHeartbeat()

```
                        if self.ListOfDevices[key]['ZDeviceID']=="0200" and self.ListOfDevices[key]['Heartbeat']>="60" :
                                # device id type shutter
                                Domoticz.Log("Request Control level of shutter via Read Attribute request : " + key)
                                EPin = "01"
                                EPout = "01"
                                direction = 0
                                manufacturer_id = 0
                                manufacturer_specific = 0
                                cluster = "0008"
                                attribute = "0000"
                                sendZigateCmd("0100", "02" + str(key) + EPin + EPout + cluster + str(direction) + str(manufacturer_specific) + str(manufacturer_id) + "1" + attribute )
                                self.ListOfDevices[key]['Heartbeat']="0"

```

Mais a priori ce n'est pas correct, car après avoir fait le Send , j'ai un retour avec comme status "Incorrect Parameters"

```

Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.802  (Zigate) ZigateRead - Message Type : 8000, Data : 0d000100, RSSI : 00, Length : 0005, Checksum : 89
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.802  (Zigate) Decode8000 - Reception status : Incorrect Parameters, SQN : 00, Message :
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.802  (Zigate) ZigateDecode - decodind data : 018002100210021589021d021002110210021003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.803  (Zigate) ZigateRead - decoded data : 0180000005890d0001000003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.803  (Zigate) ZigateRead - Message Type : 8000, Data : 0d000100, RSSI : 00, Length : 0005, Checksum : 89
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.803  (Zigate) Decode8000 - Reception status : Incorrect Parameters, SQN : 00, Message :
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.803  (Zigate) ZigateDecode - decodind data : 018002100210021589021d021002110210021003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.804  (Zigate) ZigateRead - decoded data : 0180000005890d0001000003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.804  (Zigate) ZigateRead - Message Type : 8000, Data : 0d000100, RSSI : 00, Length : 0005, Checksum : 89
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.804  (Zigate) Decode8000 - Reception status : Incorrect Parameters, SQN : 00, Message :
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.804  (Zigate) ZigateDecode - decodind data : 018002100210021589021d021002110210021003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.804  (Zigate) ZigateRead - decoded data : 0180000005890d0001000003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.805  (Zigate) ZigateRead - Message Type : 8000, Data : 0d000100, RSSI : 00, Length : 0005, Checksum : 89
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.805  (Zigate) Decode8000 - Reception status : Incorrect Parameters, SQN : 00, Message :
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.805  (Zigate) ZigateDecode - decodind data : 018002100210021589021d021002110210021003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.805  (Zigate) ZigateRead - decoded data : 0180000005890d0001000003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.805  (Zigate) ZigateRead - Message Type : 8000, Data : 0d000100, RSSI : 00, Length : 0005, Checksum : 89
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.806  (Zigate) Decode8000 - Reception status : Incorrect Parameters, SQN : 00, Message :
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.806  (Zigate) ZigateDecode - decodind data : 018002100210021589021d021002110210021003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.806  (Zigate) ZigateRead - decoded data : 0180000005890d0001000003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.806  (Zigate) ZigateRead - Message Type : 8000, Data : 0d000100, RSSI : 00, Length : 0005, Checksum : 89
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.807  (Zigate) Decode8000 - Reception status : Incorrect Parameters, SQN : 00, Message :
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.807  (Zigate) onMessage called
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.807  (Zigate) ZigateDecode - decodind data : 018002100210021589021d021002110210021003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.808  (Zigate) ZigateRead - decoded data : 0180000005890d0001000003
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.808  (Zigate) ZigateRead - Message Type : 8000, Data : 0d000100, RSSI : 00, Length : 0005, Checksum : 89
Jul 30 20:01:07 pi3 domoticz[21431]: 2018-07-30 20:01:07.808  (Zigate) Decode8000 - Reception status : Incorrect Parameters, SQN : 00, Message :
```

pipiche38, 2018-07-31:
# Data are :
                                # DeviceID 16bits
                                # EPin 8bits
                                # EPout 8bits
                                # Cluster 16bits
                                # Direction 8bits
                                # Manufacturer_spec 8bits
                                # Manufacturer_id 16 bits
                                # Number of attributes 8 bits
                                # List of attributes ( 16bits )
                                sendZigateCmd("0100", "02" + str(key) + "01" + "01" + "0008" + "00" + "00" + "0000" + "01" + "0000" )

pipiche38, 2018-07-31:
La bonne nouvelle est que je récupère bien les infos des volets. La mauvaise nouvelle c'est que bien que sValue soit correctement positionné pour le Device en question, cela n'est pas visible sur Domoticz

Dans l'exemple suivant le véritable Level est 25% je m'attendrais donc à avoir 25% de la barre bleu et non pas 100% !
[image]

pipiche38, 2018-07-31:
Salut , j'ai posté ce que j'ai fait sur le #105 . Si tu veux jeter un oeil. Il me reste le probleme du % qui n'est pas mis à jour, bien que la valeur dans sValue soit correcte.

pipiche38, 2018-08-02:
Pour adresser le problème initialement reporté, après investigation voici l'info que j'ai obtenu depuis le Forum anglophone:

> onHeartBeat events are posted to the Plugin Framework's event queue once for every 'interval' you specify. The framework itself will service the queue as quickly as possible and process any events that are ready.
> 
> If you have some they could be the problem if they are taking a long time to process events. Plugins that use urllib or http.client connectivity rather than the built in Connection objects can block the framework's event queue if they fail to connect which could cause things like this. Try disabling other plugins 1 by 1 to see if one of them is the issue 

En autres termes et après revue, j'avais un plugin qui  impacte le reste du framework. Pour votre info c'est un plugin qui permet de monitorer une MAC adresse via bluetooth. Il y a de forte chance qu'il y ai un appel bloquant dedant, et qui bloque l'ensemble.

Je clos donc le case cone not-an-issue ;-)

closed 2018-08-02 by pipiche38