zigbeefordomoticz/Domoticz-Zigbee #105: Implement Read Attribute - Cluster 0x0008 - attribute 0x0000

pipiche38, 2018-07-31:
In order to get the Level position of the Profalux shutter, the Cluster 0x0008 must be implemented.

This is for instance needed as per #80

pipiche38, 2018-07-31:
Here are what I have currently implemented and which give pretty good results.

The remaining issue is that the Level bar on the Domoticz page is not aligned with the %level !

The purpose is to send a Read Attribute Request to Zigate . The call will be placed in onHeartbeat, but could be also called when a Domoticz change is made on the device .  The response of this request will be managed in the onMessage() 
```
def ReadAttributeRequest_0008(self, key) :
        # Cluster 0x0008 with attribute 0x0000
        # frame to be send is :
        # DeviceID 16bits / EPin 8bits / EPout 8bits / Cluster 16bits / Direction 8bits / Manufacturer_spec 8bits / Manufacturer_id 16 bits / Nb attributes 8 bits / List of attributes ( 16bits )
        Domoticz.Debug("Request Control level of shutter via Read Attribute request : " + key)
        sendZigateCmd("0100", "02" + str(key) + "01" + "01" + "0008" + "00" + "00" + "0000" + "01" + "0000" )
        return
```
In ReadCluster(self, MsgData) I have added the following statement. Basically 8100 is the response of the 0x0008 / 0x0000 Read Attribute. When receiving this cluster, we will request an update of the Domoticz Device
```

        elif MsgClusterId=="0008" :  # (Cluster Level Control )
                Domoticz.Debug("ReadCluster (8100) - ClusterId=0008 - Level Control : " + str(MsgClusterData) )
                MajDomoDevice(self, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgClusterData)
                return
```

In MajDomoDevice(self, .... ) I have added the following statement. Basically when having to update for a Level Control type, we will check if there is any changes or not and in case of change we will update it

```

                        if Type==Dtypename=="LvlControl" :
                                Domoticz.Debug("MajDomoDevice LvlControl - DvID : " + str(DeviceID) + " - Device EP : " + str(Ep) + " - Value : " + str(int(value,16)) + " sValue : " + str(Devices[x].sValue) )
                                state="1"
                                percent =  round((int(value,16)/255)*100)
                                if value == "01" or value == "00" :
                                        state="0"
                                if str(percent) == "0" and str(Devices[x].sValue) != "Off" :
                                        Domoticz.Log("MajDomoDevice update DevID : " + str(DeviceID) + " from " + str(Devices[x].sValue) + " to " + str(state) )
                                        UpdateDevice(x, str(state), str(percent) ,DOptions)
                                elif str(percent) != "0" and str(percent) != str(Devices[x].sValue) :
                                        Domoticz.Log("MajDomoDevice update DevID : " + str(DeviceID) + " from " + str(Devices[x].sValue) + " to " + str(percent) )
                                        UpdateDevice(x, str(state), str(percent) ,DOptions)
```

I have then implemented the Decode8100(self, MsgData) which was doing nothing until now
```

def Decode8100(self, MsgData) :  # Report Individual Attribute response
        MsgSQN=MsgData[0:2]
        MsgSrcAddr=MsgData[2:6]
        MsgSrcEp=MsgData[6:8]
        MsgClusterId=MsgData[8:12]
        MsgAttrID=MsgData[12:16]
        MsgAttType=MsgData[16:20]
        MsgAttSize=MsgData[20:24]
        MsgClusterData=MsgData[24:len(MsgData)]
        Domoticz.Debug("Decode8100 - reception data : " + MsgClusterData + " ClusterID : " + MsgClusterId + " Attribut ID : " + MsgAttrID + " Src Addr : " + MsgSrcAddr + " Scr Ep: " + MsgSrcEp)
        ReadCluster(self, MsgData)
        return
```

pipiche38, 2018-07-31:
Je vais voir comment cela se comporte, et préparer un Pull Request. Après il faudra voir si cela à une incidence.

Entre temps, j'ai reglé le problème et maintenant j'ai bien le volet avec la bonne barre ;-)

[image]

Il y a un Bug dans la gestion de nValue versus sValue sur le Set Level.

nValue 0 and sValue 0 ==>> Closed
nValue 2 and sValue 2-99 ==>> partially opened
nValue 1 and sValue 100 ==>> Opened

Du coup j'ai changé également cela dans le code au niveau de onCommand()
```

                if Command == "Set Level" :
                        if Dtypename == "LvlControl" :
                                if Level == 0 :
                                        sendZigateCmd("0092","02" + Devices[Unit].DeviceID + EPin + EPout + "00")
                                        UpdateDevice(Unit, 0, "0",DOptions)
                                if Level == 100 :
                                        value=returnlen(2,hex(round(Level*255/100))[2:])
                                        sendZigateCmd("0081","02" + Devices[Unit].DeviceID + EPin + EPout + "00" + value + "0010")
                                        UpdateDevice(Unit, 1, "100",DOptions)
                                else :
                                        value=returnlen(2,hex(round(Level*255/100))[2:])
                                        sendZigateCmd("0081","02" + Devices[Unit].DeviceID + EPin + EPout + "00" + value + "0010")
                                        UpdateDevice(Unit, 2, str(Level) ,DOptions)

```

thiklop, 2018-08-01:
Ca m'a l'air bien, rien à redire. Si j'avais seulement une suggestion ce serait d'encapsuler tout ça dans un try except, comme je le fait beaucoup actuellement au cas où  le message reçu ne serait pas strictement égal à celui prévu (regarde mon dernier commit si tu veux).

N'hésite pas à proposer un pull request, de tout façon je ne pourrai pas tester le comportement n'ayant pas de volet de ce type.

Rq : je suis en vacances, j'ai un peu moins mon ordi donc pas de panique si je ne réponds pas toujours du tac au tac.

pipiche38, 2018-08-01:
Ok, je vais regarder cela mais je ne suis pas un pro python ;-)

PS/ Je me demande si nous devrions pas créer une version Dev , car je n'ai pas d'autres devices ZigBee que les Volets profalux, et certaines des modifs peuvent avoir une incidence.

Notamment il semble que le LvlControl soit aussi utilisé par des lampes (variation de couleur et/ou d'intensité )

pipiche38, 2018-08-10:
Implemented and release with #edb7962e7ccbccd736aa7a9042c0024bfa1a16ed

closed 2018-08-10 by pipiche38