zigbeefordomoticz/Domoticz-Zigbee #107: Strange code when handling Set Level command for ColorControl

pipiche38, 2018-08-01:
Looks like when the level is different than 0, a value is calculated and sent to Zigate, but the DomoDevice is only updated with On ! Shouldn't be the calculated value passed to Zigate ?

```

                        if Dtypename == "ColorControl" :
                                if Level == 0 :
                                        sendZigateCmd("0092","02" + Devices[Unit].DeviceID + EPin + EPout + "00")
                                        UpdateDevice(Unit, 0, "Off",DOptions)
                                else :
                                        value=returnlen(4,hex(round(Level*1700/100))[2:])
                                        sendZigateCmd("00C0","02" + Devices[Unit].DeviceID + EPin + EPout + value + "0000")
                                        UpdateDevice(Unit, 1, "On",DOptions)
```

Impacted HW: Ampoule.LED1545G12.Tradfri

pipiche38, 2018-08-02:
Anyone using a Ampoule.LED1545G12.Tradfri whom can comment

zaraki673, 2018-08-08:
Salut Pipiche38,

il me semble en avoir une (mais c est encore dans un carton) j ai bientot fini de me reinstaller, je vais pouvoir regarder toutes vos nouveautés :D 
Et surtout expliquer le fonctionnement un peu plus en détail (meme si il me semble avoir vu un message de toi ou thiklop qui resume bien le fonctionnement avec le heartbeat ... #100 )

pour la valeur du lvl, j'avais un peu bidouiller si je puis dire, ce qui n est pas forcement bon coté compatibilité avec le reste, n hesite pas a modifier,  il fallait que je sorte une valeur comprise entre 1700 et 65000 il me semble 

;)

pipiche38, 2018-08-08:
Salut @zaraki673 j'ai mis un MP sur le forum francais... 

Oui je pense que j'ai compris la partie heartbeat et j'ai fait l'implementation pour les profalux ... A priori ca fonctionne bien.

Pour le lvl, cela me semble correct, cela doit correspondre au degré Kelvin (couleur) ;-)

pipiche38, 2018-08-22:
@Smanar peut-etre que c'est adressé dans les récentes modif pour la LED IKEA ?

Smanar, 2018-08-22:
Nope, pas fini, mais suis dessus.
En fait on a a laisse tomber le "levelcontrol" pour les couleurs pour un "ColorControl" directement.
Actuellement le systeme est batard car il considere la lampe ikea comme une RGBWWCW, du coup y a tout les reglages meme les non fonctionnels.
Sinon pour le rRGB Ca fonctionne
`sendZigateCmd("00B7","02" + Devices[Unit].DeviceID + EPin + EPout + strxy + "00")`
Pour le CW WW @zaraki673 a raison sur toute la ligne (me suis loupe sur la commande d'ailleurs),
`sendZigateCmd("00C0","02" + Devices[Unit].DeviceID + EPin + EPout + strTemp + "00")`
Je pourrais le mettre en place, mais pas (encore) le tester, car mon ampoule ne fait que le RGB.

Smanar, 2018-09-07:
We can close this issue now ?

pipiche38, 2018-09-08:
Sure ...

closed 2018-09-08 by pipiche38