zigbeefordomoticz/Domoticz-Zigbee #111: Xiaomi aqara cube not functionning

nx345, 2018-08-05:
I have installed the latest version of domoticz-zigate and correctly added a xiaomi aqara cube under domoticz. When I want for instance to trigger an action with the cube at 90 degrees I have this error message in my logs : 

2018-08-05 12:25:50.157 Error: (zigate) 'onMessage' failed 'KeyError'.
2018-08-05 12:25:50.157 Error: (zigate) ----> Line 311 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-08-05 12:25:50.157 Error: (zigate) ----> Line 118 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-08-05 12:25:50.157 Error: (zigate) ----> Line 382 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ZigateDecode
2018-08-05 12:25:50.157 Error: (zigate) ----> Line 626 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ZigateRead
2018-08-05 12:25:50.157 Error: (zigate) ----> Line 808 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function Decode8102
2018-08-05 12:25:50.157 Error: (zigate) ----> Line 1374 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ReadCluster
2018-08-05 12:25:50.157 Error: (zigate) ----> Line 955 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function MajDomoDevice

Therefore this does not trigger the action I want.

It seems like there is a problem with the plugin at these line aforementioned.

nx345, 2018-08-05:
@pipiche38 How can in put the plugin in debug mode ? Is there any documentation on this ?

pipiche38, 2018-08-06:
@nx345, from Domoticz 
-> Setup
     -> Hardware
         -> Select the Zigate Hardware
                 -> You'll find a field "Debug" select "Python Only"
                       -> Press "Update" (and not Add otherwise it will add a second Zigate Harwdare)

nx345, 2018-08-06:
2018-08-06 08:49:32.759 (zigate) Envoie une demande Active Endpoint request pour avoir la liste des EP du device adresse : 402f
2018-08-06 08:49:32.759 (zigate) sendZigateCmd - length is : 0002
2018-08-06 08:49:32.759 (zigate) getChecksum - Checksum : 0x28
2018-08-06 08:49:32.759 (zigate) ZigateEncode - Encodind data : 0045
2018-08-06 08:49:32.759 (zigate) Transcode in : 0045 / out :021045
2018-08-06 08:49:32.759 (zigate) ZigateEncode - Encodind data : 0002
2018-08-06 08:49:32.760 (zigate) Transcode in : 0002 / out :02100212
2018-08-06 08:49:32.760 (zigate) ZigateEncode - Encodind data : 28
2018-08-06 08:49:32.760 (zigate) Transcode in : 28 / out :28
2018-08-06 08:49:32.760 (zigate) ZigateEncode - Encodind data : 402f
2018-08-06 08:49:32.760 (zigate) Transcode in : 402f / out :402f
2018-08-06 08:49:32.760 (zigate) sendZigateCmd - Comand send : 010210450210021228402f03
2018-08-06 08:49:32.760 (zigate) HearBeat - Devices already exist. Unit=7
2018-08-06 08:49:32.760 (zigate) HearBeat - Devices already exist. Unit=8
2018-08-06 08:49:32.763 (zigate) HB count = 21
2018-08-06 08:49:32.863 (zigate) onMessage called
2018-08-06 08:49:32.864 (zigate) ZigateDecode - decodind data : 018102120210021fe21b402f021202101202105502102102100212021002127e03
2018-08-06 08:49:32.864 (zigate) ZigateRead - decoded data : 018102000fe21b402f02001200550021000200027e03
2018-08-06 08:49:32.864 (zigate) ZigateRead - Message Type : 8102, Data : 1b402f0200120055002100020002, RSSI : 7e, Length : 000f, Checksum : e2
2018-08-06 08:49:32.864 (zigate) ZigateRead - MsgType 8102 - Report Individual Attribute response : 018102000fe21b402f02001200550021000200027e03
2018-08-06 08:49:32.864 (zigate) Decode8102 - reception data : 0002 ClusterID : 0012 Attribut ID : 0055 Src Addr : 402f Scr Ep: 02
2018-08-06 08:49:32.865 (zigate) MajDomoDevice - Device ID : 402f - Device EP : 02 - Type : 0012 - Value : 0002
2018-08-06 08:49:32.865 Error: (zigate) 'onMessage' failed 'UnboundLocalError':'local variable 'data' referenced before assignment'.
2018-08-06 08:49:32.865 Error: (zigate) ----> Line 311 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-08-06 08:49:32.865 Error: (zigate) ----> Line 118 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function onMessage
2018-08-06 08:49:32.865 Error: (zigate) ----> Line 382 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ZigateDecode
2018-08-06 08:49:32.865 Error: (zigate) ----> Line 626 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ZigateRead
2018-08-06 08:49:32.865 Error: (zigate) ----> Line 808 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function Decode8102
2018-08-06 08:49:32.865 Error: (zigate) ----> Line 1369 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function ReadCluster
2018-08-06 08:49:32.865 Error: (zigate) ----> Line 1112 in /home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py, function MajDomoDevice

pipiche38, 2018-08-06:
Super ...

Could you let me know what should be expected ? As from the source code I see that there is this value coming from Cube Xiaomi, but not expected !

Is one of them ?

>     move : s’obtient en bougeant le cube par exemple en le déplacant sur une table. Assez difficile à utiliser au quotidien en déclencheur
>     tap_twice : alors il ne faut pas tapper deux fois le cube, mais le tapper lui deux fois sur la table par exemple. Facile à utiliser
>     shake_air : shake comme shaker, voilà vous avez compris le principe on le secoue bien fort mais pas trop prêt de la porcelaine de la belle mère
>     alert : alors, le plus sensible, le plus inutilisable du coup, se déclenche dès qu’on fait quelque chose au cube ou presque
>     flip90 : tourne de 90, donc d’une face à une autre
>     flip180 : tourne de 180 (vous l’aviez vu venir je sais), donc 2 faces (attention, en une seule fois, sinon c’est 2 * flip90)
>     free_fall : tombe, alors tombe ou chute, mais si vous le lancez en l’air, après la montée vient la ?
>     rotate_left : ca c’est quand on tourne le cube sur surface plane dans le sens horaire
>     rotate_right : ca c’est … ok je sais vous avez saisi

nx345, 2018-08-06:
When I flip the cube with 90 degrees it should trigger and http request that turn on a lamp. The http url has been tested successfully on a web browser. Then I added this url to the cube (flip 90).

pipiche38, 2018-08-06:
@nx345  could you have a look to the French forum. I have send you an email. I would you send you a revised version of the plugin. 

BTW, the problem you got with Debug mode is not the same as the 1st one :-(

pipiche38, 2018-08-06:
By chance, when you do it with the browser, do you have any values which might be return when doing this action ? For instance in that case the value was "0002" which was not expected in the plugin

pipiche38, 2018-08-06:
I found this interesting thread where I thing the Cube has been reverse-engineering 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/138#issuecomment-325101635

I was as well. Let's renumber the sides, so opposite sides add up to 7 (and the opposite side can be computed by 7 - side). Taking a small sample of the dice I have at home, with 1 on top and 2 on front, 3 is on the right. So the numbers would become:

>     +---+
>     | 5    |
> +---+---+---+
> | 4   |    1 | 3  |
> +---+---+---+
>         | 2   |
>        +---+
>        | 6   |
>        +---+
> 

where side 2 holds the MI logo and side 6 has the battery door.

    For the 90° flip it is more difficult, as you could go 90° left, right, forward of backward...

The cube doesn't know its orientation. It only knows that you flip to side 1 from side 3, which could be front, right, back, left, depending on the orientation.

    I'm for an abstract simple approach like the six sides mapped to buttonevent 1000–6000 with meaningful actions press, release, shake, etc.

Let's try that for now. We need to add 7000 for gestures where there's no side (shake, rotate). As for the action, I would use the previous side here, with 0 for push, the same side for double tap, the opposite side for 180° flip, and the other four sides for 90° flip. This leads to:

    x000 for push, so 1000, 2000, 3000, 4000, 5000, 6000;
    x00x for double tap, so 1001, 2002, 3003, 4004, 5005, 6006;
    x00o for 180° flip (where o = 7 - x), so 1006, 2005, 3004, 4003, 5002, 6001;
    x00y for 90° flip (where y ≠ x and y ≠ 7 - x), so 1002, 1003, 1004, 1005, 2001, 2003, 2004, 2006, 3001, 3002, 3005, 3006, 4001, 4002, 4005, 4006, 5001, 5003, 5004, 5006, 6002, 6003, 6004, 6005;
    7007 for shake;
    7008 for drop ;
    7000 for wakeup (the cube sends raw value 2 when I just tap it after a while).

pipiche38, 2018-08-07:
One more information from the net : 
https://community.smartthings.com/t/release-xiaomi-mi-cube-magic-controller/70669/74

> 
> 
> First byte of the value contains motion type, second byte in case of flip motion contains an information about which side was flipped to and from which other side.
> 
> The second byte should be split as follows:
> 
>     bits 1 and 2 - flip type, 01 for 90 degree, 10 for 180 degree flip
>     bits 3 to 5 - number of face motion originated on (always 0 for 180 flip)
>     bits 6 to 8 - number of face motion finished on
> 
> So for example motion from face 5 to face 4 would have value of ‘006c’ where first byte ‘00’ - points to motion type (flip or shake) and second byte ‘6c’ or ‘01101100’ in binary - 01 for 90 degree flip, flowed by 101 for ‘source’ face 5 and 100 for ‘target’ face 4.

pipiche38, 2018-08-07:
I ahve also investigated into domoticz and Xiaomi Gateway implements the Cube switch selector with those levels  `flip90/flip180/move/tap_twice/shake_air/swing/alert/free_fall`

Which seems not be in sync with what the Zigate plugin implements !
`Shake|Slide|90°|Clockwise|Tap|Move|Free Fall|Anti Clockwise|180°`

pipiche38, 2018-08-09:
I have created a development branch in order to work on that issue independently to the master.

pipiche38, 2018-08-22:
Fully implemented in developement branch a5cfad26b01383ea3a4c56c3967f2bf60f452e7c

closed 2018-08-22 by pipiche38