zigbeefordomoticz/Domoticz-Zigbee #116: Housekeeping/cleaning in order to maintain databases consistency

pipiche38, 2018-08-14:
A long the Zigate plugin, we are currently dealing with 3 databases of informations:
- List of Devices paired in Zigate
- List of Devices known by Zigate plugin in the ascii file DeviceList.txt
- List of Devices known by Domoticz

=> When you remove a device from Domoticz, that one is not removed from the DeviceList.txt and not from the Zigate, which then create some inconsistency.

The idea is to have a housekeeping task done once a while to check consistency and either to report or take action.

For instance:
(1) If on a Permit Join time, the plugin must try to onboard each device known by Zigate into the plugin and Domoticz through the normal process.
(2) If not on a Permit Join time, any device known in Zigate and not present in Domoticz must be remove

zaraki673, 2018-09-26:
i think we can move write to DeviceList.txt in debug mode, then we will have only two list ?

pipiche38, 2018-09-27:
@zaraki673 could be a way.  But we need to fix the update in Domoticz DB of Zigate data

pipiche38, 2018-11-22:
Since Beta-3.1 the Zigate database is stored in the plugin dictionary (ListOfDevices) which is then dump/serialized on DeviceList.txt
This dictinary also own the relation with the DOmoticz ID.

closed 2018-11-22 by pipiche38