zigbeefordomoticz/Domoticz-Zigbee #125: EndPoints numbering and naming convention

pipiche38, 2018-08-21:
4/
J'ai commencé à ajouter le numéro des EndPoints pour le nommage par défaut des "devices Domoticz", je trouve cela pratique (surtout quand j’avais tout en double) : à continuer ou à supprimer.

cf: https://easydomoticz.com/forum/viewtopic.php?f=8&t=5254&start=480#p57711

thiklop, 2018-08-21:
Je ne comprends pas ce que tu as fait. J'ai des capteurs Température/Humidité/Pression, est-ce que cela signifie qu'il faut créer trois lignes différentes ?

pipiche38, 2018-08-21:
Voici l'exemple du fichier de DeviceConf.txt qui a été proposé par cld
'PST03A-v2.2.5':{'Ep':{'01':{'0406':'','0500':'','0400':'','0402':'','Type':'Door/Switch/Lux/Temp'},'02':{'0500':'','Type':'Motion'}},'Type':'Motion/Door/Switch/Lux/Temp'},

Je pense que l'idée est pas mal.
(1) elle permet de faire fonctionner ce capteur
(2) elle offre la flexibilité d'avoir effectivement des capteurs plus complexes avec plusieurs EPs et des fonctions différentes.

pipiche38, 2018-08-21:
Le code proposé est compatible avec les 2 modes, donc ce n'est que du plus ....

pipiche38, 2018-08-22:
Il faudrait que @cldfr commente

cldfr, 2018-08-22:
Mon commentaire 4/ faisait référence à cette modification :

Code origine :  

if t=="Temp" : # Detecteur temp
					self.ListOfDevices[DeviceID]['Status']="inDB"
Domoticz.Device(DeviceID=str(DeviceID),Name=str(t) + " - " + str(DeviceID), Unit=FreeUnit(self), TypeName="Temperature", Options={"Zigate":str(self.ListOfDevices[DeviceID]), "TypeName":t}).Create()

Le code proposé sur le forum (dans ma modification) : 
				if t=="Temp" : # Detecteur temp
					self.ListOfDevices[DeviceID]['Status']="inDB"
					Domoticz.Device(DeviceID=str(DeviceID),Name=str(t) + " - " + str(DeviceID) + " - " + str(Ep), Unit=FreeUnit(self), TypeName="Temperature", Options={"Zigate":str(self.ListOfDevices[DeviceID]), "TypeName":t}).Create() # Ajout de l'EP pour différentier dans Domoticz un même type de capteur sur deux EP diffents d'un même deviceD

J'ai juste ajouté :  + " - " + str(Ep) 
Ce qui permet d'avoir dans le nom proposé le nom du EndPoint. 

J'ai fait cette modification à 4 endroits seulement : Temp, Lux, Switch,  Door (périphérique Domoticz que j'ai ajouté pour ce dernier), d’où mon commentaire: à continuer ou à supprimer

Par ailleurs, je pense qu'il serait bien d'enlever les espaces dans les noms proposés, pour éviter d'avoir des espaces dans le noms des fichiers scripts LUA .

Actuellement :
"XXXX - Temp"  par "XXXX-Temp"
ou "XXXX - Temp - 01" (avec ma modification)  par "XXXX-Temp-01"

Par ailleurs, il n'y a pas de capteur humidité et pression dans le composant ZigBee que j'ai utilisé (juste Temp et Lux).

Pour des capteurs "Température/Humidité/Pression" ça ne change rien. 

Par exemple : la modification proposées permet à un composant ZigBee qui proposerait deux switchs, l'un sur l'EP 1 l'autre sur l'EP 2 d'avoir deux noms différents.

XXXX-Switch-01 et XXXX-Switch-02 

Sinon, actuellement on à  2 fois : XXXX-Switch (ce qui n'est pas pratique pour savoir qui est qui :-))